lbbrhzn/ocpp #2: Error in connection handler

dreimer1986, 2021-03-17:
Maybe you have a clue why this happens quite often.

Logger: websockets.server
Source: custom_components/ocpp/sensor.py:254
First occurred: 9:26:24 (42 occurrences)
Last logged: 9:51:27

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/websockets/server.py", line 191, in handler
    await self.ws_handler(self, path)
  File "/config/custom_components/ocpp/sensor.py", line 108, in on_connect
    await cp.start()
  File "/config/custom_components/ocpp/sensor.py", line 270, in start
    await asyncio.gather(super().start(), self.post_connect())
  File "/config/custom_components/ocpp/sensor.py", line 145, in post_connect
    await self.trigger_boot_notification()
  File "/config/custom_components/ocpp/sensor.py", line 177, in trigger_boot_notification
    resp = await self.call(req)
  File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 264, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/sensor.py", line 254, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 292, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

lbbrhzn, 2021-03-17:
It looks like sending the trigger_boot_notification to your charge point results in a drop in the connection, perhaps due to reboot. I think these messages should best be ignored, not sure how to suppress them to prevent them being logged.

lbbrhzn, 2021-03-17:
Another improvement would be to send the trigger_boot_notification only on request, for instance by adding it as a service. 
This would allow you to add a button to the ha dashboard that triggers the boot notification when pressed.

dreimer1986, 2021-03-17:
Funny thing is, the station just sits there. I was at work and the device should have been active all the time. But something is not quite right as it seems. Status active sometimes comes up, but most of the time it's on unknown state too. And... What is the trigger_boot_notification btw? ^^

lbbrhzn, 2021-03-17:
The charge point sends a boot notification to the central system at boot, in order to register itself with the central server The central system  then responds with a confirmation.  The trigger boot notification kickstarts this registration process from the central server.

lbbrhzn, 2021-03-17:
Here's example of a log when the boot notification is succesfully triggered:

2021-03-17 20:20:20 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-03-17 20:20:21 WARNING (MainThread) [homeassistant.loader] You are using a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"d63f7e70-68e8-4cf3-ab3d-7dade9a7efc9","TriggerMessage",{"requestedMessage":"BootNotification"}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"d63f7e70-68e8-4cf3-ab3d-7dade9a7efc9",
{
"status": "Accepted"
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"652b5206-14ef-4c10-895b-d234afb67d56","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"652b5206-14ef-4c10-895b-d234afb67d56",
{
"status": "Accepted"
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"f15d531a-9e70-4793-a4a5-18f931d37dff","RemoteStopTransaction",{"transactionId":1234}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"f15d531a-9e70-4793-a4a5-18f931d37dff",
{
"status": "Rejected"
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"aeaabe65-6b8c-45f4-a7db-29f907f39896","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"aeaabe65-6b8c-45f4-a7db-29f907f39896",
{
"status": "Accepted"
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"75e673b4-a80d-454e-ac99-bf37509d5d3d","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"75e673b4-a80d-454e-ac99-bf37509d5d3d",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": false,
"key": "MeterValuesSampledData",
"value": "Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Register,Energy.Active.Import.Register,Energy.Reactive.Export.Register,Energy.Reactive.Import.Register,Energy.Active.Export.Interval,Energy.Active.Import.Interval,Energy.Reactive.Export.Interval,Energy.Reactive.Import.Interval,Frequency,Heartbeat,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Status,Temperature,Voltage"
} 
]
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"1c79b083-6c25-4c17-a8c0-a2286a3c7664","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"1c79b083-6c25-4c17-a8c0-a2286a3c7664",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": false,
"key": "MeterValueSampleInterval",
"value": "10"
} 
]
}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: send [2,"8778c086-3358-4e3b-8f17-739e7c8d208a","RemoteStartTransaction",{"idTag":"ID4","connectorId":1,"chargingProfile":{"chargingProfileId":1,"stackLevel":999,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxProfile","chargingSchedule":{"duration":36000,"chargingRateUnit":"W","chargingSchedulePeriod":[{"startPeriod":0,"limit":22000}]}}}]
2021-03-17 20:23:00 INFO (MainThread) [ocpp] WallboxPulsar: receive message [3,"8778c086-3358-4e3b-8f17-739e7c8d208a",
{
"status": "Accepted"
}]
2021-03-17 20:23:15 INFO (MainThread) [ocpp] WallboxPulsar: receive message [2,"1540383426","BootNotification",
{
"chargePointSerialNumber": "88034",
"chargePointVendor": "Wall Box Chargers",
"meterType": "Internal NON compliant",
"meterSerialNumber": "",
"chargePointModel": "PLP1-0-2-4",
"iccid": "",
"chargeBoxSerialNumber": "88034",
"firmwareVersion": "4.7.6",
"imsi": ""
}]
2021-03-17 20:23:15 INFO (MainThread) [ocpp] WallboxPulsar: send [3,"1540383426",{"currentTime":"2021-03-17T19:23:15.220260","interval":10,"status":"Accepted"}]
2021-03-17 20:23:15 INFO (MainThread) [ocpp] WallboxPulsar: receive message [2,"35005211","StartTransaction",
{
"connectorId": 1,
"meterStart": 0,
[... 558 lines cut ...]

{
"value": "32",
"format": "Raw",
"location": "Outlet",
"context": "Sample.Periodic",
"measurand": "Current.Offered",
"unit": "A"
},

{
"value": "333",
"format": "Raw",
"location": "Outlet",
"context": "Sample.Periodic",
"measurand": "Energy.Active.Import.Register",
"unit": "Wh"
} 
]
} 
],
"connectorId": 1,
"transactionId": 1234
}]
2021-03-17 20:24:56 INFO (MainThread) [ocpp] WallboxPulsar: send [3,"927612902",{}]

dreimer1986, 2021-03-17:
Hmm, what do I have to do for these extended logs in the custom component? I guess right now I just get errors and nothing more. Maybe there is some stuff missing?

lbbrhzn, 2021-03-17:
This should be enabled by default, it is in the ocpp/sensor.py:

````
_LOGGER = logging.getLogger(__name__)
logging.getLogger("ocpp").setLevel(logging.DEBUG)
````
The log messages are  added to the home_assistant.log file.

lbbrhzn, 2021-03-17:
It's possible you configured the logger to suppress these messages. See https://www.home-assistant.io/integrations/logger/

lbbrhzn, 2021-03-17:
i've updated to code to skip the trigger of boot notification. can you check if it helps?

dreimer1986, 2021-03-17:
Might be true... I changed it to:

```
logger:
  default: info
  logs:
    custom_components.ocpp: debug
```

Now I wait a bit for more information

lbbrhzn, 2021-03-17:
The logging for HACS based integrations is actually controlled in a different way:

https://hacs.xyz/docs/basic/logs/

dreimer1986, 2021-03-17:
So more like that?
```
logger:
  default: info
  logs:
    custom_components.hacs: debug
    ocpp: debug
```

lbbrhzn, 2021-03-17:
Honestly, I don't know for sure. I don't have any logger configuration in my configuration.yaml file.

I do have default_config: in there, that could set up some logging defaults.

dreimer1986, 2021-03-17:
I got this for you: https://gist.github.com/dreimer1986/b31ffdec9ab862ea9dd00506f67bb32b Quite long, so as GIST

lbbrhzn, 2021-03-17:
Ok. Looks like home assistant never received a confirmation of the trigger boot notification. Maybe your charger goes into reboot immediately, without sending the confirmation. I can try to remove that trigger altogether, or maybe you can do this yourself. You you have the visual code plug-in installed, or do you have file access to the file system used by home assistant?

dreimer1986, 2021-03-17:
I have the file explorer addin up and running. So I can edit as you ask for.

lbbrhzn, 2021-03-17:
Ok. Just remove the line 

await self.trigger_boot_notification

In custom_components/ocpp/sensor.py

And restart HA

dreimer1986, 2021-03-17:
Seems like I got a special one here as I always seems to prefer... ^^
```
2021-03-17 21:31:13 DEBUG (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 connected.
2021-03-17 21:31:13 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"d5079a8d-6c03-4b0f-a032-4ebe7ebad6fc","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e52390-df13-411d-b782-739be3c33cab","BootNotification",{"chargePointVendor":"EV-BOX","chargePointModel":"G4-WIFI-F6017","chargePointSerialNumber":"20482857","firmwareVersion":"P0417B0417v0_785.200220_U6.0.0-050"}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e52390-df13-411d-b782-739be3c33cab",{"currentTime":"2021-03-17T20:31:14.200482","interval":10,"status":"Accepted"}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e52392-df13-451d-b782-739bf23bb6ac","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError","timestamp":"2021-03-17T20:31:14Z"}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e52392-df13-451d-b782-739bf23bb6ac",{}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e52392-df13-451d-b782-739b144d3bae","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","timestamp":"2021-03-17T20:31:14Z"}]
2021-03-17 21:31:14 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e52392-df13-451d-b782-739b144d3bae",{}]
2021-03-17 21:31:19 INFO (MainThread) [hass_nabucasa.google_report_state] Connected
2021-03-17 21:31:23 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e5239c-df13-451d-b782-739bc33b26e4","Heartbeat",{}]
2021-03-17 21:31:23 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e5239c-df13-451d-b782-739bc33b26e4",{"currentTime":"2021-03-17T20:31:23.721565"}]
2021-03-17 21:31:33 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523a5-df13-461d-b782-739b2111bc1f","Heartbeat",{}]
2021-03-17 21:31:33 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523a5-df13-461d-b782-739b2111bc1f",{"currentTime":"2021-03-17T20:31:33.726689"}]
2021-03-17 21:31:43 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523af-df13-461d-b782-739bb260595b","Heartbeat",{}]
2021-03-17 21:31:43 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523af-df13-461d-b782-739bb260595b",{"currentTime":"2021-03-17T20:31:43.717205"}]
2021-03-17 21:31:43 ERROR (MainThread) [websockets.server] Error in connection handler
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/websockets/server.py", line 191, in handler
await self.ws_handler(self, path)
File "/config/custom_components/ocpp/sensor.py", line 110, in on_connect
await cp.start()
File "/config/custom_components/ocpp/sensor.py", line 275, in start
await asyncio.gather(super().start(), self.post_connect())
File "/config/custom_components/ocpp/sensor.py", line 147, in post_connect
await self.trigger_status_notification()
File "/config/custom_components/ocpp/sensor.py", line 171, in trigger_status_notification
resp = await self.call(req)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 264, in call
await self._get_specific_response(call.unique_id,
File "/config/custom_components/ocpp/sensor.py", line 259, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 292, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(),
File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2021-03-17 21:31:48 DEBUG (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 connected.
2021-03-17 21:31:48 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"ef2fd0ed-b8a5-4070-a9ee-8f7d43105432","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-03-17 21:31:53 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523b9-df13-461d-b782-739b7c90f496","Heartbeat",{}]
2021-03-17 21:31:53 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523b9-df13-461d-b782-739b7c90f496",{"currentTime":"2021-03-17T20:31:53.718608"}]
2021-03-17 21:32:03 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523c3-df13-461d-b782-739baccd91d2","Heartbeat",{}]
2021-03-17 21:32:03 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523c3-df13-461d-b782-739baccd91d2",{"currentTime":"2021-03-17T20:32:03.729934"}]
2021-03-17 21:32:13 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523cd-df13-461d-b782-739b3677270e","Heartbeat",{}]
2021-03-17 21:32:13 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523cd-df13-461d-b782-739b3677270e",{"currentTime":"2021-03-17T20:32:13.716836"}]
[... cut ...]
2021-03-17 21:32:23 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"e504c4c4-48bc-4497-a7e9-6532f73a4008","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-03-17 21:32:23 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e523d7-df13-461d-b782-739b2ad19a4a","Heartbeat",{}]
2021-03-17 21:32:23 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e523d7-df13-461d-b782-739b2ad19a4a",{"currentTime":"2021-03-17T20:32:23.867768"}]
```

dreimer1986, 2021-03-17:
After it now comlained about the next line, I commented that out too, then it complained about the one after that... Now I have commneted out the three await before sending configure and now it complains about this one.

```
2021-03-17 21:40:16 DEBUG (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 connected.
2021-03-17 21:40:16 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"355a0e1c-82ce-4697-a7e1-de829767f30d","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525af-df13-411d-b782-739b743e82ab","BootNotification",{"chargePointVendor":"EV-BOX","chargePointModel":"G4-WIFI-F6017","chargePointSerialNumber":"20482857","firmwareVersion":"P0417B0417v0_785.200220_U6.0.0-050"}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525af-df13-411d-b782-739b743e82ab",{"currentTime":"2021-03-17T20:40:17.124945","interval":10,"status":"Accepted"}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525b1-df13-451d-b782-739bb637ecac","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError","timestamp":"2021-03-17T20:40:17Z"}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525b1-df13-451d-b782-739bb637ecac",{}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525b2-df13-451d-b782-739b99b471ae","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","timestamp":"2021-03-17T20:40:17Z"}]
2021-03-17 21:40:17 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525b2-df13-451d-b782-739b99b471ae",{}]
2021-03-17 21:40:21 INFO (MainThread) [hass_nabucasa.google_report_state] Connected
2021-03-17 21:40:26 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525bb-df13-451d-b782-739b633426e4","Heartbeat",{}]
2021-03-17 21:40:26 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525bb-df13-451d-b782-739b633426e4",{"currentTime":"2021-03-17T20:40:26.621734"}]
2021-03-17 21:40:36 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525c4-df13-461d-b782-739b550dbc1f","Heartbeat",{}]
2021-03-17 21:40:36 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525c4-df13-461d-b782-739b550dbc1f",{"currentTime":"2021-03-17T20:40:36.618501"}]
2021-03-17 21:40:46 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525ce-df13-461d-b782-739b4864595b","Heartbeat",{}]
2021-03-17 21:40:46 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525ce-df13-461d-b782-739b4864595b",{"currentTime":"2021-03-17T20:40:46.616133"}]
2021-03-17 21:40:46 ERROR (MainThread) [websockets.server] Error in connection handler
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/websockets/server.py", line 191, in handler
await self.ws_handler(self, path)
File "/config/custom_components/ocpp/sensor.py", line 110, in on_connect
await cp.start()
File "/config/custom_components/ocpp/sensor.py", line 276, in start
await asyncio.gather(super().start(), self.post_connect())
File "/config/custom_components/ocpp/sensor.py", line 150, in post_connect
await self.configure(
File "/config/custom_components/ocpp/sensor.py", line 231, in configure
resp = await self.call(req)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 264, in call
await self._get_specific_response(call.unique_id,
File "/config/custom_components/ocpp/sensor.py", line 260, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 292, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(),
File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2021-03-17 21:40:51 DEBUG (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 connected.
2021-03-17 21:40:51 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"ab711f8c-2579-4136-9177-1b785f221810","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-03-17 21:40:56 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525d8-df13-461d-b782-739b9597f496","Heartbeat",{}]
2021-03-17 21:40:56 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525d8-df13-461d-b782-739b9597f496",{"currentTime":"2021-03-17T20:40:56.618710"}]
2021-03-17 21:41:06 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525e2-df13-461d-b782-739b56d191d2","Heartbeat",{}]
2021-03-17 21:41:06 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [3,"27e525e2-df13-461d-b782-739b56d191d2",{"currentTime":"2021-03-17T20:41:06.618852"}]
2021-03-17 21:41:16 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: receive message [2,"27e525ec-df13-461d-b782-739b3772270e","Heartbeat",{}]
[... cut ...]
File "/config/custom_components/ocpp/sensor.py", line 260, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 292, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(),
File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

For me it seems almost like whatever we pass, it hates what it gets.

lbbrhzn, 2021-03-17:
Yes, looks like it. Have you tried rebooting your charge point, and then home assistant?

lbbrhzn, 2021-03-17:
and is your charge point really OCPP 1.6 compliant?

dreimer1986, 2021-03-17:
It claims to be. We even had that support as reason to buy. https://evbox.com/en/products/home-chargers/elvi Check certifications. In tech specs we have: OCPP 1.5 S and 1.6 J. Trying the reboot now.

lbbrhzn, 2021-03-17:
and they are celebrating their certification here: https://news.evbox.com/nl-NL/184402-evbox-is-eerste-ocpp-1-6-gecertificeerde-partner-wereldwijd-door-open-charge-alliance

dreimer1986, 2021-03-17:
Yeah. Btw, I rebooted the wallbox and had a HassOS reboot parallel to it. HA was back on after the box. Results: No change at all.
OK, before we go crazy here, I call the support tomorrow. They say no support for custom OCPP solutions, but I will ask if the firmware installed is up to date. The app always says "unknown version" and that is strange. Rest of the app works fine though. We did set up the whole thing that way.

What makes me wonder a bit is that: https://webcache.googleusercontent.com/search?q=cache:vLRnn2gId0UJ:https://intercom.help/true-energy-8e0af048b9de/en/articles/4886123-evbox+&cd=3&hl=de&ct=clnk&gl=de This is a older cached version of a manual set up how to of a  OCPP server provider by now officially supported by EVBOX. Why that special /evbox/ at the end of the url? Why do they need the serial? Some login needed or what?

lbbrhzn, 2021-03-17:
yes, there is a login / authentication scheme.  See section 3.1.1 of the specification:
[ocpp-j-1.6-specification.pdf](https://github.com/lbbrhzn/ocpp/files/6159734/ocpp-j-1.6-specification.pdf)

dreimer1986, 2021-03-17:
And 6.2.2 tells something about a key needed for authentication... I got that EVB... chargepoint ID which seems to be already known AND I have a 10 digit security code. Both were on a sticker. Maybe thats the problem? I need the key for Install mode in app too btw aka for changing the OCPP server, activating WIFI etc. Well Well, at least I already got the firmware version out of it thx to the addin. Now I can go ask whats maybe wrong.

lbbrhzn, 2021-03-17:
In OCPP the charge point authenticates to the central system. There is no mechanism for the central system to authenticate to the charge point.

dreimer1986, 2021-03-18:
So, had a nice chat with the EVBOX hotline. Somehow changing the OCPP from Everon (default) to something else kills... almost everything. So... we have to wait for a update. The box right now never leaves the Everon service at all. Regardless what you do. And the settings on Everon are completely randomized after trying to. Somehow the Box now had 2 charge connectors and was always charging O_o. OK, thanks for the support. For now we have to wait.

lbbrhzn, 2021-03-18:
Ok, good to know. Are you paying a monthly fee currently? I can imagine vendors that charge such a fee would not like you to leave, and using OCPP would enable you to do exactly that.

dreimer1986, 2021-03-18:
Nope, all free. Lets see and wait ^^ I did set up the box again now and they are releasing a new firmware next week they said.

dreimer1986, 2021-03-29:
Still no update. As the problem is the wallbox and not your solution, I will for now close this report.

dreimer1986, 2021-06-16:
```
2021-06-16 18:23:41 INFO (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 connected.
2021-06-16 18:23:41 INFO (MainThread) [ocpp] http://192.168.181.42:9000/EVB-P20482857: send [2,"230ed1aa-52de-4bfd-a2d0-4c9642179f3c","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-06-16 18:24:11 ERROR (MainThread) [websockets.server] Error in connection handler
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/websockets/server.py", line 191, in handler
await self.ws_handler(self, path)
File "/config/custom_components/ocpp/sensor.py", line 175, in on_connect
self._cp_metrics = await cp.start()
File "/config/custom_components/ocpp/sensor.py", line 388, in start
await asyncio.gather(super().start(), self.post_connect())
File "/config/custom_components/ocpp/sensor.py", line 214, in post_connect
await self.get_supported_features()
File "/config/custom_components/ocpp/sensor.py", line 238, in get_supported_features
resp = await self.call(req)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 264, in call
await self._get_specific_response(call.unique_id,
File "/config/custom_components/ocpp/sensor.py", line 371, in _get_specific_response
resp = await super()._get_specific_response(unique_id, timeout)
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 292, in _get_specific_response
response = await asyncio.wait_for(self._response_queue.get(),
File "/usr/local/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2021-06-16 18:24:16 DEBUG (MainThread) [custom_components.ocpp.sensor] Charger http://192.168.181.42:9000/EVB-P20482857 reconnected.
```

Finally new firmware. Sadly it seems like this box does not wanna be used away from their supported charge management sites...

Tested with c84ff1f but with reverted 893f15f as it causes HA to not restart anymore due to syntax errors.

drc38, 2021-06-17:
It should not be reporting `http://` in the log as this is the wrong protocol it should be `ws://` try updating your ocpp url to either `ws://192.168.181.42:9000` or simply `192.168.181.42:9000`

dreimer1986, 2021-06-17:
Well, it is ws://192.168.181.42:9000/ Will verify later at home again, but I am only able to use ws:// in that charger.

drc38, 2021-06-17:
You can also use Wireshark to verify the initial websocket handshake is conforming to the ocpp specification. It won't show in the HA log.

drc38, 2021-06-17:
A charger that has connected correctly to the websocket server should report `Charger EVB-P20482857 connected`

dreimer1986, 2021-06-17:
Will create some Wireshark logs from my network with the default Online OCPP target it uses and with the correct ws:// target later today. Maybe we can find out a few things that way. They claim to be OCPP supportive and even celebrated that confirmed conformity last year. (See Mar 17th discussion in here) Firmware is the newest right now.

dreimer1986, 2021-06-17:
[WS Logs.zip](https://github.com/lbbrhzn/ocpp/files/6672069/WS.Logs.zip)

Here war are.
Timestamp 1857 goes from 192.168.181.62 to evbox.everon.io and works.
Timestamp 1916 goes from 192.168.181.62 to 192.168.181.42:9000 and does not work.

I have zero clue about this stuff, but if I see things somehow right in there it first sends infomration about supported OCPP and some Token stuff to the addin and then the addin answers that it's not valid http request?

drc38, 2021-06-17:
Afraid I can't help with websockets. Wikipedia has details on the handshake and standards the client should adhere to https://en.m.wikipedia.org/wiki/WebSocket

dreimer1986, 2021-06-18:
Well, maybe someone else already tinkered with it. I can only show what my box does and that it claims to have OCPP 1.6 in that http packets. I leave that here for others to confirm it or confirm that the box is not OCPP 1.6 supportive at all as it does stupid things instead of the way it should go.

dreimer1986, 2021-06-19:
I don't even know what SOAP is ^^ But yes... maybe.

lbbrhzn, 2021-06-19:
https://www.educba.com/json-vs-soap/

dreimer1986, 2021-06-19:
Well, if that is the case then the everon logs should contain SOAP protocol data passing around ...

lbbrhzn, 2021-06-19:
@dreimer1986, I see a TLSv1.2 request being made by your charge point to port 9000. Could that be related to the connection issue? I don't think the occp server currently supports TLS

dreimer1986, 2021-06-19:
I used ws:// and not wss:// so I think not, buuuut I would say this special case of a charge station could do it anyways? When I try wss:// nothing seems to happen at all.

drc38, 2021-06-19:
I'm guessing it is something in the chargers config, can you screenshot it?

dreimer1986, 2021-06-20:
Not much to see there.... https://photos.app.goo.gl/wXxDAjC9ij1j6Hx39

drc38, 2021-06-20:
Suggest logging a ticket with the supplier and providing your wireshark outputs

dreimer1986, 2021-06-20:
The only thing they will tell me is that they give a f*ck about that. There is only support for their included list of providers. As these work, noone will help me at all. The config app even shows a full screen message telling me that there is zero support for not supported solutions whenever I enter a custom ws:// or wss:// url. So that will not help us at all... Last time I tried to find something out they just told me that IN THEORY all OCPP 1.6 supporting solutions do work, but they only support the ones they tested and the rest is up to the user if it does anything. So for me it looks like it does some special things still sort of fine in the protocol, but not yet in the addin? I got to that assumption as you are talking about not working TLS and SOAP. Maybe I am wrong though ^^

jeremycoppey, 2021-07-06:
@dreimer1986 Did you retry the new version of this integration maybe in the meantime something got fixed?
Please remove the config from you config file and install the integration though the UI (see updated readme for info)

dreimer1986, 2021-07-06:
Already reconfigured it, but not yet tested it. Right now I have my car on it charging from 10%, so I could try tomorrow.

dreimer1986, 2021-07-07:
Sooo, tested it. Still not much coming from the device. Only thing which has information is charger.ID. There you can read http://192.168.181.42:9000/EVB-P20482857 I tried to open that link in a browser as http looks strange and got this:

```
Failed to open a WebSocket connection: invalid Connection header: keep-alive.

You cannot access a WebSocket server directly with a browser. You need a WebSocket client.
```

Rest all the same. I added all checkboxes on installing and all stuff it passes on is shown on my testing site. Only this one does show anything. I am quite sure that as the behavior is the same as before that the dumps don't need an update. What about this one? https://github.com/lbbrhzn/ocpp/issues/55 Didn't you say that it seems like my box wants to open a secured connection maybe?

lbbrhzn, 2021-08-23:
@dreimer1986, looks like your charger is trying to connect to your home assistant server (IP 192.168.181.32) on port 9000, with charger id EVB-P20482857. The OCPP integration is listening to 192.168.181.42:9000, and indeed it is a websocket that is not meant to be opened directly  from a browser. 

Can you try to grab the communication using wireshark again?

dreimer1986, 2021-08-24:
Attached a few new ones. IP of HA is 192.168.181.42, IP of the charger is 192.168.181.62. Just for fun I redid the same with a external forwarding to be 100% sure as this must be the correct IP as all my HA hosted stuff is running over this one either. Of course I used ws:// aka insecure.

[WS Logs NEW.zip](https://github.com/lbbrhzn/ocpp/files/7041032/WS.Logs.NEW.zip)
[image]
[image]

lbbrhzn, 2021-08-28:
@dreimer1986, I finally got around to take a look at your logs. I see a lot of strange TCP packets (TCP Out-Of_order, TCP Dup Ack, TCP Retransmission] from charger to HA

There are some packets coming through though:

From Charger to HA:
```
GET http://192.168.181.42:9000/EVB-P20482857 HTTP/1.1\r\n
```
Response:
```
Hypertext Transfer Protocol
    HTTP/1.1 101 Switching Protocols\r\n
    Upgrade: websocket\r\n
    Connection: Upgrade\r\n
    Sec-WebSocket-Accept: KJsG7JYcfQ2KBj3nLE62483cGH0=\r\n
    Date: Tue, 24 Aug 2021 17:09:03 GMT\r\n
    Server: Python/3.9 websockets/9.1\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 0.007920000 seconds]
    [Request in frame: 4696]
    [Request URI: http://192.168.181.42:9000/EVB-P20482857]
```
Followed by a Websocket packet from HA to charger:
```
Line-based text data (1 lines)
    [2,"368f39ff-02e5-4514-8636-ec4dc6cc5f66","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
```
This response is acknowledged on TCP level, but there is no response from the charger.

This pattern is repeated over and over again. So it looks like the charger does not respond the GetConfiguration command.

lbbrhzn, 2021-08-28:
On the ext logs, the pattern is the same, only withouth the ACKs

dreimer1986, 2021-08-29:
Network is A-OK. I can sort of guarantee that as it's my job to make these for customers and hopefully I know what I am doing there :P WIFI is a bit on the weaker side though as the way to the box is a lil longer, maybe that's the problem here. But as you said it seems to dislike something regarding the connection being accepted. Maybe the older logs where I left the configuration on Everon service are of any help? It communicates fine with that service as it's the one maintained by EVBOX.

dreimer1986, 2021-08-30:
Stupid question, what does "db dogs in spa" mean?
Any need for a few more logs/dumps/anything?

lbbrhzn, 2021-08-30:
it appears i accidentally created a milestone by typing in the wrong window

lbbrhzn, 2021-08-30:
This is the response from the Everon server:
``
Hypertext Transfer Protocol
    HTTP/1.1 101 Switching Protocols\r\n
        [Expert Info (Chat/Sequence): HTTP/1.1 101 Switching Protocols\r\n]
        Response Version: HTTP/1.1
        Status Code: 101
        [Status Code Description: Switching Protocols]
        Response Phrase: Switching Protocols
    Date: Thu, 17 Jun 2021 16:58:56 GMT\r\n
    Connection: upgrade\r\n
    Set-Cookie: ISESSIONID=node01ff5d7ai56yujwue9t38xir3g25516.node0; Path=/\r\n
    Expires: Thu, 01 Jan 1970 00:00:00 GMT\r\n
    Sec-WebSocket-Protocol: ocpp1.6\r\n
    Sec-WebSocket-Accept: 7dsHRS6AgV6wWNgNayVVCKVwoK8=\r\n
    Upgrade: WebSocket\r\n
    traceId: 45a607a8547869c8aaf22cdd3ee1df3d\r\n
    Via: 1.1 google\r\n
    CF-Cache-Status: DYNAMIC\r\n
    cf-request-id: 0abc82696f0000d7295d9d9000000001\r\n
    Server: cloudflare\r\n
    CF-RAY: 660dd35578d0d729-FRA\r\n
    \r\n
    [HTTP response 1/1]
```

[20 more replies]

closed 2021-11-15 by lbbrhzn