lbbrhzn/ocpp #6: EV car not charging after restart HASS/OCPP service

jeremycoppey, 2021-05-02:
I restarted HASS after some config changes (another component) and the car stopped charging.
On the wallbox I had a (un)/plug error but he re-established the connection with the OCPP service without any errors but didn't restart charging.
Restarted HASS around 11:45
At 12 I went to the car and removed and inserted the plug back into the car.
[image]

The hass config:
```
- platform: ocpp
   monitored_variables:
      - "Current.Export"
      - "Current.Import"
      - "Current.Offered"
      - "Energy.Active.Export.Register"
      - "Energy.Active.Import.Register"
      - "Energy.Reactive.Export.Register"
      - "Energy.Reactive.Import.Register"
      - "Energy.Active.Export.Interval"
      - "Energy.Active.Import.Interval"
      - "Energy.Reactive.Export.Interval"
      - "Energy.Reactive.Import.Interval"
      - "Frequency"
      - "Heartbeat"
      - "Power.Active.Export"
      - "Power.Active.Import"
      - "Power.Factor"
      - "Power.Offered"
      - "Power.Reactive.Export"
      - "Power.Reactive.Import"
      - "RPM"
      - "SoC"
      - "Status"
      - "Temperature"
      - "Voltage"
   port: 9000
   scan_interval:
    seconds: 60
```

My OCPP config on the wallbox: 
```
# Ocpp Conf

[OCPP]
ocppVersion=              1.6
ocppCentralAddress=       ws://192.168.1.21:9000
ocppBoxLocalPort=         8080
ocppBoxPublicIp=          
ocppBoxPublicPort=        8080
ocppBoxLocalSSL=          0
ocppBoxAddressReplyTo=    http://www.w3.org/2005/08/addressing/anonymous
ocppModemIpAddress=       192.168.0.254
ocppModemPresence=       0
modemType=                
confModemRetryInterval=   30
boxIdentity=              UNDEFINED
defaultIdTag=             UNDEFINED
isMeterValueSimulated=    false
MeterValuesAlignedData=   Energy.Active.Import.Register
MeterValuesSampledData=   Energy.Active.Import.Register
meterValueSampleInterval= 900
clockAlignedDataInterval= 0
ocppConnectTimeOut=       60
WebSocketPingInterval=    120
MinimumStatusDuration=    0
TransactionMessageRetryInterval=60
TransactionMessageAttempts=500
truncateBootNotificationsSerialNumbers=    true
serverCertificatePassword=UNDEFINED
clientHttpsOptions=       DEFAULT
cacheListEnabled=         false
enablePlugNumbering=      false
enableDoubleNotification=      false
lastAcceptedOcppCentralAddress=ws://192.168.1.21:9000/00:80:F4:43:0B:34
lastAcceptedBoxIdentity=       00:80:F4:43:0B:34
lastAcceptedHeartBeatInterval= 10
compressDiagnostic=            1
ocppEMDegradedModeEnabled=     0
EnergyLimitValue=       0
isEnergyLimitedActive=     false
```

jeremycoppey, 2021-07-06:
Keeping this issue open until I can retest this scenario

lbbrhzn, 2021-07-06:
ok, thnx!

lbbrhzn, 2021-11-11:
@jeremycoppey , any updates? Otherwise i'll close this issue due to inactivity...

lbbrhzn, 2022-02-04:
Closed due to inactivity

closed 2022-02-04 by lbbrhzn