lbbrhzn/ocpp #9: Wrap ChargePoint in a ChargePointEntity

lbbrhzn, 2021-06-14:
The current ChargePoint is an extension from ocpp.v16.ChargePoint.  To enable entity services, and allow multiple chargepoint instances, a chargepoint should be modeled as an homeassistant Entity. There are two options:

1. Use multiple inheritance to make ChargePoint both an ocpp.v16.ChargePoint as well as an homeassistant Entity
2. Create a new new ChangePointEntity class that contains a occp.v16.Chargepoint object.

I prefer option 2, this makes it much clearer which methods belong to which interface, and avoids naming clashes.

drc38, 2021-06-17:
These could be good starting points https://github.com/hesselonline/wallbox or https://github.com/fondberg/easee_hass

lbbrhzn, 2021-06-17:
Wallbox is based on reading data from the cloud. It is now integrated into home assistant core:
https://github.com/home-assistant/core/tree/dev/homeassistant/components/wallbox
It does not create separate sensor entities, only a single sensor with a lot of attributes.

Homematic uses local push, like us, and can create new objects on the fly.
https://github.com/home-assistant/core/tree/master/homeassistant/components/homematic

hesselonline, 2021-06-19:
hi guys, quick review here based on a request by @drc38. I'm the author of the wallbox component (which does actually seperate the sensor entities ;-)).

I'm just going to trow some first thoughts here, feel free to ignore them. Also, feel free to ask if you need more info. My point of reference is what you would need to do to get this added to HA core.

1. I think you are currently putting to much in the component (and in a single sensor.py file). Basic HA rule is use a Pypi package to do the communication and only do the HA implementation in the HA component. I think (but don't really have OCPP experience) that you are currently doing too much in the component. To remedy this I would start by moving stuff to a seperate py (chargepoint.py) file, I think that if you want to get this integrated you would need to make a seperate pypi package in the future.
2. I'm not sure if HA core allows opening websockets, might be a thing. Again, not sure. Also, you have a lot of sleep timers in you async code, specific reason for that?
3. https://developers.home-assistant.io/docs/development_environment explains how to setup a developer environment (I use VS code and devcontainers), this is really beneficial. It includes stuff like githooks for testing and other useful stuff. Also, https://developers.home-assistant.io/docs/creating_component_index explains how you can create a basic 'component scaffold' which uses the new config flow structure. This is a good place to start. If you get stuck here, let me know -> really worth the effort.
4. I understand that you want to implement multiple platforms for OCPP, best way to do that (after starting with point 3) is to move you main logic to __init__. You want to create a DataUpdateCoordinator there which handles all communication, this coordinator in turn populates entities (if you look at the wallbox files in HA core you will notice that the sensor file is quite bare, init is where the magic happens.)

hesselonline, 2021-06-19:
5. forgot to say: really nice and interesting project! I like the fact that you don't need cloud for monitoring you EV charger ;-)

lbbrhzn, 2021-06-19:
Thanks for your comments!
I have installed the wallbox integration in my ha setup, and it works great! 

As a future idea I would also like to make an integration for obd2, to connect with an elm327 based Bluetooth dongle in de car. This allows you to monitor things like State of Charge and battery temperature while charging, so you can adjust the charging profile accordingly. All without cloud!

drc38, 2021-06-19:
@lbbrhzn, rather than going down the elm327 route I would recommend getting one of these units if your vehicle is supported https://www.openvehicles.com/ there is a mqtt client that can be used to integrate with HA, and it has its own app to pre-heat/AC or lock the car. I use it with my Leaf.

lbbrhzn, 2021-06-19:
@drc38 thanks for the tip. Does not seem to support VW ID.4 yet, though

lbbrhzn, 2021-06-19:
@hesselonline I'm considering using this template: https://cookiecutter-homeassistant-custom-component.readthedocs.io/en/2021.1.1/guide.html Can you recommend it?

hesselonline, 2021-06-19:
I would start with the link I posted above, setting up your dev environment there and then create a scaffold component. If you get that running properly it will save you ages of time. Do you use a windows machine? If so, get docker running with the WSL2 integration, that will speed up the devcontainer considerably.

lbbrhzn, 2021-06-19:
I'm using visual studio code on MacOs, got the docker container up and running.

drc38, 2021-06-19:
> @drc38 thanks for the tip. Does not seem to support VW ID.4 yet, though

If the ID.4 is using TP 2.0 it wouldn't take much to add it. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/vwtp20/vehicle/OVMS.V3/components/vehicle/docs/VW-TP-2.0.txt

lbbrhzn, 2021-06-19:
There's some code in EVNotify to support VW ID.x: https://github.com/EVNotify/EVNotify/blob/master/app/www/components/cars/ID_3.vue

closed 2021-06-26 by lbbrhzn