lbbrhzn/ocpp #18: KeyError measurand

Rep-Hueman, 2021-06-21:
Hi I'm testing this integration on all the Alfen charge stations

have a Error in the log.
I have a Alfen Eve Single Pro-line -  NG910-60023 
OCPP 1.6

Error in log HA:

This error originated from a custom integration.

Logger: ocpp
Source: custom_components/ocpp/sensor.py:420
Integration: ocpp (documentation, issues)
First occurred: 14:43:32 (13 occurrences)
Last logged: 15:02:45

Error while handling request '<Call - unique_id=128, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1624271340, 'meterValue': [{'timestamp': '2020-08-29T19:53:37Z', 'sampledValue': [{'value': '0.010', 'context': 'Transaction.Begin', 'unit': 'kWh'}]}]}>'
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 185, in _handle_call
    response = handler(**snake_case_payload)
  File "/usr/local/lib/python3.8/site-packages/ocpp/routing.py", line 33, in inner
    return func(*args, **kwargs)
  File "/config/custom_components/ocpp/sensor.py", line 420, in on_meter_values
    self._units[sampled_value["measurand"]] = sampled_value["unit"]
KeyError: 'measurand'

HA image: entity:  
[image]

Rep-Hueman, 2021-06-21:
I have changed the configuration of the Alfen charging station!
[image]

And I made an adjustment in sensors in the same order as the charging station

```
  - platform: ocpp
    name: charger_alfen
    meter_interval: 60
    monitored_variables:
      - "Energy.Active.Import.Register"
      - "Current.Import"
      - "Current.Offered"
      - "Frequency"
      - "Power.Active.Import"
      - "Power.Factor"
      - "Temperature"
      - "Voltage"
    port: 9000
    scan_interval:
      seconds: 60
```

No error in log now!

[image]

lbbrhzn, 2021-06-21:
I guess this can be closed now?

Rep-Hueman, 2021-06-21:
Yes..

closed 2021-06-21 by Rep-Hueman