lbbrhzn/ocpp #19: Support for 3 phases

Rep-Hueman, 2021-06-21:
Are phases also available?

"Power.Active.Import" is now (L3 ?!)

### Back office (OCPP) configuration key list for Alfen chargers

MeterValuesSampledData Sampled measurand(s) to be included in every meter value. The interval can be changed by 
changing the key "MeterValueSampleInterval". A combination, up to 9, of measurands is 
supported. The measurands can be combined by separating the value by a comma ","
Value example: 'Energy.Active.Import.Register,Voltage.L1-N,Current.Import.L1'

Supported values:
Energy.Active.Import.Register, Power.Active.Import, Current.Import, Voltage, 
Temperature, Current.Offered, Frequency, Power.Factor
Measurand(s) must be combined with the phase (separated by a dot ‘.’)

Supported phases:
Voltage meter values support phase configuration:
L1-N, L2-N, L3-N, L1-L2, L2-L3, L3-L1
Value example: 'Voltage.L1-N'

Current, power and power factor meter values support phase configuration:
L1, L2, L3
Value example: 'Current.Import.L1'

When the 'SmartChargingMode' (OCPP 1.5+) is beeing used, the following measurands are 
available:

Supported values:
Current.L1 ; Current.L2 ; Current.L3

lbbrhzn, 2021-06-21:
That's currently not available, because it is not part of the OCCP1.6 standard. Should not be that difficult to add though.

Rep-Hueman, 2021-06-21:
Log Home Assistant..

2021-06-21 18:14:32 INFO (MainThread) [ocpp] CH1Beheer: send [3,"220",{}]
2021-06-21 18:15:02 INFO (MainThread) [ocpp] CH1Beheer: receive message [2,"221","Heartbeat",{}]
2021-06-21 18:15:02 INFO (MainThread) [ocpp] CH1Beheer: send [3,"221",{"currentTime":"2021-06-21T16:15:02.648607"}]
2021-06-21 18:15:28 INFO (MainThread) [ocpp] CH1Beheer: receive message [2,"222","MeterValues",{"connectorId":1,"transactionId":9223372036854766666,"meterValue":[{"timestamp":"2021-06-21T16:15:09Z","sampledValue":[{"value":"1305570.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"16.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"50.010","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"38.500","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"},{"value":"228.000","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"227.000","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"229.300","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"395.900","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"396.300","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"398.900","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"}]}]}]

lbbrhzn, 2021-06-21:
Hmm, I double checked. It is actually in the standard. So you can get a measurand multiple times, for different phases. The integration currently only has one sensor entity to represent multiple phases, so it will show what has been received last.

This is something that should be fixed.

lbbrhzn, 2021-06-21:
I guess location should be added too

drc38, 2021-06-22:
I suggest waiting till after the recoding of the integration. If present these could be added as attributes of the sensor.

lbbrhzn, 2021-07-06:
I think we would should to create different metrics for the different phases. The additional phase related metrics can then be added as attribute to the sensor.

drc38, 2021-07-19:
@Rep-Hueman , you can test using my repo https://github.com/drc38/ocpp and report back

Rep-Hueman, 2021-07-21:
Just got back from vacation I'll test it tonight or tomorrow..

lbbrhzn, 2021-07-21:
Great! @Rep-Hueman, what charger model are you using? Just to add it to the list of supported chargers...

Rep-Hueman, 2021-07-21:
### I can confirm that it works perfectly!

I am testing here with the following chargers:

**Alfen - Eve Single S-line
Alfen - Eve Single Pro-line
https://alfen.com/en/ev-charge-points/alfen-product-range**
These work perfectly with this OCCP integration

### **before starting to charge the car:**
.
[image]
[image]
[image]
[image]
.
### **While charging the car:**
.
[image]
[image]
[image]

.
Thanks for the perfect adjustment

lbbrhzn, 2021-07-21:
👍 thanks for the feedback!

lbbrhzn, 2021-07-21:
I will release v0.3.2 shortly..

lbbrhzn, 2021-07-21:
Solved and released.

closed 2021-07-21 by lbbrhzn

bobf41, 2021-07-31:
What did you guys set in ACE Service installer to get the data in HA?
I tried OCPP 1.6 with ws://homeassistant.local:9000 which resolves fine but get no data in HA.

TravisWilder, 2021-09-28:
Will this also work for the Alfen Eve Double Pro-line?
Are charges with two outlets supported?
Or shall I just give it a try?

drc38, 2021-09-28:
It may report some odd results if both outlets are in use, but both outlets should still charge. You're  welcome to have a go at amending the code to support more than 1 outlet 😉

TravisWilder, 2021-09-29:
thanks - I opened a new FR with some details and some documentation from Alfen

dailyenergy, 2022-02-05:
Do you have to pick a certain Backoffice preset in ACE in order for the Alfen to work with HA?

amurray81, 2023-04-18:
Hi, I have this working to a point. I can grab the Diagnostic information into HA but the sensors remain as 'Unknown'

Any ideas as to why this would happen? Thanks