lbbrhzn/ocpp #30: Titles for the measurand selection are not showing?

drc38, 2021-06-26:
@lbbrhzn, any ideas on why I'm not seeing the labels when adding the new integration?
[image]

lbbrhzn, 2021-06-26:
Hmm, strange. I renamed the translation file to i-default.json, which should be loaded when there is no language specific translation.

lbbrhzn, 2021-06-26:
i've tried to reproduce the issue, but failed. I added nl.json, this is user language is set to 'Nederlands'. Selecting a language that does not have a .json file will default to i-default.json.

drc38, 2021-06-27:
A minimum HA version may need to be added. I was running a Docker version dated 2021/04/19, once I updated it to 2021.6.6 then it worked perfectly!

lbbrhzn, 2021-06-27:
Ok, I guess this can be done in the manifest.json?

closed 2021-06-27 by lbbrhzn