lbbrhzn/ocpp #34: Setup entry on discovery

drc38, 2021-06-27:
@lbbrhzn , it would be useful to split the integration setup. The first `user` form essentially sets up the central system server and can be set when adding the integration, then when a charger successfully connects to the central server the default sensors can be automatically added by calling `get_configuration("MeterValuesSampledData")`. If the user wanted to set extra options for the charger device the `measurands` form could be used with extra items  relating to the charger also added such as `MeterValueSampleInterval`, `WebSocketPingInterval`, etc. These extra options could be auto-generated from the charger by looking at whether `read-only` is false. Is that possible?

lbbrhzn, 2021-06-27:
Yes, that would be a nice approach. There are some drawbacks though.

Adding the sensors upon connect is a bit complicated. I've been reading about device discovery and it seems a bit daunting. It would need a callback function or event subscription. Currently CentralSystem and ChargePoint know very little about home assistant, they just use websockets and ocpp. As hesselonline pointed out we will probably need to make the websocket server external (as an addition?) to get this integration merged into core, I would like to decouple them even more. Not sure what the best way is, a full blown Jason rest api seems overkill and would require polling instead of push.

For now can take a simple approach: we can let the ChargePoint find out which measurands are supported, and use an OptionFlow to provide a means to select which ones are visible in home assistant.

drc38, 2021-06-27:
The simple approach sounds good, now that the device details have been added #35 the integration is looking great.

lbbrhzn, 2021-06-27:
Well, unfortunately the device details (vendor, model, serial) are not yet showing up in my case. Probably because the entities are created before the information was retrieved. Any clue how to solve this?

lbbrhzn, 2021-06-27:
Oh, I added switch, but and it shows up as an entity, but it does not do anything useful yet. I would like it to enable / disable charging. So, trigger a transaction when enabled, and allow new transactions, stop any ongoing transaction when disabled, and disallow new transactions.

drc38, 2021-06-27:
> Well, unfortunately the device details (vendor, model, serial) are not yet showing up in my case. Probably because the entities are created before the information was retrieved. Any clue how to solve this?

When info becomes available iterate devices and use `async_update_device` perhaps?

lbbrhzn, 2021-06-27:
> When info becomes available iterate devices and use `async_update_device` perhaps?

Looking at the [documentation](https://developers.home-assistant.io/docs/device_registry_index/) of the DeviceRegistry,  `async_get_or_create()` might just be what we need. It will call `async_update_device` when the device is found, and create a new device if not.

I'm a bit hesitant to call this directly from the ChargePoint class though, for reasons provided above. Perhaps use an asyncio.Event()?

drc38, 2021-06-27:
I guess it comes down to which approach is easier, move to setting up Charger and platforms when it registers with Central sever  (then information is available for setup) or setup as we do now then look to update on actual registration.

lbbrhzn, 2021-06-27:
Well, the detailed device info only becomes available through a boot notification, and it is not guaranteed when this will occur, especially because not all chargers support a remote trigger.

I think a charger should be represented by a home assistant device, which is created when there is a boot notification. The device name should be the cp_id, and the unique_id should be derived from the manufacturer, model and serial number. If charger is offline then the device should become unavailable.

In a ConfigFlow you specify the port, host and which cp_id and cp_pw is accepted. It is convenient to be able to inspect / modify these and other settings, so we can also put them in an optionflow and restart the centralsystem upon change.

How should we support multiple chargers? If we use different websocket port for each charger this is easy, you can just add multiple instances of the OCPP integration, so there is a CentralSystem for every ChargePoint. This also makes it easy to have separate settings for each Charger, and even different OCPP versions.

Multiple devices per OCPP instance is more difficult. Not sure whether an OptionFlow can be initiated from a device instance.

drc38, 2021-06-27:
I like your idea of using a separate port per charger, that certainly makes the whole integration easier. 

Why not simply use `cp_id` for the `unique_id`? Saves having to worry whether other variables are reported and as it is used in websocket initiation is unique I would expect.

lbbrhzn, 2021-06-28:
Perhaps I am being too strict. For my charger, I can modify the `cp_id` freely through it's settings, so I see it as a username, and this is explicitly mentioned as an unacceptable source for a unique ID [here](https://developers.home-assistant.io/docs/entity_registry_index/). However, I agree that it makes our life a lot simpler, so let's just go for it.

drc38, 2021-06-28:
I think it will do, also I have switches working for basic functions like reset, availability, charge (start/stop) and will send a draft PR with details later tonight.

lbbrhzn, 2021-07-02:
I have come up with some changes that will allow automatic discovery of a new charge point, creation of the device and the related sensor/switch entities.

I would like to introduce some new classes:
- Metric. It has name, value and unit properties.
- DeviceManager. Response for creating new devices and entities.

- A ChargePoint creates collection of Metrics on instantiation.
- The DeviceManager creates a new device for the chargepoint, and the entities for the metrics.
- The entities subscribe to the metrics, using  a callback function.
- When the metrics are updated by the ChargePoint, subscribers will be notified through the callback
- In the callback function, write_ha_state() is called, updating the hass gui.
- This changes the data retrieval for the sensor from polling to push

I've got most of this working already, will add a PR this weekend.

drc38, 2021-07-02:
Awesome,  look forward to talking a look at the PR when it's ready

drc38, 2021-07-05:
Hi @lbbrhzn 
A couple more features that I think would be worthwhile adding:

- a custom service to call `set_charge_rate` so that HA automations/scripting can be used for limiting the charger output
- a custom service to call `clear_profile` for the same reason above
- a custom service to set the update server and initiate a firmware update
- config flow to make the `unlock` switch optional as it only applies to a charger with sockets (as opposed to fixed cable)
- config flow to set the logging level (if that is possible) as once setup and running debug messages are not really required
- extra attributes for metrics that are capable of reporting 3-phase 

Are there any others?

lbbrhzn, 2021-07-05:
That's quite a list! Some remarks...

Wrt set_charge_rate:  Would it be enough to just have this service or in addition / instead offer a service for setting a complete profile?

Wrt to config flow: Actually, I would like to reduce the settings in the config flow to the bare minimum (host, port, central system id, charge point id), and move as much settings as possible to an option flow. This flow is initiated by a button on the integration, and allows you to change settings without the need to re-add the integration.

drc38, 2021-07-06:
> Wrt set_charge_rate: Would it be enough to just have this service or in addition / instead offer a service for setting a complete profile?

I think it is preferable to keep it simple, the ocpp standard allows for a huge range of complex options, but most of these are already available via the HA automation scheduler/timers and simply changing the charging rate. By not trying to replicate this complexity by setting a charge profile the chance of issues (eg incorrect format, unsupported rate units etc) is reduced.

> Wrt to config flow: Actually, I would like to reduce the settings in the config flow to the bare minimum (host, port, central system id, charge point id), and move as much settings as possible to an [...]

Sorry, yes I meant option flow.

closed 2022-02-26 by github-actions[bot]