lbbrhzn/ocpp #42: First time setup - HA throws error on sensor configuration

bpenglase, 2021-07-05:
## Version of the custom_component
97b83fe

## Configuration

```yaml
sensor:
  - platform: ocpp
    name: charger_1
    meter_interval: 60
    monitored_variables:
      - "Current.Export"
      - "Current.Import"
      - "Current.Offered"
      - "Energy.Active.Export.Register"
      - "Energy.Active.Import.Register"
      - "Energy.Reactive.Export.Register"
      - "Energy.Reactive.Import.Register"
      - "Energy.Active.Export.Interval"
      - "Energy.Active.Import.Interval"
      - "Energy.Reactive.Export.Interval"
      - "Energy.Reactive.Import.Interval"
      - "Frequency"
      - "Power.Active.Export"
      - "Power.Active.Import"
      - "Power.Factor"
      - "Power.Offered"
      - "Power.Reactive.Export"
      - "Power.Reactive.Import"
      - "RPM"
      - "SoC"
      - "Temperature"
      - "Voltage"
    port: 9000
    scan_interval:
      seconds: 60

```

## Describe the bug

I'm trying to set this up for the first time (just got a charger that supports ocpp) and I setup HACS and added the repo. When I add the sensor configuration it passes config check but nothing shows up in entities, and the below shows up in Configuration -> Logs.

HAOS on a Pi4
Core: core-2021.6.6
Supervisor: 2021.06.6
Host: Home Assistant OS 6.1

## Debug log

```text
 Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:165
Integration: Sensor (documentation, issues)
First occurred: 1:03:27 PM (1 occurrences)
Last logged: 1:03:27 PM
The ocpp platform for the sensor integration does not support platform setup. Please remove it from your config.

```

lbbrhzn, 2021-07-05:
Hi Brandon,

The cpp integration no longer requires an entry in configuration.yaml, so you can remove it. 
Please see the updated readme.md for the steps required to add the integration using the gui.

bpenglase, 2021-07-05:
> Hi Brandon,
> 
> The cpp integration no longer requires an entry in configuration.yaml, so you can remove it.
> Please see the updated readme.md for the steps required to add the integration using the gui.

Hello!
Go figure, looks like it was updated while I was working on it. I'll keep working at it and let you know. Thanks!

bpenglase, 2021-07-05:
At this point, any issue I would have would be different than this topic, so closing it out.

closed 2021-07-05 by bpenglase

lbbrhzn, 2021-07-05:
ok!