lbbrhzn/ocpp #45: charger.Session.Time negative value

Rep-Hueman, 2021-07-06:
I get a negative value for charger.Session.Time

## Version of the custom_component

Version 0.2 (code 6-7-2021)

## Configuration

central
Open Charge Point Protocol (OCPP)
3 devices and 26 entities

sensors
``
"Energy.Active.Import.Register":` "Active energy imported from the grid",
"Current.Import": "Instantaneous current flow to EV",
 "Current.Offered": "Maximum current offered to EV",
"Frequency": "Powerline frequency",
"Power.Active.Import": "Instantaneous active power imported by EV",
"Power.Factor": "Instantaneous power factor of total energy flow",
 "Temperature": "Temperature reading inside Charge Point",
 "Voltage": "Instantaneous AC RMS supply voltage"
``
## Describe the bug

I get a negative value for charger.Session.Time

charger.Session.Time
10 minutes ago
-153.722.867.253.819.900 min

## Debug log

```
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"49278991-dc53-4459-964d-e115f455d078","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"49278991-dc53-4459-964d-e115f455d078",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,Reservation,LocalAuthListManagement,RemoteTrigger,SmartCharging"}]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"3095f400-15e5-4e4e-8f46-cc87656e121b","TriggerMessage",{"requestedMessage":"BootNotification"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"3095f400-15e5-4e4e-8f46-cc87656e121b",{"status":"Accepted"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"c06070c5-f8a5-4d5b-bbdc-e07ab3b00947","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"c06070c5-f8a5-4d5b-bbdc-e07ab3b00947",{"status":"Accepted"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"713f3172-e854-49f6-9dca-a4ebb2028282","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"713f3172-e854-49f6-9dca-a4ebb2028282",{"status":"Accepted"}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"ecfd42a4-07fe-48f8-a809-5cfd19ef30f0","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"ecfd42a4-07fe-48f8-a809-5cfd19ef30f0",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"30"}]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"f9b9a86d-fb08-4e53-ad50-87ed03fd831f","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"f9b9a86d-fb08-4e53-ad50-87ed03fd831f",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"ac159344-d472-474c-bce4-4616a998a841","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"ac159344-d472-474c-bce4-4616a998a841",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Temperature,Voltage"}]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"3009a3cd-59cf-4fbd-a7cd-f02a2a5fe43c","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"3009a3cd-59cf-4fbd-a7cd-f02a2a5fe43c",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: send [2,"26bf53aa-ee21-4ba2-b25d-871becc0fc93","GetConfiguration",{"key":["NumberOfConnectors"]}]
2021-07-06 16:10:51 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [3,"26bf53aa-ee21-4ba2-b25d-871becc0fc93",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-07-06 16:10:56 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [2,"8","BootNotification",{"chargePointVendor":"Alfen BV","chargePointModel":"NG910-60023","chargePointSerialNumber":"ace0025893","chargeBoxSerialNumber":"Reppel Beheer","firmwareVersion":"4.15.0-3910"}]
2021-07-06 16:10:56 INFO (MainThread) [ocpp] Alfen-Beheer: send [3,"8",{"currentTime":"2021-07-06T14:10:56.330803","interval":30,"status":"Accepted"}]
2021-07-06 16:11:36 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [2,"9","MeterValues",{"connectorId":1,"transactionId":9223372036854775807,"meterValue":[{"timestamp":"2021-07-06T14:08:31Z","sampledValue":[{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"1318920.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"50.050","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"34.875","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"},{"value":"227.200","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"225.300","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"228.600","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"394.900","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"393.700","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"396.700","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"}]}]}]
2021-07-06 16:11:36 INFO (MainThread) [ocpp] Alfen-Beheer: send [3,"9",{}]
2021-07-06 16:12:08 INFO (MainThread) [ocpp] Alfen-Beheer: receive message [2,"10","MeterValues",{"connectorId":1,"transactionId":9223372036854775807,"meterValue":[{"timestamp":"2021-07-06T14:09:32Z","sampledValue":[{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"1318920.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"50.070","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"34.875","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}, [... cut ...] surand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"396.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"394.700","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"397.400","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"}]}]}]
2021-07-06 16:13:15 INFO (MainThread) [ocpp] Alfen-Beheer: send [3,"14",{}]
```

lbbrhzn, 2021-07-06:
@drc38, can you take a look? You've implement the session related metrics...

drc38, 2021-07-06:
@Rep-Hueman, the transactionId reported appears to be incorrect eg 9223372036854775807, it should be set by the ocpp integration to epoch time eg 1625606905. Suggest a clean install of ocpp integration in HACS and then starting the charge session after the unit has connected to HA.

Rep-Hueman, 2021-07-06:
I have a clean install and it is stil the same here..

I get this error when I connect the car to the charger (first test)

```
2021-07-07 01:06:01 INFO (MainThread) [ocpp] RepHome-1: send [3,"822",{}]
2021-07-07 01:06:02 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"823","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2021-07-06T23:06:01Z"}]
2021-07-07 01:06:02 INFO (MainThread) [ocpp] RepHome-1: send [3,"823",{}]
2021-07-07 01:06:03 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"824","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-07-06T23:06:01Z"}]
2021-07-07 01:06:03 INFO (MainThread) [ocpp] RepHome-1: send [3,"824",{}]
2021-07-07 01:06:04 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"825","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2021-07-06T23:06:03Z"}]
2021-07-07 01:06:04 INFO (MainThread) [ocpp] RepHome-1: send [3,"825",{}]
2021-07-07 01:06:34 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"826","Heartbeat",{}]
2021-07-07 01:06:34 INFO (MainThread) [ocpp] RepHome-1: send [3,"826",{"currentTime":"2021-07-06T23:06:34.847897"}]
2021-07-07 01:06:40 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"827","MeterValues",{"connectorId":1,"transactionId":9223372036854775807,"meterValue":[{"timestamp":"2021-07-06T23:06:34Z","sampledValue":[{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"16.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"1318920.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"50.040","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"33.500","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"},{"value":"226.700","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"226.900","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"227.600","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"394.800","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"395.000","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"396.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"}]}]}]
2021-07-07 01:06:40 INFO (MainThread) [ocpp] RepHome-1: send [3,"827",{}]
2021-07-07 01:06:42 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"828","StartTransaction",{"timestamp":"2021-07-06T23:06:38Z","connectorId":1,"meterStart":1318920,"idTag":"REP00088"}]
2021-07-07 01:06:42 INFO (MainThread) [ocpp] RepHome-1: send [3,"828",{"transactionId":1625612802,"idTagInfo":{"status":"Accepted"}}]
2021-07-07 01:07:08 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"829","MeterValues",{"connectorId":1,"transactionId":1625612802,"meterValue":[{"timestamp":"2021-07-06T23:06:38Z","sampledValue":[{"value":"1318.920","context":"Transaction.Begin","unit":"kWh"}]}]}]
2021-07-07 01:07:08 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=829, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1625612802, 'meterValue': [{'timestamp': '2021-07-06T23:06:38Z', 'sampledValue': [{'value': '1318.920', 'context': 'Transaction.Begin', 'unit': 'kWh'}]}]}>'
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 185, in _handle_call
    response = handler(**snake_case_payload)
  File "/usr/local/lib/python3.8/site-packages/ocpp/routing.py", line 33, in inner
    return func(*args, **kwargs)
  File "/config/custom_components/ocpp/api.py", line 473, in on_meter_values
    self._units[sampled_value["measurand"]] = sampled_value["unit"]
KeyError: 'measurand'
2021-07-07 01:07:08 INFO (MainThread) [ocpp] RepHome-1: send [4,"829","InternalError","An unexpected error occurred.",{}]
2021-07-07 01:08:47 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"830","MeterValues",{"connectorId":1,"transactionId":1625612802,"meterValue":[{"timestamp":"2021-07-06T23:06:38Z","sampledValue":[{"value":"1318.920","context":"Transaction.Begin","unit":"kWh"}]}]}]
[... cut ...]
  File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 185, in _handle_call
    response = handler(**snake_case_payload)
  File "/usr/local/lib/python3.8/site-packages/ocpp/routing.py", line 33, in inner
    return func(*args, **kwargs)
  File "/config/custom_components/ocpp/api.py", line 473, in on_meter_values
    self._units[sampled_value["measurand"]] = sampled_value["unit"]
KeyError: 'measurand'
2021-07-07 01:08:47 INFO (MainThread) [ocpp] RepHome-1: send [4,"830","InternalError","An unexpected error occurred.",{}]

```

drc38, 2021-07-06:
Not sure about why the `measrand` error is occuring yet, but it is now obtaining the correct transactionId eg `"transactionId":1625612802`

drc38, 2021-07-07:
The `measurand` key is optional so the `on_meter_values` will need to be reworked so that the setting of units only occurs if the `measurand` and `unit` keys are present in the message.

PR #58 (2021-07-07): ocpp: revise approach to measurand metrics reading
Should resolve #45
closed 2021-07-07 by lbbrhzn