lbbrhzn/ocpp #47: on_stop_transaction() missing 1 required positional argument: 'reason'

jeremycoppey, 2021-07-06:
## Version of the custom_component
Version of 06/07/2021 (where can I find it in the const.py file?)

## Configuration

```yaml
1 central
1 charepoint

Sensors: all that could be checked for m the list
```

## Describe the bug
on_stop_transaction() missing 1 required positional argument: 'reason'

## Debug log

```text

2021-07-06 18:44:50 INFO (MainThread) [ocpp] 00:80:F4:43:0B:34: receive message [2,"2107","StopTransaction",{"meterStop":322,"timestamp":"2021-05-02T06:52:08Z","transactionId":1625568827}]
2021-07-06 18:44:50 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=2107, action=StopTransaction, payload={'meterStop': 322, 'timestamp': '2021-05-02T06:52:08Z', 'transactionId': 1625568827}>'
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/ocpp/charge_point.py", line 185, in _handle_call
response = handler(**snake_case_payload)
File "/usr/local/lib/python3.8/site-packages/ocpp/routing.py", line 33, in inner
return func(*args, **kwargs)
TypeError: on_stop_transaction() missing 1 required positional argument: 'reason'
2021-07-06 18:44:50 INFO (MainThread) [ocpp] 00:80:F4:43:0B:34: send [4,"2107","InternalError","An unexpected error occurred.",{}]

```

lbbrhzn, 2021-07-06:
Looks like this is a codign error. Reason is optional, but timestamp is required. See
https://github.com/mobilityhouse/ocpp/blob/a6f57307d2a8a1c8f85f802742b5855ccf48756d/ocpp/v16/call.py#L273

closed 2021-07-06 by lbbrhzn

jeremycoppey, 2021-07-08:
tested: Approved