lbbrhzn/ocpp #52: use constants instead of string literals for key names and measurands

lbbrhzn, 2021-07-06:
Currently we use string literals for measurands and configuration keys.
It would be much better to use constants, defined in the ocpp package.

The ocpp.v16.enums already has a Measurand class, but there is no ConfigurationKey class yet.

I've created an issue and a PR for this on the ocpp pakcage:
https://github.com/mobilityhouse/ocpp/issues/229

lbbrhzn, 2021-07-06:
@drc38, if we adopt this approach the sensor names will change as well.  That breaks backwards compatiblity, but I don't have an issue with that at this stage of the project. Do you agree?

drc38, 2021-07-06:
Can you explain why the sensor names change, the strings for the measurands still remain the same don't they?

lbbrhzn, 2021-07-06:
For the measurands that is true, but for some sensors that are currently defined may change.

For instance:

| sensor name | ocpp package name |  official OCPP name |
|-|-|-|
| Fw.Version | firmware_version | firmwareVersion | 
| Serial         | charge_point_serial_number | chargePointSerialNumber |
|Stop.Reason | StopTransaction.reason | StopTransaction.reason |
| Connectors | number_of_connectors | NumberOfConnectors |

drc38, 2021-07-06:
If they are not going to merged upstream I don't think we need to adopt the official names necessarily. Originally, my thinking was to use the same convention as the measurands eg "Current.Export" and removed elements that I considered superfluous to interpreting the metric. `Fw` could be expanded to `Firmware` on reflection. I'm comfortable either way though... as between the HA, python, json and OCPP conventions it is a bit of a mishmash.

closed 2021-07-08 by lbbrhzn