lbbrhzn/ocpp #55: Support for secure authentication

lbbrhzn, 2021-07-06:
We currently accept connections from any charger, and don't use any encryption. This may be ok for a private network, but is  not acceptable when you connect to home assistant through a public network.

Hence, we should support password based authentication, and TLS for encryption.

lbbrhzn, 2021-07-06:
based on the websockets documentation, it does not look that difficult to support TLS:
https://websockets.readthedocs.io/en/stable/intro.html

```
ssl_context = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
localhost_pem = pathlib.Path(__file__).with_name("localhost.pem")
ssl_context.load_cert_chain(localhost_pem)

start_server = websockets.serve(
    hello, "localhost", 8765, ssl=ssl_context
)
```
It would be nice if we can reuse a .pem file provided by home assistant. Not sure whether that is possible though!

lbbrhzn, 2021-07-06:
The http component allows specification for an SSL key, see
https://github.com/home-assistant/core/blob/16f1647ad0aeb0c4295d741d6b9dbb3f60e49909/homeassistant/components/http/__init__.py#L80

Maybe we can re-use that?

drc38, 2021-07-06:
Personally, I think we should keep the scope to personal applications/networks. There are plenty of commercial solutions for use through a public network... that's not to say it should not support encryption its just lower down the priority order from my perspective.

lbbrhzn, 2021-07-07:
I would agree if everyone would run home assistant on a local server. But there is also a user base that runs home assistant in the cloud... so we better support a secure connection to that.

dreimer1986, 2021-07-08:
The problem with the commercial solutions is that you hardly can add these to home assistant at all. That's why I follow this addon. Working for me or not right now, this is the favored solution for me. As mine is a special case in many ways as it seems, I am all for any OCPP 1.6 supported corner cases until my box finally does a bit more than .... nothing. If it's quite easy to add, why not adding it? It's no safety risk, too. Quite the opposite.

lbbrhzn, 2021-07-17:
[Here](https://www.floriangaechter.com/blog/valid-https-certificate-for-local-home-assistant-installation/) is a simple tutorial for generating an https key and certificate for a local network.

Should be fairly easy to check if this works for an ocpp websocket as well.

dreimer1986, 2021-07-18:
So the support is already implemented? We can hand it a certificate and I can forward to my DYNDNS?

lbbrhzn, 2021-07-18:
No, it still needs some work.

dreimer1986, 2021-07-18:
Ah sorry. Misunderstood you then.

dreimer1986, 2021-07-27:
My 2 cents to that one. I use the reverse proxy way: https://dummylabs.com/posts/2021-03-13-the-easiest-way-to-https-home-assistant/ and with it's help many addons have their own subdomain which I feed to it with some CNAME help on my domain I already own. So I have nas.xyz.bla, ha.xyz.bla, tautulli.xyz.bla and many more. I have no clue if a reverse proxy is able to pass a websocket at all, but that's no problem too. I can just forward to my dyndns domain with my router, too. So to support all these gimmicks as most official addons do, I recommend to just allow to set a path or name to the cert files.

An example from my MQTT addon:
certfile: cert.pem
keyfile: privkey.pem

These are the default certs used by Home Assistant and can be generated with many different addons by using Let's encrypt or other free variants. In my case right now this cert is completely invalid as the Reverse Proxy is doing the real job. But by feeding it a valid cert to the ddns domain (I already own that one for some other tinkering and it's being auto refreshed either) it will become valid, too.
Here how MQTT is doing the job: https://github.com/home-assistant/addons/blob/master/mosquitto/rootfs/etc/cont-init.d/mosquitto.sh

So in short. Just copy the cert searching from some other official addon. Of course we can do almost everything automated, but as it happened with many official integrations, as soon as they are ported to the WebUI install process, many less commonly used settings are not useable anymore. Some config parameters dont hurt ^^

lbbrhzn, 2021-07-27:
Ok, that's useful feedback!

I now think we should support configuration of the key and certificate through configuration.yaml, since you need to upload these files anyway that should not be an issue.

drc38, 2022-04-30:
see #506