lbbrhzn/ocpp #78: Split Features list into separate items

lbbrhzn, 2021-07-16:
**Is your feature request related to a problem? Please describe.**
The charger.Features sensor shows the list of supported features as a list of comma separated strings. This does not display nicely on small screens (phone) and is cumbersome to access in scripts. 

**Describe the solution you'd like**
It would be better to create separate (binary) sensor metrics for each possible feature.

**Describe alternatives you've considered**
None

**Additional context**
[image]

drc38, 2021-07-17:
You might be able to follow this example to set them up as flags https://community.home-assistant.io/t/sensors-with-flags-in-automation/311415

lbbrhzn, 2021-07-17:
I think the example uses state attributes. That would require a custom sensor that adds these attributes. That would work, but what would be the main sensor value to show, perhaps Status?

drc38, 2021-07-17:
Main sensor value could be `Core` which all chargers must support?

lbbrhzn, 2021-07-17:
Ok, I'll try when I get back from my holiday. There's a limit to what you can do from your mobile phone! 😎

PR #122 (2021-08-23): Change supported features to using flags
Fixes #78 using same approach as HA supported_features attribute
closed 2021-08-23 by lbbrhzn