lbbrhzn/ocpp #80: Add to HACS default integrations

lbbrhzn, 2021-07-16:
## Version of the custom_component
v0.3.0

## Configuration

Not applicable

## Describe the bug

Ocpp currently needs to be added as an additional repository to hacs. By adding ocpp to the HACS default integration list, this step is no longer needed

## Debug log

Not applicable.

lbbrhzn, 2021-07-16:
I've opened a [pull request](https://github.com/hacs/default/pull/1027) on hacs/default. If we pass all checks, and the PR is approved, we can remove the addition of the repository from our installation instructions.

lbbrhzn, 2021-07-16:
Note that restart of home assistant should not be needed after adding the integration, since we use a configflow for our setup. See [here](https://hacs.xyz/docs/categories/integrations/)

lbbrhzn, 2021-08-15:
pull request for adding to HACS has finally been accepted!

drc38, 2021-08-20:
Worth bumping the release?

lbbrhzn, 2021-08-22:
well, it does not need any code changes... but I did update the readme.

drc38, 2021-08-22:
HACS defaults to 0.3.4 (rather than main) so now 25+ days old

lbbrhzn, 2021-08-23:
done

closed 2021-08-23 by lbbrhzn