lbbrhzn/ocpp #84: Support for a charger-card

lbbrhzn, 2021-07-17:
**Is your feature request related to a problem? Please describe.**
Our ocpp integration represents the ChargePoint as a set of sensors. Some users may find it more user friendly to use a single graphical representation of the charger that shows the status and access to services, like  lock/unlock, reset, fw upgrade etc.

**Describe the solution you'd like**
The is a HACS project for a custom
[charger-card](https://github.com/tmjo/charger-card) 
We should see what would be required to facilitate users to combine our integration with this.

**Describe alternatives you've considered**
Alternatively, we can create our own card that is customised for ocpp. Not sure if we can integrate the .js file of the card with the custom_component in a single hacs integration, though.

**Additional context**
I’ve opened an [issue](https://github.com/tmjo/charger-card/issues/10) at  the charge-card project to discuss how to facilitate  integration.

jeremycoppey, 2021-07-19:
Hey
I use this card already in my hass setup.
looks like this:
[image]

My yaml config for the card 
`
type: custom:charger-card
entity: sensor.wallbox_status
image: default
customCardTheme: theme_default
chargerImage: White
`

Note that I called my charger sensor wallbox

lbbrhzn, 2021-07-19:
Great! Did you also connect it to ocpp services?

jeremycoppey, 2021-07-19:
No I didn't look into that yet.
Was already happy I had some graphical way to show the charger on my homepage 😄

jeremycoppey, 2021-07-19:
Thinking it would be better to recreate the charger card that is already there and make it OCPP complaint, 
I went through the repo and found some comment by the dev: https://github.com/tmjo/charger-card/issues/6#issuecomment-853045278 
Here he states he created it specific for a specific charger: ease charger.

So my 2 cent, as OCPP is a standard we (I can code JS) should maybe try to start from scratch and make it OCPP complainant.

jeremycoppey, 2021-07-19:
And I just noticed that Easee charger user can still use our card via the ocpp service that Easee provides through their API:
https://easee.com/uk/wp-content/uploads/sites/3/2020/03/V1_01-UK_Easee_Charge_Productsheet.pdf
page 6
[image]

lbbrhzn, 2021-07-19:
Here's another card that could be used for inspiration:

https://github.com/JonahKr/power-distribution-card

This has animations and a versatile editor.

jeremycoppey, 2021-07-22:
Quick question, do you add it into this integration or do we need to create a new repo with only the card?

lbbrhzn, 2021-07-22:
I would prefer to keep things together, but I am not sure that this is supported by HACS

nagydavid, 2021-11-28:
Hi @jeremycoppey and @lbbrhzn, 

I am interested about the status of the card. If you need help I could look into it during the Christmas holiday :)

zbuh, 2022-02-09:
The Alarmo component also includes a lovelace card. This could be a similar approach

closed 2022-04-18 by github-actions[bot]