lbbrhzn/ocpp #91: Setting max current using sliders

lbbrhzn, 2021-07-19:
**Is your feature request related to a problem? Please describe.**
We provide services through switches, and service calls. Some simple services like setting the max charging current could be represented in the user interface as simple controls.

**Describe the solution you'd like**
Use a slider for setting max charging current.

**Describe alternatives you've considered**
None

**Additional context**
See https://github.com/cathiele/homeassistant-goecharger/blob/master/doc/ha_entity_view.png

lbbrhzn, 2021-07-21:
We should be able to use the `input_number` integration for this. Not sure how to set this up without user interaction though, maybe similar to the switch and sensor platforms?

jeremycoppey, 2021-07-22:
Guys I found another hass integration using the slider:
https://github.com/cathiele/homeassistant-goecharger

drc38, 2021-07-23:
@jeremycoppey, do you want to add to the readme initially? Here is my setup:
Configuration.yaml
`input_number:
  charge_limit_slider:
    name: Charge limit (A)
    min: 0
    max: 32
    step: 1`

Automations.yaml
`- id: '1627073230475'
  alias: Max Charge Limit
  description: ''
  trigger:
  - platform: state
    entity_id: input_number.charge_limit_slider
  condition: []
  action:
  - service: ocpp.set_charge_rate
    data:
      limit_amps: '{{ states(''input_number.charge_limit_slider'')|int }}'
  mode: single`

lbbrhzn, 2021-07-23:
Yes, that's a nice starting point!

I would like the ocpp integration to create the slider entities by itself, without the need to update yaml files.

drc38, 2021-07-23:
> Yes, that's a nice starting point!
> 
> I would like the ocpp integration to create the slider entities by itself, without the need to update yaml files.

That would be nice, may be a little tricky as you will need to set it up after everything else so that the charger's support for either power or current can be queried.

lbbrhzn, 2021-07-23:
Can we convert to whatever is supported in the service handler?

drc38, 2021-07-24:
> Can we convert to whatever is supported in the service handler?

That would work, for AC chargers I expect most set the limit based on current.

lbbrhzn, 2021-08-01:
ok, i found out how to setup a slider control in a similar way to a switch. You just create a file `number.py` with a `async_setup_entry` function, which is initiate by forwarding from the async_setup_entry in __init__.py. The slider class can be derived from homeassistant.components.input_number.InputNumber, it just needs the device info added to make sure the entity is nicely grouped in the device overview.

@drc38:
- I want to do an async call to ChargePoint.set_charge_rate on change.  Will it work if I just use the current limit?
- Should we keep the set_charge_rate service, or can the slider replace it completely?
- If we keep the service, we should make sure the slider value is always reflecting the last charge_rate set.

drc38, 2021-08-01:
Sounds great, in response to your questions:
- the async call will work with only a current limit specified 
- the handle service can go when the slider is implemented, the custom services are not ideal
- the slider should reflect the last charge rate set, if not accepted then it should revert to value before sliding it

drc38, 2021-08-24:
@lbbrhzn thought I'd give it a try but I am stuck on how the entry forwarding needs to work. If you could take a look https://github.com/drc38/ocpp/tree/slider any assistance appreciated!

lbbrhzn, 2021-08-25:
@drc38, I suspect you need to rename input_numeric.py to number.py to get it working.

Here's my take at it:
https://github.com/lbbrhzn/ocpp/tree/add-number-platform

The number shows up in the gui, but is not calling any service.

lbbrhzn, 2021-08-25:
[image]