lbbrhzn/ocpp #92: Charger offline

drc38, 2021-07-20:
## Describe the bug

When a charger disconnects from the websocket eg powered down the sensors/switches should be set to offline. At the moment they remain online with the last known value.  This results in a websocket error when operating a switch for example.

lbbrhzn, 2021-07-20:
There are twoaspects to this:

1. Detection of offline status: This can be done by setting a timeout on the reception of a heartbeat notification, timeout on receiving a confirmation to a request, or the websocket disconnecting.

2. Notification of offline status: When the charger goes online (or back online) all entities should be informed to change their status. I guess this is true for services as well. We could use a metric for this, with subscriber notification on status changes.

drc38, 2021-07-21:
In terms of offline I'm detection pretty sure all the timeouts mentioned above are already handled by the MobilityHouse ocpp code and ultimately result in a websocket disconnect. So the trigger can be the websocket disconnect.

For implementing the offline status how about simply updating the device registry to disable the charger? HA should take care of the sensors/switches/services from there?

lbbrhzn, 2021-07-21:
That could work, would need to try.

drc38, 2021-07-26:
Setting the device to offline/disabled was not workable, hence went back to using `available` state for sensors/switches based on charger `status`

closed 2021-07-28 by lbbrhzn