lbbrhzn/ocpp #95: UnboundLocalError: local variable 'sum' referenced before assignment

Rep-Hueman, 2021-07-21:
## Version of the custom_component
 OCPP version  0.3.2  21-07-2021

## Configuration

```yaml

```

## Describe the bug

UnboundLocalError: local variable 'sum' referenced before assignment.

No problems with the test version for 3 phase - version 0.3.1 i test today.
## Debug log

```text

This error originated from a custom integration.

Logger: ocpp
Source: custom_components/ocpp/api.py:719
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 20:03:15 (1 occurrences)
Last logged: 20:03:15

Error while handling request '<Call - unique_id=18, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1626878818, 'meterValue': [{'timestamp': '2021-07-21T17:42:42Z', 'sampledValue': [{'value': '1336750.000', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'location': 'Outlet', 'unit': 'Wh'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L1'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L2'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L3'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Factor', 'location': 'Outlet'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L1'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L2'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L3'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'location': 'Outlet', 'unit': 'A'}, {'value': '224.700', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L1-N'}, {'value': '224.600', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L2-N'}, {'value': '226.500', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L3-N'}, {'value': '391.400', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L1-L2'}, {'value': '391.200', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L2-L3'}, {'value': '393.100', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L3-L1'}, {'value': '50.040', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'location': 'Outlet'}, {'value': '43.000', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'location': 'Body', 'unit': 'Celsius'}]}]}>'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 185, in _handle_call
    response = handler(**snake_case_payload)
  File "/usr/local/lib/python3.9/site-packages/ocpp/routing.py", line 47, in inner
    return func(*args, **kwargs)
  File "/config/custom_components/ocpp/api.py", line 754, in on_meter_values
    self.process_phases(unprocessed)
  File "/config/custom_components/ocpp/api.py", line 719, in process_phases
    self._metrics[metric] = round(sum, 1)
UnboundLocalError: local variable 'sum' referenced before assignment

```

```
2021-07-21 20:00:05 INFO (MainThread) [ocpp] RepHome-1: send [4,"15","InternalError","An unexpected error occurred.",{}]
2021-07-21 20:01:23 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"16","MeterValues",{"connectorId":1,"transactionId":1626878818,"meterValue":[{"timestamp":"2021-07-21T17:42:42Z","sampledValue":[{"value":"1336750.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"224.700","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"224.600","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"226.500","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"391.400","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"391.200","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"393.100","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"},{"value":"50.040","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"43.000","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}]}]}]
2021-07-21 20:01:23 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=16, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1626878818, 'meterValue': [{'timestamp': '2021-07-21T17:42:42Z', 'sampledValue': [{'value': '1336750.000', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'location': 'Outlet', 'unit': 'Wh'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L1'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L2'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'location': 'Outlet', 'unit': 'W', 'phase': 'L3'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Power.Factor', 'location': 'Outlet'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L1'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L2'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'location': 'Outlet', 'unit': 'A', 'phase': 'L3'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'location': 'Outlet', 'unit': 'A'}, {'value': '224.700', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L1-N'}, {'value': '224.600', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L2-N'}, {'value': '226.500', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L3-N'}, {'value': '391.400', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L1-L2'}, {'value': '391.200', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L2-L3'}, {'value': '393.100', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'location': 'Outlet', 'unit': 'V', 'phase': 'L3-L1'}, {'value': '50.040', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'location': 'Outlet'}, {'value': '43.000', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'location': 'Body', 'unit': 'C [... cut ...]
2021-07-21 20:03:00 INFO (MainThread) [ocpp] RepHome-1: receive message [3,"04e578a5-b41d-49a0-895f-b1625249b4ce",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-07-21 20:03:04 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"17","BootNotification",{"chargePointVendor":"Alfen BV","chargePointModel":"NG910-60023","chargePointSerialNumber":"ace0025893","chargeBoxSerialNumber":"RepHome-1","firmwareVersion":"4.15.0-3910"}]
2021-07-21 20:03:04 INFO (MainThread) [ocpp] RepHome-1: send [3,"17",{"currentTime":"2021-07-21T18:03:04.791874+00:00","interval":30,"status":"Accepted"}]
```

lbbrhzn, 2021-07-21:
Oh oh.. I'll try to fix this!

lbbrhzn, 2021-07-21:
I think there are some spaces missing on lines 719 of api.py

lbbrhzn, 2021-07-21:
Released v0.3.3 as a hotfix. @Rep-Hueman, can you give it a spin?

drc38, 2021-07-21:
> I think there are some spaces missing on lines 719 of api.py

Thanks for resolving I shouldn't have tried to update from my phone! I've noticed a small issue in how ` self._extra_attr` is set as it should also add location and its value from the payload. I can resolve next week unless you want to fix @lbbrhzn?

lbbrhzn, 2021-07-21:
Shouldn't line 720 be removed? The phase related attributes are set in the for loop earlier.

drc38, 2021-07-21:
> Shouldn't line 720 be removed? The phase related attributes are set in the for loop earlier.

The loop earlier gathers the phase information together into 1 dictionary entry. Once gathered and used to calculate overall metric value then the extra attributes are updated. I guess the first loop could access/update the `self._extra_attr` instead of local variable.

drc38, 2021-07-21:
Line 749 should be:
`(self._extra_attr[sv[om.meaurand.value]])[om.location.value] = ...`

Rep-Hueman, 2021-07-21:
> Released v0.3.3 as a hotfix. @Rep-Hueman, can you give it a spin?

It is working now..

no error in HA

```
2021-07-21 22:38:28 INFO (MainThread) [ocpp] RepHome-1: receive message [2,"134","MeterValues",{"connectorId":1,"transactionId":1626878818,"meterValue":[{"timestamp":"2021-07-21T17:47:43Z","sampledValue":[{"value":"1336750.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Factor","location":"Outlet"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L1"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L2"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A","phase":"L3"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"226.200","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-N"},{"value":"226.500","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-N"},{"value":"228.000","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-N"},{"value":"394.300","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L1-L2"},{"value":"394.400","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L2-L3"},{"value":"395.900","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V","phase":"L3-L1"},{"value":"50.050","context":"Sample.Periodic","measurand":"Frequency","location":"Outlet"},{"value":"43.000","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}]}]}]
2021-07-21 22:38:28 INFO (MainThread) [ocpp] RepHome-1: send [3,"134",{}]
```

closed 2021-07-21 by Rep-Hueman