lbbrhzn/ocpp #99: Let Enums inherit from Str class also

lbbrhzn, 2021-07-21:
**Is your feature request related to a problem? Please describe.**
Key values are currently defined in enums, which is great for preventing typos through linting. The syntax is a bit verbose though:

```python
self._metrics[om.Frequency.value]=value
```

**Describe the solution you'd like**
Using multiple inheritance, you can force the enum value to become a string:

```python
self._metrics[om.Frequency]=value
```

**Describe alternatives you've considered**
None

**Additional context**
See [here](https://stackoverflow.com/questions/58608361/string-based-enum-in-python#58608362) for more info.

drc38, 2021-07-21:
Agree, we should raise this in upstream repo for consistency.

lbbrhzn, 2021-07-21:
mobilityhouse/ocpp is already doing this.

drc38, 2021-07-22:
The approach used in mobilityhouse/ocpp provides a string but not of the value assigned. Eg  Measurand.voltage returns the string "Measurand.voltage". Instead the class could be defined as a `StrEnum` as per this

lbbrhzn, 2021-11-11:
in mobilityhouse/occp repo the enums are used to translate from pythonic snake case enum keys to OCPP camel case strings. Since these strings are part of the OCPP standard, these values should not be changed. I don't feel we should propose upstream to change their approach.

lbbrhzn, 2021-11-11:
closed, nothing to be done.

closed 2021-11-11 by lbbrhzn