lbbrhzn/ocpp #133: Cannot write to maximum current in HA 2021.9

smbunn, 2021-09-02:
That this seems to be wrong: class Number(InputNumber):
Seems like it should derive from https://github.com/home-assistant/core/tree/dev/homeassistant/components/number

Log error is:
Logger: homeassistant.components.websocket_api.http.connection
Source: components/number/__init__.py:61
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 17:44:08 (3 occurrences)
Last logged: 18:21:09

[140123908151952] 'Number' object has no attribute 'min_value'
[140123874083888] 'Number' object has no attribute 'min_value'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1492, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1527, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 843, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 61, in async_set_value
    if value < entity.min_value or value > entity.max_value:
AttributeError: 'Number' object has no attribute 'min_value'

lbbrhzn, 2021-09-03:
Hmm, It seems I got confused here. I deliberately chose to derive from input_number, because it is derived from RestoreEntity, so the state is restored on reboot.  This seems to be incompatible with number, so it needs to be changed.

drc38, 2021-09-03:
@lbbrhzn , looks like HA 2021.9 has also already deprecated the last_rest feature added in HA 2021.8:
```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:213
Integration: Sensor (documentation, issues)
First occurred: 9:13:27 (24 occurrences)
Last logged: 9:20:10

Entity sensor.evnex_error_code (<class 'custom_components.ocpp.sensor.ChargePointMetric'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
```

PR #143 (2021-09-06): Update Number platform
@lbbrhzn, I'd already done some work on this. Feel free to amend. Fixes #133
closed 2021-09-06 by lbbrhzn