lbbrhzn/ocpp #135: Timeout when talking to charger

smbunn, 2021-09-03:
My Home Assistant is running inside docker inside a Ubuntu 20.04 computer that also uses nginx reverse proxy.  Do you think the reverse proxy could be screweing up communications?  If the charger sends a message on the defined port to my ubuntu machine how does that know to pass that message to Home Assistant inside the docker container?  

## Version of the custom_component

custom_components/ocpp/api.py:382

## Configuration

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:382
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 15:25:18 (1 occurrences)
Last logged: 15:25:18

Exception occurred: Waited 30s for response on [2,"01de410a-8efb-4817-a7d2-34816aff4fb4","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 265, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 681, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 298, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 160, in on_connect
    await self.charge_points[self.cpid].start()
  File "/config/custom_components/ocpp/api.py", line 698, in start
    await asyncio.gather(super().start(), self.post_connect())
  File "/config/custom_components/ocpp/api.py", line 324, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 382, in get_supported_features
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 268, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"01de410a-8efb-4817-a7d2-34816aff4fb4","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].

lbbrhzn, 2021-09-03:
OCPP uses websockets to communicate with the charger. These are two way connections. The server (home assistant) can only send something when the client (charger) is connected. 

A reverse proxy for websockets needs some special care to setup. I have no experience myself, but this seems like a good link:
https://www.serverlab.ca/tutorials/linux/web-servers-linux/how-to-reverse-proxy-websockets-with-apache-2-4/

lbbrhzn, 2021-09-06:
BTW, your docker container should expose the port on your ubuntu machine. Did you configure this correctly?

closed 2021-09-19 by lbbrhzn