lbbrhzn/ocpp #141: Sensors come and go

smbunn, 2021-09-05:
## Version of the custom_component

v0.3.7   (with the input_number code)
-->

## Configuration

No yaml requried

## Describe the bug

When first installed and launched OCPP gest most of the parameters from my Evnex unit.  I see them under 'configuration..Device''"  and in the panel copied to Lovelace.  The gradually most of them dissapear.   The log file linked below shows this occuring, you can see parameters like serial number and firmware version lilsted at 17:36:10 but by 17:40:24 the same query (getConfiguration) returns significantly less parameters

## Debug log

https://pastebin.ubuntu.com/p/jtnXCr3Nnm/

lbbrhzn, 2021-09-06:
solved in v0.3.8

lbbrhzn, 2021-09-06:
Can you check if this issue is solved in v0.3.8?  We have corrected an issue with ping timeouts.

smbunn, 2021-09-06:
Yes, this seems to be all fixed now in v0.3.9.  Thanks for the very fast releases!  Much appreciated

lbbrhzn, 2021-09-06:
great!

closed 2021-09-06 by lbbrhzn