lbbrhzn/ocpp #148: Maximum Current not working

jonasbkarlsson, 2021-09-06:
Using v0.3.9, I used the Maximum_Current slider to set 10 A, which sent `"limit":10.0` to the charger.
After that, clicking on the Charge_Control switch, `"limit":32.0` was sent to the charger.
Is that correct behavior?

```
2021-09-06 20:31:39 INFO (MainThread) [ocpp] CHARGER: send [2,"98b3812a-7dae-40f7-affd-253b1831435f","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":20,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10.0}]}}}]
2021-09-06 20:31:39 INFO (MainThread) [ocpp] CHARGER: receive message [3,"98b3812a-7dae-40f7-affd-253b1831435f",{"status":"Accepted"}]
```

```
2021-09-06 20:32:31 INFO (MainThread) [ocpp] CHARGER: send [2,"a874459f-24d6-4b4c-9c7f-efb779ce8ab2","RemoteStartTransaction",{"idTag":"CHARGER","connectorId":1,"chargingProfile":{"chargingProfileId":1,"stackLevel":20,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
2021-09-06 20:32:31 INFO (MainThread) [ocpp] CHARGER: receive message [3,"a874459f-24d6-4b4c-9c7f-efb779ce8ab2",{"status":"Accepted"}]
```

lbbrhzn, 2021-09-06:
hmm, doesn't seem that logical to me.
start_transaction queries the maximum allowable rate for chargers that support the smart charging profile, and then uses that limit in the charging profile. It would make more sense to use the limit set by the slider. @drc38, do you agree?

drc38, 2021-09-06:
Agree, it makes more sense to remotely start with the current limit set by the slider. I'll take a look when I can.

drc38, 2021-09-06:
@jonasbkarlsson, can you answer the following for me on your charger https://github.com/lbbrhzn/ocpp/discussions/111

jonasbkarlsson, 2021-09-07:
Sorry, but I'm not able to completely test my charger yet. No access to an EV at the moment.

smbunn, 2021-09-08:
If i move the slider I get an error straight away.
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:721
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 15:19:36 (1 occurrences)
Last logged: 15:19:36

Rejected while setting MeterValuesSampledData to Energy.Active.Import.Register,Power.Active.Import,Power.Offered

smbunn, 2021-09-08:
Also getting this every time I restart HA
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:531
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 15:39:44 (1 occurrences)
Last logged: 15:39:44

Failed with response: Scheduled

PR #149 (2021-09-08): Remove setting charging profile from start_transaction
Fixes #148, maximum charge rate to be set by maximum_current slider only
commit 38d4d6c8 (2021-09-08): Remove setting charging profile from start_transaction (#149)

* Remove setting charging profile from start_transaction

Fixes #148, maximum charge rate to be set by maximum_current slider only

* fix testing error

* fix data transfer service in tests
closed 2021-09-08 by lbbrhzn