lbbrhzn/ocpp #152: Using services in automation actions

jonasbkarlsson, 2021-09-07:
### Discussed in https://github.com/lbbrhzn/ocpp/discussions/150

<div type='discussions-op-text'>

<sup>Originally posted by **jonasbkarlsson** September  7, 2021</sup>
Not sure what the problem is so I try to ask here.

If I configure the automation action as

```
service: ocpp.configure
data:
  ocpp_key: MeterValueSampleInterval
  value: "120"
```
it works well. But if I try

```
service: ocpp.configure
data:
  ocpp_key: MeterValueSampleInterval
  value: "{{ states('input_number.ocpp_test') | round (0)}}"
```
I get the ERROR:

```
2021-09-07 09:47:56 ERROR (MainThread) [homeassistant.components.automation.ocpp_test] OCPP test: Error executing script. Invalid data for call_service at pos 1: expected str for dictionary value @ data['value']
2021-09-07 09:47:56 ERROR (MainThread) [homeassistant.components.automation.ocpp_test] Error while executing automation automation.ocpp_test: expected str for dictionary value @ data['value']
```
How to convert the numerical value to a string? Or what should be done?</div>

jonasbkarlsson, 2021-09-07:
I got some comments on this issue here: (https://community.home-assistant.io/t/creating-string-value-from-input-number-state-in-automation-actions/336473)
So it seems like it needs to be fixed in this integration.

drc38, 2021-09-07:
@jonasbkarlsson , I have not tested #153 but according to your link it should solve the templating issue. Please test.

jonasbkarlsson, 2021-09-07:
I have tested #153. It solves the templating issue. Great!

PS. I have noticed that this component floods the `home-assistant.log` with a lot of INFO-level logger messages. Should it really be like that? I needed to add the following four lines in `configuration.yaml` in order to turn it off.  (And note `ocpp` and not `custom_components.ocpp`, which didn't have any effect.)
```
logger:
  default: warning
  logs:
    ocpp: warning
```

drc38, 2021-09-07:
Ocpp controls the underlying MobilityHouse python logger used to communicate with the charger. Up to you if you want to disable the messages but while the integration is in its infancy it is useful for debugging (eg if an ocpp message is rejected)

drc38, 2021-09-07:
Also if your default is warning then I don't think you'll get anything logged when using the custom ocpp services as they are at info level.

PR #153 (2021-09-08): use config_validation on services
Fixes #152
closed 2021-09-08 by lbbrhzn