lbbrhzn/ocpp #154: Keep losing the OCPP device from Home Assitant

smbunn, 2021-09-09:
## Version of the custom_component

Version v0.3.10
## Configuration

```yaml

```

## Describe the bug

If I restart HA, I get the EV Charge info appearing on Lovelace.  After only a few minutes it disappears again.  I look at the log and can see heartbeat still runnning.  I can also see the request for features succesfully bringing verything back, 
## Debug log

```text

2021-09-09 12:42:26 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"8e108dec-61fd-4b94-b0e9-e95c23779dd1","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-09 12:42:26 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"6d455b48-e9b8-b3cc-c924-2aa402067eef","Heartbeat",{}]
2021-09-09 12:42:26 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"6d455b48-e9b8-b3cc-c924-2aa402067eef",{"currentTime":"2021-09-09T00:42:26.832610+00:00"}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"8e108dec-61fd-4b94-b0e9-e95c23779dd1",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging"}]}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"2fa8cb23-6590-465d-a86c-1b6fbcac030b","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"2fa8cb23-6590-465d-a86c-1b6fbcac030b",{"status":"Accepted"}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"d6d3cd5c-0bc7-491c-8e7e-c9d24ec9ae46","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"e603e441-efab-435d-0c57-1fc113cc7716","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-09-09T00:42:26Z"}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"e603e441-efab-435d-0c57-1fc113cc7716",{}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"d6d3cd5c-0bc7-491c-8e7e-c9d24ec9ae46",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"30"}]}]
2021-09-09 12:42:27 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"addc4470-c3ee-4499-a062-6d3a0ffa919b","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"addc4470-c3ee-4499-a062-6d3a0ffa919b",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"ae9edd56-9a51-4505-8e2a-d7413abcc4e7","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"ae9edd56-9a51-4505-8e2a-d7413abcc4e7",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import"}]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"452e81d8-f931-4d08-97ae-a1481913289c","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"452e81d8-f931-4d08-97ae-a1481913289c",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2021-09-09 12:42:28 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"eda4e565-6b84-4591-8245-916baf88c3c2","GetConfiguration",{"key":["NumberOfConnectors"]}]
2021-09-09 12:42:29 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"eda4e565-6b84-4591-8245-916baf88c3c2",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-09-09 12:45:36 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"a3edb558-3885-4bef-a75a-cff02530563d","Reset",{"type":"Hard"}]
2021-09-09 12:45:36 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"a3edb558-3885-4bef-a75a-cff02530563d",{"status":"Accepted"}]
2021-09-09 12:45:54 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"43c4164f-844b-275b-3afc-e2898a915c53","BootNotification",{"chargePointModel":"E7-T2S","chargePointVendor":"Evnex","firmwareVersion":"1.1.0.26E4747","chargePointSerialNumber":"AR21120320","meterType":"D111"}]
2021-09-09 12:45:54 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"43c4164f-844b-275b-3afc-e2898a915c53",{"currentTime":"2021-09-09T00:45:54.739409+00:00","interval":30,"status":"Accepted"}]
2021-09-09 12:45:54 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"ff1f4448-ef1f-4ff0-b945-47611db19169","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"f96b7478-92ae-8004-818f-46ef99f7c1fe","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2021-09-09T00:45:45Z"}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"f96b7478-92ae-8004-818f-46ef99f7c1fe",{}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"ff1f4448-ef1f-4ff0-b945-47611db19169",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging"}]}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"29ff9d44-60bb-40ce-86b9-ca1b9f49bf2d","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"386cd2d0-c2c0-cb69-9438-9d8a1df04fb9","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-09-09T00:45:45Z"}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"386cd2d0-c2c0-cb69-9438-9d8a1df04fb9",{}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"29ff9d44-60bb-40ce-86b9-ca1b9f49bf2d",{"status":"Accepted"}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"497b6824-bb03-413a-bd7e-ee3db2fd0207","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"c2c0bd0d-d0f3-b57a-7cc8-d121b29d5210","DataTransfer",{"vendorId":"Evnex","messageId":"info","data":"Reboot Reason: 7, Reboot Flags: 0x0b"}]
2021-09-09 12:45:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"c2c0bd0d-d0f3-b57a-7cc8-d121b29d5210",{"status":"Accepted"}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"497b6824-bb03-413a-bd7e-ee3db2fd0207",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"30"}]}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"858c955b-940c-4301-aab7-02fff1ed73ef","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"089a0057-5de4-7b92-31ad-8ef4fe86c949","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-09-09T00:45:54Z"}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"089a0057-5de4-7b92-31ad-8ef4fe86c949",{}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"858c955b-940c-4301-aab7-02fff1ed73ef",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2021-09-09 12:45:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"12fd7f81-6ee4-4b4a-bec2-a8c01f09c304","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-09 12:45:57 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"12fd7f81-6ee4-4b4a-bec2-a8c01f09c304",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import"}]}]
2021-09-09 12:45:57 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"97e3ae1c-0cf7-477f-ae99-40e099c4bb48","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2021-09-09 12:45:57 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"97e3ae1c-0cf7-477f-ae99-40e099c4bb48",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2021-09-09 12:45:57 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"f536c931-1d7c-4f37-a06e-c645de825469","GetConfiguration",{"key":["NumberOfConnectors"]}]
2021-09-09 12:45:57 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"f536c931-1d7c-4f37-a06e-c645de825469",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-09-09 12:46:25 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"032eb682-bee8-7b9e-73a7-ac07e80d25ee","Heartbeat",{}]
[... cut ...]
2021-09-09 12:49:25 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"be5ea9b8-3243-eb68-2beb-770d3bd8dd6e","Heartbeat",{}]
2021-09-09 12:49:25 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"be5ea9b8-3243-eb68-2beb-770d3bd8dd6e",{"currentTime":"2021-09-09T00:49:25.238923+00:00"}]
2021-09-09 12:49:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"5ee803ee-4212-dbb5-3e17-7536a74fbeee","Heartbeat",{}]
2021-09-09 12:49:55 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"5ee803ee-4212-dbb5-3e17-7536a74fbeee",{"currentTime":"2021-09-09T00:49:55.546061+00:00"}]

```

drc38, 2021-09-09:
Hi Simon, can you include a screenshot of your Lovelace. From the above log I would expect most to be Unknown with exception of those returned by the BootNotification and StatusNotification messages.

smbunn, 2021-09-10:
Thats exactly what I was getting, everything unknown.  Iupgraded this morning to the latest OCPP.  I had to delete the integrtaion and re-install it.  I saw it connect and load most values.  I hit reset and saw it load almost all values, then seconds later it went all greyed out and 'unknown'

Here is my log

2021-09-11 10:16:08 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"21f6fabc-8097-4f66-8b92-c64b064d87d5","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2021-09-11 10:16:09 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"21f6fabc-8097-4f66-8b92-c64b064d87d5",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2021-09-11 10:16:09 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"c55a233e-8878-460f-a53c-0144f55ecd19","GetConfiguration",{"key":["NumberOfConnectors"]}]
2021-09-11 10:16:09 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"c55a233e-8878-460f-a53c-0144f55ecd19",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-09-11 10:16:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"c436740f-be73-4b65-84b1-772dc460bba4","Reset",{"type":"Hard"}]
2021-09-11 10:16:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"c436740f-be73-4b65-84b1-772dc460bba4",{"status":"Accepted"}]
2021-09-11 10:16:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"0600da8f-06a4-afb7-bff8-c5419b58370d","StopTransaction",{"timestamp":"2021-09-10T22:16:55Z","meterStop":685538,"transactionId":1631312098,"reason":"HardReset"}]
2021-09-11 10:16:56 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"0600da8f-06a4-afb7-bff8-c5419b58370d",{"idTagInfo":{"status":"Accepted"}}]
2021-09-11 10:17:14 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"582b5db0-a0e1-4549-b2f3-7e0c2e63377e","BootNotification",{"chargePointModel":"E7-T2S","chargePointVendor":"Evnex","firmwareVersion":"1.1.0.26E4747","chargePointSerialNumber":"AR21120320","meterType":"D111"}]
2021-09-11 10:17:14 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"582b5db0-a0e1-4549-b2f3-7e0c2e63377e",{"currentTime":"2021-09-10T22:17:14.968030+00:00","interval":30,"status":"Accepted"}]
2021-09-11 10:17:14 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"9b8e233b-d420-43b6-9368-6faa99b239e1","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"242821f3-6ccc-3fd3-27c4-10edff79732e","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2021-09-10T22:17:06Z"}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"242821f3-6ccc-3fd3-27c4-10edff79732e",{}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"9b8e233b-d420-43b6-9368-6faa99b239e1",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging"}]}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"5928a341-53d8-40a8-b9df-89fa1130043b","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"12c11e04-59c5-dc38-c17b-7a07a4b8da21","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-09-10T22:17:06Z"}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"12c11e04-59c5-dc38-c17b-7a07a4b8da21",{}]
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"5928a341-53d8-40a8-b9df-89fa1130043b",{"status":"Scheduled"}]
2021-09-11 10:17:15 WARNING (MainThread) [custom_components.ocpp] Failed with response: Scheduled
2021-09-11 10:17:15 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"d13b7e3a-c5eb-4b03-a186-9dff3b26a20a","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"c538848c-1f75-e7c3-b4c9-e178c5ba1d02","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2021-09-10T22:17:06Z"}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"c538848c-1f75-e7c3-b4c9-e178c5ba1d02",{}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"d13b7e3a-c5eb-4b03-a186-9dff3b26a20a",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"30"}]}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"c3895b2f-97d2-48fc-81e3-0f1f422c1c37","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"fde75eb5-fc8c-0243-6f3d-45a586d45f97","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV","timestamp":"2021-09-10T22:17:08Z"}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"fde75eb5-fc8c-0243-6f3d-45a586d45f97",{}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [3,"c3895b2f-97d2-48fc-81e3-0f1f422c1c37",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2021-09-11 10:17:16 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [2,"1c9a7a34-3dbf-4fef-b386-e4fa92de592c","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-11 10:17:17 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"3124995b-08b5-eff2-932f-9d472b7ab4f0","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2021-09-10T22:17:10Z"}]
[... cut ...] iodic","unit":"kWh","value":"685.540"}]}]}]
2021-09-11 10:22:48 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"9d750a56-2f95-4111-408d-3d8d38c53de6",{}]
2021-09-11 10:23:00 INFO (MainThread) [ocpp] EVNEX-JAYDCA: receive message [2,"a001e11e-4bc8-a828-311e-eae29f8e4d1a","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV","timestamp":"2021-09-10T22:22:59Z"}]
2021-09-11 10:23:00 INFO (MainThread) [ocpp] EVNEX-JAYDCA: send [3,"a001e11e-4bc8-a828-311e-eae29f8e4d1a",{}]

smbunn, 2021-09-10:
I have returned to v3.10 and that is not giving me the error messages in the log about :

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:532
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 10:56:59 (1 occurrences)
Last logged: 10:56:59

Failed with response: Scheduled

drc38, 2021-09-10:
The log looks fine, what is it showing in Lovelace? You can ignore the `Failed with response: Scheduled` message it is in response to the ChangeAvailability request. You can find out more about the `Scheduled` response in the Ocpp specification.

smbunn, 2021-09-14:
I went back to 3.11 and I ahve had crashes each day taking out Home assistant.  Today it also took out the entire VM and then the Ubuntu server the VM was running on.  Maybe unrelated.  My error log showed this right at the moment everything crashed

2021-09-15 09:32:29 ERROR (MainThread) [custom_components.ocpp] Exception occurred:
Waited 30s for response on [2,"9ea9785a-fd8a-48bb-8b95-29e5babe87f7","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 265, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 696, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 298, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 176, in on_connect
    await self.charge_points[self.cpid].start()
  File "/config/custom_components/ocpp/api.py", line 713, in start
    await asyncio.gather(super().start(), self.post_connect())
  File "/config/custom_components/ocpp/api.py", line 350, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 413, in get_supported_features
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 268, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"9ea9785a-fd8a-48bb-8b95-29e5babe87f7","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].

drc38, 2021-09-14:
There's a couple of PR's awaiting approval that might help. They are already included in my repo.

smbunn, 2021-09-16:
OK. I restart often as the OCPP device drops out all the time.  I had it running for 2 days then today I have had to restart 3 times.
the error is always:
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:532
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 15:10:57 (2 occurrences)
Last logged: 15:11:44

Failed with response: Scheduled

lbbrhzn, 2021-09-19:
Please check with release [v0.3.12](https://github.com/lbbrhzn/ocpp/releases/tag/v0.3.12)

lbbrhzn, 2021-11-11:
Hi @smbunn, are you still seeing this with recent versions? If not, i'll close this isue.

closed 2022-01-13 by lbbrhzn