lbbrhzn/ocpp #160: Maximum_Current is set to 32 after HA restart

jonasbkarlsson, 2021-09-10:
## Version of the custom_component

v0.3.11

## Configuration

```yaml
```

## Describe the bug

Maximum_Current is set to 32 after HA restart. Wouldn't it be better to set Maximum_Current to the value it had before the restart? 

## Debug log

```yaml
```

Rep-Hueman, 2021-09-10:
My charger supports 32 (3 phase) but my installation can only handle max 3x16A! 
I have a 3x25 fuse for the whole house!

The charger must all the time max 16A (3 phase)

My Alfen charging station is linked to the smart meter via the P1 port. If the connection value (3x25A) of my house is exceeded, my electric car will be charged more slowly.

ps,
Is it also possible to switch from 3 phase to 1 phase and vice versa with OCPP?

jonasbkarlsson, 2021-09-11:
I assume it's the Charging Point's responsibility to offer a suitable charging current, so it should be possible to locally (i.e. not via OCPP) configure the Charging Point with information about the house fuse (or an even lower value if that is desirable). And then the offered current should become the minimum of the locally configured limit and the limit requested by OCPP.

Reading the OCPP spec, it should be possible to request either 3 phase and 1 phase with OCPP.  However, some EVs does not support switching from 3 phases to 1 phase during a charging session. They can even be physically damaged!

```
The amount of phases used during charging is limited by the capabilities of: The
Charge Point, EV and Cable between CP and EV. If any of these 3 is not capable of
3 phase charging, the EV will be charged using 1 phase only.

Switching the number of used phases during a schedule or charging session
should be done with care. Some EVs may not support this and changing the
amount of phases may result in physical damage. With the configuration key:
ConnectorSwitch3to1PhaseSupported The Charge Point can tell if it supports
switching the amount of phases during a transaction.
```

drc38, 2021-09-11:
As @jonasbkarlsson says a charger should have dip switches or similar set to protect upstream cables etc by limiting the maximum current offered or a local smart limit that will override the Ocpp setting. So initialising the `maximum_current` slider to 32A on a restart is not an issue per se.

On a HA restart the charger current limit is not known, one approach (for those chargers that support a trigger message and current.offered measurand) would be to request this in the `post_connect` and then use the value returned to set the `maximum_current` slider. I don't have a charger that supports either though so can't test.

lbbrhzn, 2021-09-11:
It would be nice to store the value of the max current persistently in HA, so it is restored on reboot. The input_number class is derived from RestoreEntity, which should be able to handle this. More info can be found here:
https://aarongodfrey.dev/programming/restoring-an-entity-in-home-assistant/

jonasbkarlsson, 2021-09-11:
On the same theme, the values received by the Boot Notification (Vendor, Model, Serial, FW.Version) are also lost when HA reboots. Not as important as the Maximum_Current, but it would be nice if they also were persistent.

lbbrhzn, 2021-09-19:
I agree. Will take some time to implement though.

drc38, 2022-01-02:
Refer v0.4.3 and PR #266

closed 2022-01-02 by drc38