lbbrhzn/ocpp #169: ABB charger working, but incorrect voltage reported

lbbrhzn, 2021-09-21:
### Discussed in https://github.com/lbbrhzn/ocpp/discussions/168

<div type='discussions-op-text'>

<sup>Originally posted by **eogmau** September 21, 2021</sup>
Have tested the ABB Terra AC model W7-G5-R-0 and working, can be added to the list. The charger.Voltage is reporting 76.6V, not sure why. 
[image]
</div>

lbbrhzn, 2021-09-21:
Looks like your charger is reporting only a single phase voltage, while OCPP is assuming three phases, so it is averaging (229.9 + 0 + 0) / 3 = 76.6 V

lbbrhzn, 2021-09-21:
@eogmau, can you share a debug log? I'm curious why only L1-N is reported, while this is a 3 phase charger.

eogmau, 2021-09-22:
@lbbrhzn Thanks for looking at this, will get the logs later today.

eogmau, 2021-09-22:
Extracted the logs.
`2021-09-22 18:34:59 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "4289088", "BootNotification", {"chargePointModel": "CDT_TACW7::NET_WIFI", "chargePointVendor": "Chargedot", "chargeBoxSerialNumber": "ABBTerraAC", "firmwareVersion": "TAC1Z9118606710273::V1.4.2", "meterType": "V1"}]
2021-09-22 18:34:59 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"4289088",{"currentTime":"2021-09-22T17:34:59.634265+00:00","interval":3600,"status":"Accepted"}]
2021-09-22 18:35:00 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "9274563", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x00"}]
2021-09-22 18:35:00 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"9274563",{}]
2021-09-22 18:35:00 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"4ddf9f48-7410-4b4e-bdba-cdda51f67aa5","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-22 18:35:01 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "5706945", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Preparing", "vendorErrorCode": "0x00"}]
2021-09-22 18:35:01 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"5706945",{}]
2021-09-22 18:35:02 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "4ddf9f48-7410-4b4e-bdba-cdda51f67aa5", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement"}], "unknownKey": []}]
2021-09-22 18:35:02 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"81f1b569-2735-4dd8-a7b7-6c71ed41dab8","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-22 18:35:02 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "81f1b569-2735-4dd8-a7b7-6c71ed41dab8", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement"}], "unknownKey": []}]
2021-09-22 18:35:02 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"fbd47206-b8d5-4e71-8813-8063c0e5bf2b","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-22 18:35:03 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "fbd47206-b8d5-4e71-8813-8063c0e5bf2b", {"status": "Accepted"}]
2021-09-22 18:35:03 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"4a9d937b-6805-47e9-8176-5d64c468f7e5","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-22 18:35:04 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "4a9d937b-6805-47e9-8176-5d64c468f7e5", {"status": "Accepted"}]
2021-09-22 18:35:04 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"7cef5eb3-5603-4407-a29b-6f5d9916bc9f","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-22 18:35:05 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "7cef5eb3-5603-4407-a29b-6f5d9916bc9f", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2021-09-22 18:35:05 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"72ecb4a7-2af2-4c5f-888d-ece253b167d9","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-22 18:35:06 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "72ecb4a7-2af2-4c5f-888d-ece253b167d9", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2021-09-22 18:35:06 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"7162de1b-85e3-47f1-adb7-9ecca99d146e","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-22 18:35:07 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "7162de1b-85e3-47f1-adb7-9ecca99d146e", {"configurationKey": [], "unknownKey": ["WebSocketPingInterval"]}]
2021-09-22 18:35:07 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"dc3e00b1-535c-47ba-8051-9ecf3eaf5f93","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-22 18:35:07 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "dc3e00b1-535c-47ba-8051-9ecf3eaf5f93", {"configurationKey": [], "unknownKey": ["WebSocketPingInterval"]}]
2021-09-22 18:35:07 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"f64effb7-7ffa-43ee-9b78-33e34b3990d3","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"60"}]
2021-09-22 18:35:08 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "f64effb7-7ffa-43ee-9b78-33e34b3990d3", {"status": "NotSupported"}]
2021-09-22 18:35:08 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"7d60e30f-f4b1-477c-9f3e-a375084c9563","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"60"}]
2021-09-22 18:35:09 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "7d60e30f-f4b1-477c-9f3e-a375084c9563", {"status": "NotSupported"}]
2021-09-22 18:35:09 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"9300bd1e-4c1c-47a5-8b3b-13355787b890","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-22 18:35:10 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "9300bd1e-4c1c-47a5-8b3b-13355787b890", {"configurationKey": [], "unknownKey": ["MeterValuesSampledData"]}]
2021-09-22 18:35:10 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"029f228f-1354-4e53-9286-d516a9eedbae","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-22 18:35:11 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "029f228f-1354-4e53-9286-d516a9eedbae", {"configurationKey": [], "unknownKey": ["MeterValuesSampledData"]}]
[... cut ...] and": "Current.Import", "phase": "L1", "unit": "A"}, {"value": "1740", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "463", "context": "Sample.Periodic", "format": "Raw", "measurand": "Energy.Active.Import.Register", "unit": "Wh"}, {"value": "8", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}]}]}]
2021-09-22 18:52:21 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"0510572",{}]
`

eogmau, 2021-09-25:
@drc38 have tested your fix and unfortunately the Voltage is not displayed.
Thanks for looking at this.
 
[image]

Have extracted new logs
`
2021-09-25 08:22:44 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "4289088", "BootNotification", {"chargePointModel": "CDT_TACW7::NET_WIFI", "chargePointVendor": "Chargedot", "chargeBoxSerialNumber": "ABBTerraAC", "firmwareVersion": "TAC1Z9118606710273::V1.4.2", "meterType": "V1"}]
2021-09-25 08:22:44 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"4289088",{"currentTime":"2021-09-25T07:22:44.454537+00:00","interval":3600,"status":"Accepted"}]
2021-09-25 08:22:45 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "9274563", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x00"}]
2021-09-25 08:22:45 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"9274563",{}]
2021-09-25 08:22:46 INFO (MainThread) [ocpp] ABBTerraAC: receive message [2, "5706945", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x00"}]
2021-09-25 08:22:46 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"5706945",{}]
2021-09-25 08:22:46 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"e1808b46-181a-42ef-b565-e8421eaa52f8","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-25 08:22:46 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "e1808b46-181a-42ef-b565-e8421eaa52f8", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement"}], "unknownKey": []}]
2021-09-25 08:22:46 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"e1806bfc-694f-4dd2-a182-d78aa501c869","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2021-09-25 08:22:47 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "e1806bfc-694f-4dd2-a182-d78aa501c869", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement"}], "unknownKey": []}]
2021-09-25 08:22:47 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"045e83c7-9afe-4c59-a1f8-1e992e58cc63","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-25 08:22:48 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "045e83c7-9afe-4c59-a1f8-1e992e58cc63", {"status": "Accepted"}]
2021-09-25 08:22:48 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"c82f2b02-09e1-4eb7-b0bc-316225878bf8","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-09-25 08:22:49 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "c82f2b02-09e1-4eb7-b0bc-316225878bf8", {"status": "Accepted"}]
2021-09-25 08:22:49 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"f807c6b8-8934-4734-9322-4990db3938a4","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-25 08:22:50 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "f807c6b8-8934-4734-9322-4990db3938a4", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2021-09-25 08:22:50 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"5ed4258d-6563-453b-9bc1-431827a5e134","GetConfiguration",{"key":["HeartbeatInterval"]}]
2021-09-25 08:22:50 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "5ed4258d-6563-453b-9bc1-431827a5e134", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2021-09-25 08:22:50 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"e8fb849f-902b-4d5a-ae7b-566aa70b6f7c","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-25 08:22:51 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "e8fb849f-902b-4d5a-ae7b-566aa70b6f7c", {"configurationKey": [], "unknownKey": ["WebSocketPingInterval"]}]
2021-09-25 08:22:51 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"ec5e11e8-7667-4efc-adb1-0544345a41ef","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2021-09-25 08:22:52 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "ec5e11e8-7667-4efc-adb1-0544345a41ef", {"configurationKey": [], "unknownKey": ["WebSocketPingInterval"]}]
2021-09-25 08:22:52 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"b243d630-5582-43d7-bf16-3d95b74faa36","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"60"}]
2021-09-25 08:22:53 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "b243d630-5582-43d7-bf16-3d95b74faa36", {"status": "NotSupported"}]
2021-09-25 08:22:53 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"3e2363f4-c603-4ef2-82ae-a0c5a988ea04","GetConfiguration",{"key":["AuthorizeRemoteTxRequests"]}]
2021-09-25 08:22:54 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "3e2363f4-c603-4ef2-82ae-a0c5a988ea04", {"configurationKey": [{"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}], "unknownKey": []}]
2021-09-25 08:22:54 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"6b919659-df4b-46d9-848b-9923e0a8707e","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"60"}]
2021-09-25 08:22:55 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "6b919659-df4b-46d9-848b-9923e0a8707e", {"status": "NotSupported"}]
2021-09-25 08:22:55 INFO (MainThread) [ocpp] ABBTerraAC: send [2,"1d0ca6bc-13c0-4749-a2d5-06a36b0430b8","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2021-09-25 08:22:55 INFO (MainThread) [ocpp] ABBTerraAC: receive message [3, "1d0ca6bc-13c0-4749-a2d5-06a36b0430b8", {"configurationKey": [], "unknownKey": ["MeterValuesSampledData"]}]
[... cut ...]  "Current.Import", "phase": "L1", "unit": "A"}, {"value": "1710", "context": "Transaction.Begin", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "0", "context": "Transaction.Begin", "format": "Raw", "measurand": "Energy.Active.Import.Register", "unit": "Wh"}, {"value": "8", "context": "Transaction.Begin", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}]}]}]
2021-09-25 08:27:23 INFO (MainThread) [ocpp] ABBTerraAC: send [3,"0422594",{}]

`

drc38, 2021-09-25:
Thanks, try again with the latest PR

eogmau, 2021-09-27:
Thanks @drc38. The latest PR fixes the voltage.

PR #173 (2021-10-02): Add context to metrics, fix single phase voltage
Should fix #169 but will need testing @eogmau, context also added to metrics to separate charging metrics from clock sampled ones
closed 2021-10-02 by drc38