lbbrhzn/ocpp #171: Unknown values

confucius78, 2021-09-23:
Description:
it looks like my charger only returns total energy values when charging. The device is connected and heartbeat is passed consistently, but total energy values are not refreshed so HA starts to report ‘unknown’ after a while (despite showing the correct graph when checking the entity). Values start refreshing when charging.

No logs retrieved yet, if needed I can provide them.

Config:
HA running on rpi, Alfen pro-line charger in same subnet. Again: connecting like a charm and all functionality is working consistently and continuously

jonasbkarlsson, 2021-09-24:
I had the same problem with my charger. I then realized that the integration currently only configure measurement to be done while charging (the ocpp parameters `MeterValueSampleInterval `and `MeterValuesSampledData`), and my charger by default did not report anything while not chariging.

Reporting while not charging is configured with `ClockAlignedDataInterval `(seconds between reports, a 0 value means no reporting) and `MeterValuesAlignedData `(defines what to report).

In the HA->Development Tools->Services, try 

```
service: ocpp.get_configuration
data:
  ocpp_key: ClockAlignedDataInterval
``` 

```
service: ocpp.get_configuration
data:
  ocpp_key: MeterValuesAlignedData
```

and check the response in the logs or in the Config.Response entity.

drc38, 2021-10-02:
They can also be set:
```
service: ocpp.configure
data:
  ocpp_key: ClockAlignedDataInterval
  value: 900
```
which will report every 15min when not charging for example.

closed 2021-11-05 by lbbrhzn