lbbrhzn/ocpp #172: Error setting up entry central for ocpp

fwestenberg, 2021-09-24:
## Version of the custom_component
v0.3.12

## Configuration
Error occurs at the config flow, no additional YAML configuration.
[image]

## Describe the bug
Error setting up entry central for ocpp. 
This error occurs when trying to configure the integration from the config flow, with the default configuration.
HA version: core-2021.9.7
HACS version: 1.15.2

## Debug log
```
Error setting up entry central for ocpp
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/config/custom_components/ocpp/__init__.py", line 28, in async_setup_entry
    central_sys = await CentralSystem.create(hass, entry)
  File "/config/custom_components/ocpp/api.py", line 137, in create
    server = await websockets.serve(
  File "/usr/local/lib/python3.9/site-packages/websockets/imports.py", line 81, in __getattr__
    source = deprecated_aliases[name]
  File "/usr/local/lib/python3.9/site-packages/websockets/imports.py", line 26, in import_name
    assert level < len(source), "importing from parent isn't supported"
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/server.py", line 26, in <module>
    from ..connection import State
  File "/usr/local/lib/python3.9/site-packages/websockets/connection.py", line 17, in <module>
    from .frames import (
ImportError: cannot import name 'OK_CLOSE_CODES' from 'websockets.frames' (/usr/local/lib/python3.9/site-packages/websockets/frames.py)
```

drc38, 2021-10-06:
I've raised with websockets repo: https://github.com/aaugustin/websockets/issues/1063

drc38, 2021-10-06:
To confirm it is related to websockets 10 change your manifest.json file to `websockets==9.1` and see whether the error occurs.

drc38, 2021-10-08:
> I ran into this, and modified the manifest.json to set `websockets==9.1` and was able to get it start.

If you now reinstall v0.3.13 or change back to `websockets>=10.0` I take it you no longer have the error?

tube0013, 2021-10-08:
confirmed changing it back to `websockets>=10.0` with the integration already set up, and it loads after a restart without issues.

chiefdeputy, 2021-10-08:
Yes, that's the case in my setup, too. It's now runnibg fine with 10.0.

drc38, 2021-11-02:
This should be fixed with the merge of #177, there haven't been any reported issues since

closed 2021-11-02 by drc38