lbbrhzn/ocpp #174: Alfen - NG920 - Eve Double Pro-line

TravisWilder, 2021-09-29:
Hi

here the status of this box after adding.
OCPP 2.0.1 did not work, so this is 1.6
I added as well the documentation from alfen to help supporting this wallbox with 2 Sockets

[image]

[image]

[image]

[Backoffice Configuration keys 4.14.0.pdf](https://github.com/lbbrhzn/ocpp/files/7250138/Backoffice.Configuration.keys.4.14.0.pdf)

Hope this helps to fully support this and other Alfen boxes

Rep-Hueman, 2021-09-29:
Only these meter values are supported by the box

    monitored_variables:
      - "Energy.Active.Import.Register"
      - "Current.Import"
      - "Current.Offered"
      - "Frequency"
      - "Power.Active.Import"
      - "Power.Factor"
      - "Temperature"
      - "Voltage"

Choose this in the OCPP setup in HA!

"Instantaneous current flow to EV",
"Maximum current offered to EV",
"Active energy imported from the grid",
"Powerline frequency",
"Instantaneous active power imported by EV",
"Instantaneous power factor of total energy flow",
"Temperature reading inside Charge Point",
"Instantaneous AC RMS supply voltage"
.
[image]
.
[image]

closed 2021-11-05 by lbbrhzn

TravisWilder, 2021-11-10:
Hi "issue" was not the values alone but that I have two sockets.