lbbrhzn/ocpp #198: Error Unsupported state class and one Unsupported unit in Developer Tools - Statistics

Rep-Hueman, 2021-11-16:
## Version of the custom_component

- 0.3.18
- HA core-2021.11.4

## Describe the bug

Error Unsupported state class and one Unsupported unit in Developer Tools - Statistics

## Debug log

```
charger.Session.Energy sensor.charger_session_energy
kWh
recorder
The state class '' of this entity is not supported.

charger.Meter.Start sensor.charger_meter_start
kWh
recorder
The state class '' of this entity is not supported.

charger.Voltage sensor.charger_voltage
V
recorder
The state class '' of this entity is not supported.

charger.Frequency sensor.charger_frequency
recorder
The state class '' of this entity is not supported.

charger.Connectors sensor.charger_connectors
recorder
The state class '' of this entity is not supported.

charger.Transaction.Id sensor.charger_transaction_id
recorder
The state class '' of this entity is not supported.

charger.Session.Time sensor.charger_session_time
min
recorder
The state class '' of this entity is not supported.

charger.Reconnects sensor.charger_reconnects
recorder
The state class '' of this entity is not supported.

***
FIX ISSUE:

Unsupported state class
The state class of this entity, is not supported.
Statistics can not be generated until this entity has a supported state class.

If this state class was provided by an integration, this is a bug. Please report an issue.

If you have set this state class yourself, please correct it. The different state classes and when to use which can be found in the developer documentation.

***
```
and:
```
charger.Power.Factor  sensor.charger_power_factor:

***
FIX ISSUE:

Unsupported unit
The unit of your entity is not a supported unit for the device class of the entity, power.
Statistics can not be generated until this entity has a supported unit.

If this unit was provided by an integration, this is a bug. Please report an issue.

If you have set this unit yourself, and want to have statistics generated, make sure the unit matches the device class. The supported units are documented in the developer documentation.
***
```

lbbrhzn, 2021-11-16:
thanks for reporting! What are te steps to produce these logs?
If I just look at the statistics tab in the developer tools pane I see different set of issues (see below).
In my case, there seem to be only unit related issues, and no state class related issues.
Can you try to remove the OCCP Integration and add your charger with a different name?

[image]

lbbrhzn, 2021-11-16:
To resolve the unit issues I think the best approach is to provide a constant ha_unit for each measurand upfront, and convert the values reported by the chargepoint from the ocpp_unit to the ha_unit. This involves converting Wh to kWh, and Celcius, Fahrenheit and Kelvin to °C.

lbbrhzn, 2021-11-16:
Please check if the issues are solved in v0.3.19

dreimer1986, 2021-11-19:
I can't speak for his HA instance, but here I had remnants of my renames, when these were removed I reinstalled OCPP addin and selected all features: The sensors of the addin are only shown once:

[image]

In short. The only formerly mentioned flaw I still get from the Android mobile integration, nowhere else.

lbbrhzn, 2021-11-22:
thnx, closing the issue.

closed 2021-11-22 by lbbrhzn