lbbrhzn/ocpp #206: Error in sensor.py for v0.3.20

jonasbkarlsson, 2021-11-21:
## Version of the custom_component
v0.3.20

## Describe the bug

I get errors in sensor.py after upgrading from v0.3.19. 

## Debug log

```
2021-11-21 20:36:34 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform ocpp
Traceback (most recent call last):
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
await asyncio.gather(*tasks)
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/helpers/entity_platform.py", line 517, in _async_add_entity
capabilities=entity.capability_attributes,
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/sensor/__init__.py", line 204, in capability_attributes
if state_class := self.state_class:
File "/home/homeassistant/.homeassistant/custom_components/ocpp/sensor.py", line 142, in state_class
if self.device_class is DEVICE_CLASS_ENERGY:
File "/home/homeassistant/.homeassistant/custom_components/ocpp/sensor.py", line 169, in device_class
or self.metric.lower().startwith("frequency")
AttributeError: 'str' object has no attribute 'startwith'

2021-11-21 20:36:34 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up ocpp platform for sensor
Traceback (most recent call last):
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/helpers/entity_platform.py", line 257, in _async_setup_platform
await asyncio.gather(*pending)
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
await asyncio.gather(*tasks)
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/helpers/entity_platform.py", line 517, in _async_add_entity
capabilities=entity.capability_attributes,
File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/sensor/__init__.py", line 204, in capability_attributes
if state_class := self.state_class:
File "/home/homeassistant/.homeassistant/custom_components/ocpp/sensor.py", line 142, in state_class
if self.device_class is DEVICE_CLASS_ENERGY:
File "/home/homeassistant/.homeassistant/custom_components/ocpp/sensor.py", line 169, in device_class
or self.metric.lower().startwith("frequency")
AttributeError: 'str' object has no attribute 'startwith'
```

And warnings for all sensors. Only one shown below:
`2021-11-21 20:36:42 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.ctek_config_response not found.`

dreimer1986, 2021-11-22:
Likely should be "startswith" not "startwith" in line 169. Fixing that causes all values without any metrix to have Hz though. I have no clue if that "fix" was your intention, but I tried to semi fix that my noobish way (line 103 and 163 are my changes):

sensor.py:
```
"""Sensor platform for ocpp."""

import homeassistant
from homeassistant.components.sensor import (
    STATE_CLASS_MEASUREMENT,
    STATE_CLASS_TOTAL_INCREASING,
    SensorEntity,
)
import homeassistant.const as ha
from homeassistant.const import (
    CONF_MONITORED_VARIABLES,
    DEVICE_CLASS_BATTERY,
    DEVICE_CLASS_CURRENT,
    DEVICE_CLASS_ENERGY,
    DEVICE_CLASS_POWER,
    DEVICE_CLASS_TEMPERATURE,
    DEVICE_CLASS_TIMESTAMP,
    DEVICE_CLASS_VOLTAGE,
)

from .api import CentralSystem
from .const import CONF_CPID, DEFAULT_CPID, DOMAIN, ICON, Measurand
from .enums import HAChargerDetails, HAChargerSession, HAChargerStatuses

# To be added when home assistant supports it
DEVICE_CLASS_FREQUENCY = None
FREQUENCY_RPM = "rpm"
FREQUENCY_HERTZ = "Hz"

async def async_setup_entry(hass, entry, async_add_devices):
    """Configure the sensor platform."""
    central_system = hass.data[DOMAIN][entry.entry_id]
    cp_id = entry.data.get(CONF_CPID, DEFAULT_CPID)
    entities = []
    for measurand in list(
        set(
            entry.data[CONF_MONITORED_VARIABLES].split(",")
            + list(HAChargerDetails)
            + list(HAChargerSession)
            + list(HAChargerStatuses)
        )
    ):
        entities.append(
            ChargePointMetric(
                central_system,
                cp_id,
                measurand,
            )
        )

    async_add_devices(entities, False)

class ChargePointMetric(SensorEntity):
    """Individual sensor for charge point metrics."""

    def __init__(
        self,
        central_system: CentralSystem,
        cp_id: str,
        metric: str,
    ):
        """Instantiate instance of a ChargePointMetrics."""
        self.central_system = central_system
        self.cp_id = cp_id
        self.metric = metric
        self._extra_attr = {}
[... 102 lines cut ...]
            device_class = DEVICE_CLASS_TEMPERATURE
        elif self.metric.lower().startswith("timestamp.") or self.metric in [
            HAChargerDetails.config_response.value,
            HAChargerDetails.data_response.value,
            HAChargerStatuses.heartbeat.value,
        ]:
            device_class = DEVICE_CLASS_TIMESTAMP
        elif self.metric.lower().startswith("soc."):
            device_class = DEVICE_CLASS_BATTERY
        return device_class

    @property
    def native_value(self):
        """Return the state of the sensor."""
        return self.central_system.get_metric(self.cp_id, self.metric)

    @property
    def native_unit_of_measurement(self):
        """Return the native unit of measurement."""
        return self.central_system.get_ha_unit(self.cp_id, self.metric)

    async def async_update(self):
        """Get the latest data and update the states."""
        pass
```

lbbrhzn, 2021-11-22:
my bad... sorry!
There were some unexpected complications because the current released version of home assistant does not yet support the frequency device class. I think it will be added in 2012.12.0. Meanwhile I've made a workaround.

closed 2021-11-22 by lbbrhzn