lbbrhzn/ocpp #210: Any chance we can get OpenEVSE supported?

omriasta, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I set this up and then connected OpenEVSE to this OCPP server, the OpenEVSE shows as connected but none of the sensors or switches work
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Not sure how to report exactly what the issue is, I'll post the errors I received in HA down below, any guidance on what info is needed or steps to troubleshoot would be helpful
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
More info on OpenEVSE OCPP implementation can be found here:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/182
https://github.com/matth-x/ArduinoOcpp

```
This error originated from a custom integration.

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:570
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 3:43:26 PM (9 occurrences)
Last logged: 4:17:59 PM

Failed with response: Rejected
```

```
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/ocpp/api.py:662
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 3:43:20 PM (10 occurrences)
Last logged: 4:17:58 PM

[140625518845472] list index out of range
[140625164988720] list index out of range
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 667, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 863, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 704, in _handle_entity_call
    await result
  File "/config/custom_components/ocpp/switch.py", line 70, in async_turn_on
    self._state = await self.central_system.set_charger_state(
  File "/config/custom_components/ocpp/api.py", line 240, in set_charger_state
    resp = await self.charge_points[cp_id].start_transaction()
  File "/config/custom_components/ocpp/api.py", line 548, in start_transaction
    resp = await self.get_configuration(ckey.authorize_remote_tx_requests.value)
  File "/config/custom_components/ocpp/api.py", line 662, in get_configuration
    value = resp.configuration_key[0][om.value.value]
IndexError: list index out of range
```
```
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/ocpp/api.py:717
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 3:43:12 PM (23 occurrences)
Last logged: 4:17:56 PM

[140625518845472] NotImplementedError: , {}
[140625164988720] NotImplementedError: , {}
[... 73 lines cut ...]

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 184, in on_connect
    await charge_point.reconnect(websocket)
  File "/config/custom_components/ocpp/api.py", line 745, in reconnect
    await super().start()
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 128, in start
    await self.route_message(message)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 146, in route_message
    await self._handle_call(msg)
  File "/config/custom_components/ocpp/api.py", line 723, in _handle_call
    await super()._handle_call(msg)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 170, in _handle_call
    validate_payload(msg, self._ocpp_version)
  File "/usr/local/lib/python3.9/site-packages/ocpp/messages.py", line 194, in validate_payload
    raise ValidationError(f"Payload '{message.payload} for action "
ocpp.exceptions.ValidationError: Payload '{'connectorId': 1, 'meterValue': [{'timestamp': '2021-11-22T20:38:32.000Z', 'sampledValue': [{'value': 604.213, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2021-11-22T20:39:32.000Z', 'sampledValue': [{'value': 604.213, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2021-11-22T20:40:32.000Z', 'sampledValue': [{'value': 604.213, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2021-11-22T20:41:32.000Z', 'sampledValue': [{'value': 604.213, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}]} for action 'MeterValues' is not valid: 604.213 is not of type 'string'

Failed validating 'type' in schema['properties']['meterValue']['items']['properties']['sampledValue']['items']['properties']['value']:
    {'type': 'string'}

On instance['meterValue'][0]['sampledValue'][0]['value']:
    604.213

```

drc38, 2021-11-22:
Not sure about the others but for the last error you will need to raise it with the OpenEVSE team as the charger is not conforming to the OCPP standard. 
`Exception occurred: Payload '{'connectorId': 1, 'meterValue': [{'timestamp': '2021-11-22T21:11:49.000Z', 'sampledValue': [{'value': 604.213,...`
The value returned needs to be a `String` 
[image]

drc38, 2021-11-22:
The 3rd issue also points to an issue with the OpenEVSE charger as it is returning "Not Implemented" for the "Change Availability" function which is a "Core" function in OCPP - so it is required to be supported for a charger to be OCPP compliant.
[image]

lbbrhzn, 2022-02-04:
Closed, due to inactivity.

matth-x, 2022-02-23:
Hi, I'm the developer of the OCPP library used by OpenEVSE and I stumbled over this thread as another Home Assistant user has reported a similar issue to me.

Since the error report here, a lot has happened with the OpenEVSE OCPP integration. You can find the most recent version in [my OpenEVSE fork](https://github.com/matth-x/ESP32_WiFi_V4.x/tree/master). If you prefer a compiled version, here is a snapshot: [firmware.zip](https://github.com/lbbrhzn/ocpp/files/8125101/firmware.zip) (note that it's zipped because GitHub requires it so)

It would be a great help if someone can test it with Home Assistant and report the results in this issue. Unfortunately, I haven't managed to try Home Assistant yet, though I heard so many positive things and think it's definitely time to start using it!

lbbrhzn, 2022-02-23:
@omriasta can you check whether the most recent version of OpenEVSE still has issues?

lbbrhzn, 2022-02-23:
Based on the log it seems OpenOVSE is still not OCPP 1.6 compliant.

Can you try with the option 'skip validation' enabled?

omriasta, 2022-02-23:
Sorry, I only tested with the original firmware, not your fork as per @lbbrhzn 
I am not home so don't want to flash my firmware since I will probably lose connectivity....

matth-x, 2022-02-23:
@omriasta Okay, then I'm looking forward to the test result when it's possible again.

drc38, 2022-02-23:
Hi @matth-x, the main issue preventing OpenEVSE from working with the MobilityHouse ocpp library we use in the HA occp integration is the energy and power sampled values are not being sent as a `string`. I had a quick look at your repo but couldn't spot where they are being cast from `float` to `string`?

matth-x, 2022-02-24:
@drc38 Thank you for the test! Looks like I've actually sent the old firmware file ...

This is the new FW file. To verify that the installation succeeded, I've changed the FW version from "4.1.1" into "preview4.1.1" (will be printed on the screen and on the FW tile in the dashboard): [firmware.zip](https://github.com/lbbrhzn/ocpp/files/8130824/firmware.zip)

drc38, 2022-02-24:
> @drc38 Thank you for the test! Looks like I've actually sent the old firmware file ...
I haven't tested your firmware as I don't have an OpenEVSE charger, just had a quick scan of your ArduinoOcpp code. Have you made changes to it so it casts those two variables?

matth-x, 2022-02-24:
@drc38 Finally I got a Home Assistant instance and checked out the OCPP integration. Nice work!

The main issue was that the validator of SteVe is more "tolerant" and allows wrong data types if the value is semantically correct. Good to have my library checked against a stricter validator, there were actually a few other mistakes.

Now, I get a new error message. This pattern repeats with a WebSocket reconnect in between. Do you have an idea how to proceed from here?
```
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"e2f47b55-4014-4d7d-b77e-a68dbf7581c5","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"e2f47b55-4014-4d7d-b77e-a68dbf7581c5",{"configurationKey":[],"unknownKey":["SupportedFeatureProfiles"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"98ada1c9-2768-4e47-9a90-5546d589eb9b","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"98ada1c9-2768-4e47-9a90-5546d589eb9b",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"2"}]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"1b3836e0-2fde-46af-b913-2e59fb175b2f","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"1b3836e0-2fde-46af-b913-2e59fb175b2f",{"status":"Accepted"}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"304cb872-7c9c-40ed-b230-6d991b616e48","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"304cb872-7c9c-40ed-b230-6d991b616e48",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"64a7eec1-0a4d-40a8-912c-f3844507154e","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"64a7eec1-0a4d-40a8-912c-f3844507154e",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"7f080db9-9b7a-4400-b45d-39d712c441aa","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"7f080db9-9b7a-4400-b45d-39d712c441aa",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import"}]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"f29a3a8d-33b9-45ba-9540-5a61445d16b6","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"f29a3a8d-33b9-45ba-9540-5a61445d16b6",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: send [2,"ab723c1a-5905-4093-a9a8-2cbfe5a2c24a","GetConfiguration",{"key":["ClockAlignedDataInterval"]}]
2022-02-24 14:03:01 INFO (MainThread) [ocpp] charger: receive message [3,"ab723c1a-5905-4093-a9a8-2cbfe5a2c24a",{"configurationKey":[{"key":"ClockAlignedDataInterval","readonly":false,"value":"900"}]}]
2022-02-24 14:03:01 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to 'charger': 'argument of type 'int' is not iterable'
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 880, in run
    await asyncio.gather(*self.tasks)
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 435, in post_connect
    if prof.SMART in self._attr_supported_features:
TypeError: argument of type 'int' is not iterable
2022-02-24 14:03:08 INFO (MainThread) [ocpp] charger: send [2,"e0037058-8db0-4d44-aa6f-866eac414d07","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
[...]
```

Btw. if you have by chance an ESP at hand and PIO, you can reproduce this behavior if you want. In the develop branch, the `ESP` example initiates an idle connection and then only reacts on the Central System requests. In `main.cpp`, the `STASSID` and `STAPSK` need to be configured. The following OCPP credentials worked for me:

```cpp
#define OCPP_HOST "192.168.Y.Z"
#define OCPP_PORT 9000
#define OCPP_URL "ws://192.168.Y.Z:9000/charger"
```

lbbrhzn, 2022-02-24:
Looks like the ocpp integration has an issue when the value of SupportedFeatureProfiles is not defined.
When an exception occurs it closed the websocket connection and start over.

drc38, 2022-02-24:
Looking at your wiki the charger should respond with "Core,FirmwareManagement,SmartCharging,RemoteTrigger" to show it supports these feature profiles.

omriasta, 2022-02-24:
@matth-x there is quite a bit of back and forth and I prefer not to flash new firmware to my openevse (it's very cold here and the charger is mounted outside and have had a few experiences where I had to open it up and flash using a serial connection and prefer not to do this in the cold). If you get closer and want me to run a test I'll give it a shot though. Keep me posted.

matth-x, 2022-02-25:
@lbbrhzn @drc38 @omriasta Thank you so much for your input! It seems that the error messages originates in the Python OCPP package. I will take a deeper look into it and post an update here as soon as the integration works or if I need further feedback.

drc38, 2022-04-22:
@matth-x interested to hear how you've got on. Is it now working with our integration ok?

closed 2022-06-30 by github-actions[bot]

matth-x, 2023-06-28:
@lbbrhzn Hi, I would like to follow up on this. It would be great to do a guided compatibility test. I'm curious about the scenarios which are challenging to charge points in your experience and of course, the final steps to reach full compatibility of the OpenEVSE OCPP integration. Are you open to that? If yes, I would be happy if you reach out ✉️ : matthias A⊤ arduino-ocpp DО⊤ com