lbbrhzn/ocpp #211: not working with EN-Plustech AC7000-AE-25

killee, 2021-11-24:
Hi,

thanks for the great work!

I'll try to get the AC7000-AE-25 to work with the integration.
There are a view things to solve...
The WB seems to supports no websocket ping. After disabling the connection is stable:

        server = await websockets.server.serve(
            self.on_connect,
            self.host,
            self.port,
            subprotocols=[self.subprotocol],
            ping_interval=None,
            ping_timeout=None,
            close_timeout=10,
        )

Perhaps you can add the possibility to deactivate the websocket ping in the configurations.

Another think is that the WB sends some errorCode after a while:
2021-11-24 15:42:24 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"10","StatusNotification",{"connectorId":1,"status":"Faulted","errorCode":"OtherError","timestamp":"2021-11-24T14:42:22Z","vendorId":"EN+","info":"{\\"reason\\":\\"faults\\",\\"cpv\\":0,\\"rv\\":0}","vendorErrorCode":"00001000000000000000000000000000"}]', rsv1=False, rsv2=False, rsv3=False)
2021-11-24 15:42:24 INFO (MainThread) [ocpp] SN10052108183214: receive message [2,"10","StatusNotification",{"connectorId":1,"status":"Faulted","errorCode":"OtherError","timestamp":"2021-11-24T14:42:22Z","vendorId":"EN+","info":"{\"reason\":\"faults\",\"cpv\":0,\"rv\":0}","vendorErrorCode":"00001000000000000000000000000000"}]

I've contact the support of the WB. I'll post the answer of the support.

Regards
Kilian

drc38, 2021-11-24:
Hi Kilian, does the charger respond with a pong when the `ping_timeout` is set? Details on debugging/logging the websocket can be found here websockets.readthedocs.io/en/stable/
If it does then disabling the ping is simply masking an unstable connection between the server and charger. It may also have something to do with the fault the charger is throwing. Keep us posted. Cheers

killee, 2021-11-24:
The network is stable, the WP is very close to the AP.
I've checkt it again, enabling the ping, no pong and also the error code...

drc38, 2021-11-25:
A pong response is required to be compliant with the websocket standard https://datatracker.ietf.org/doc/html/rfc6455#section-5.5  let us know how you get on with the charger manufacturer.

mterwoord, 2021-12-28:
Sorry for hijacking this thread. I have the same WB, and can't find any documentation on how to configure an OCPP endpoint on it. @killee Maybe you can help me out?

killee, 2021-12-28:
Hi @mterwoord 

I've got documentation in German from the supplier.
It's a bit of ...

I'll send you the documentation with a private message.

mterwoord, 2021-12-28:
I cannot send you private messages. Github should let you contact me via direct message.

killee, 2021-12-29:
@mterwoord I've send you an mail.

lbbrhzn, 2022-02-04:
@killee , the latest release supports measurement of ping and pong times, and configuration options for timeout.
Hope that helps.

closed 2022-02-04 by lbbrhzn

jeffborg, 2023-12-06:
@killee sorry to reopen this issue, but I have this charger branded as another brand and I have had to use router tricks to get it to connect to my own ocpp server, does the documentation you have on this say how to configure this for your own ocpp server? Thanks