lbbrhzn/ocpp #212: NRGKICK 2 (KfW)

thecem, 2021-11-28:
## Version of the custom_component
v0.3.21

## Configuration
Configured like in the repo description:

charge point Identity: NRGKICK

```yaml
charge point Identity: NRGKICK
```

## Describe the bug

The NRGKICK could not connected. Is the config right? 

## Debug log

```text

Add your logs here.

```

lbbrhzn, 2021-11-28:
It appears your charger did not contact the OCPP central server. 

- You may need to reset you charger before it switches to the new OCPP server.
- Is the charger on the same network as your home assistant server?

thecem, 2021-11-28:
Same network, access to ocpp server in HAis possible.
I tried to sniff the packets from the NRGKICK no response.
I captured some traffic between the app and the NRGKICK via Charles, seems to be ws:// too.
But was never able to get a try (packet) from NRGKICK to the OCPP Server.
Seems to be that there is some issue with the firmware, I try to get some information from the vendor Dinitech.at.

Actually installed:

[image]

thecem, 2021-11-29:
captured some packets:
[nrg21.zip](https://github.com/lbbrhzn/ocpp/files/7617763/nrg21.zip)

dreimer1986, 2021-12-13:
Just a idea on my side. As I NEED the last / on my wallbox, did you try ws://192.168.176.100:9000/ ?

thecem, 2021-12-13:
I tried, but even with / no result. Thanks a lot..

lbbrhzn, 2021-12-27:
did you try withouth ws:// ?

bardahlm, 2021-12-27:
The packet dump show traffic from HA to the web port (port 80) on the charging station, there is no OCPP traffic in the capture.

```
1	0.000000	192.168.76.100	192.168.76.151	TCP	74	36938 → 80 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=126991611 TSecr=0 WS=128
2	0.024789	192.168.76.151	192.168.76.100	TCP	60	80 → 36938 [RST, ACK] Seq=1 Ack=1 Win=28694 Len=0
```

thecem, 2022-01-21:
In between there is some fw update:

[image]

lbbrhzn, 2022-02-04:
This seems to be a charger issue, not related to the integration. Closing...

closed 2022-02-04 by lbbrhzn

KETRF, 2024-08-06:
Hi,

I wanted to share an alternative software that might be more suitable for this application and is compatible with the current software version of the NRGkick. You can check out the OCPP.Core project, which is an OCPP (Open ChargePoint Protocol) server written in .NET 8. It includes a management Web-UI for administration of charge points and charge tokens (RFID-Token).

You can find more information and the software at the following link: [OCPP.Core GitHub Repository]
(https://github.com/dallmann-consulting/OCPP.Core?tab=readme-ov-file#status).

Best regards,