lbbrhzn/ocpp #213: charge_box_serial_number on Siemens EVlink Smart Wallbox is too long

bardahlm, 2021-11-28:
## Version of the custom_component

0.3.21

## Configuration

```yaml
[...]
logger:
  default: info
  logs:
    custom_components.hacs: debug
    queueman: debug
    aiogithubapi: debug
    custom_components.ocpp: debug
    websocket: debug

```

## Describe the bug

The charge_box_serial_number on my Siemens EVlink Smart Wallbox is too long, so the BootNotification message fails validation.

## Debug log

```text

ocpp.exceptions.ValidationError: Payload '{'chargeBoxSerialNumber': 'EVB1A22P2RI3N182341200100350795E5', 'chargePointModel': 'EVlink Smart Wallbox', 'chargePointSerialNumber': '3N18310641B55', 'chargePointVendor': 'Schneider Electric', 'firmwareVersion': '3.4.0.2'} for action 'BootNotification' is not valid: 'EVB1A22P2RI3N182341200100350795E5' is too long
```

## Workaround 1

Modify ocpp/v16/schemas/BootNotification.json
```
        "chargeBoxSerialNumber": {
            "type": "string",
            "maxLength": 64
        },
```

## Workaround and possibly fix:

See also https://github.com/mobilityhouse/ocpp/issues/273

Modify line 905 in custom_components/ocpp/api.py:

```
@on(Action.BootNotification)
```

to 

```
@on(Action.BootNotification, skip_schema_validation=True)
```

lbbrhzn, 2021-11-28:
The OCPP standard specifies chargeBoxSerialNumber to be a CiString25Type. This means your charge point is not OCPP compliant. Also, chargeBoxSerialNumber is marked as deprecated, chargePointSerialNumber should be used instead. This is what we do in our custom integration. Note that chargePointSerialNumber is also present in your boot notification, but does comply to the limit of 25 characters. 

Workaround 1 is upstream, not under our control. 

Workaround 2 will completely disable validation for the boot notification, which opens the door to a  much wider range of non-compliant boot notifications. I think we should continue to do schema validation default. We could consider adding a config option to optionally skip the schema validation. However, such a feature it should also be tested, which is not that trivial, so there is some effort involved. I would rather put this effort in new features that enhance the experience for compliant charge points than making workarounds for non-compliant ones, so this is low priority for me.

Please check with your vendor (Siemens), and ask them to update their firmware. They are a reputable company, so they should fix this. Meanwhile, I recommend you fork our repo and add the workaround yourself.

bardahlm, 2021-11-29:
I mistook this to be a Siemens box, it is made by Schneider Electric. But they also count as a reputable company,

HWiese1980, 2021-12-07:
I find it disturbing that a box claims to support OCPP 1.6 and then violates the schema. It's true that as a developer of a downstream component you shouldn't have to worry about what's happening upstream because you have a schema and everything that violates that schema simply isn't your problem. The point is, you're losing people by this. Me for example (I also have an EVLink Smart WallBox by Schneider Electric). Or the OP. We're pretty much powerless unless we decided to use our own fork of your repo and work around the issue ourselves. We can request a fix from Schneider Electric but we all know how such requests usually end: by falling on deaf ears. "Dear HWiese1980, Thank you for your inquiry. I've forwarded your message to our devs. Blah blah..." never to be heard of again. 

My experience tells me, as long as we're dealing with consumer products from big companies, we have to work around every fuck-up on their side because they won't fix it. If we're very, very lucky it's still taking months, sometimes years. And especially for something as hard-wired as a serial number, we cannot even expect them to change it at all. 

So there are just two options: work around it or ignore it. I'd be really happy if we could choose option 1 because option 2 would mean, I and other owners of any affected box, not just from SE, just couldn't use your library without implementing each their own workarounds. 

My suggestion: treat it as an exception, a true, well-known workaround with all its implications, and let the user decide whether to accept any issues arising from switching on thw workaround or not, maybe put a big yellow warning on it or something. Just please don't go "Nope, sorry guys, it's not standard! Period!"

drc38, 2022-01-03:
@bardahlm other than the boot exception in the log I take it the integration stills functions?

bardahlm, 2022-01-03:
@drc38 No, it fails unless I patch the api.py file. I've made a script to patch it everytime I upgrade the integration.

Patch:
```
--- old/api.py  2021-12-28 17:32:42.113627292 +0100
+++ new/api.py  2021-12-28 17:33:10.448855525 +0100
@@ -951,7 +951,7 @@
         self.hass.async_create_task(self.central.update(self.central.cpid))
         return call_result.MeterValuesPayload()

-    @on(Action.BootNotification)
+    @on(Action.BootNotification, skip_schema_validation=True)
     def on_boot_notification(self, **kwargs):
         """Handle a boot notification."""
         resp = call_result.BootNotificationPayload(

```
Script:

```
cd custom_components/ocpp/
patch --verbose -u -l -p1  </path/to/patch/fix-ocpp.patch
```

bardahlm, 2022-01-04:
I just read the release notes for 3400-3 and stumbled across this:

* New serial number truncate in Boot Notification message to avoid doubloon;

Maybe it helps to  open cases with them?

lbbrhzn, 2022-02-04:
There's now a release v0.4.17 that allows the user to skip schema validation on the gui. Closing...

closed 2022-02-04 by lbbrhzn