lbbrhzn/ocpp #220: 2021.12.0 Beta deprecation warning

dreimer1986, 2021-12-06:
## Version of the custom_component

0.3.21

## Configuration

```yaml
            {
                "entry_id": "fe8d66744ea7798573a91ffdfb389031",
                "version": 1,
                "domain": "ocpp",
                "title": "central",
                "data": {
                    "host": "0.0.0.0",
                    "port": 9000,
                    "csid": "central",
                    "cpid": "evbox",
                    "meter_interval": 60,
                    "monitored_variables": ""
                },
                "options": {},
                "pref_disable_new_entities": false,
                "pref_disable_polling": false,
                "source": "user",
                "unique_id": null,
                "disabled_by": null
            },
```

## Describe the bug

2021.12.0 Beta reveals some new information on what Home Assistant want's to see coming from the addin and what will be deprecated soon. As the log entry asks to do, I inform you about the new information. I only get one warning right now from sensor.evbox_timestamp_config_response, but very likely sensor.evbox_timestamp_data_response will result in the same warning. No hurry though. Seems like the deprecation will not come before the February release.

## Debug log

```text

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:376
Integration: Sensor (documentation, issues)
First occurred: 15:50:25 (1 occurrences)
Last logged: 15:50:25

sensor.evbox_timestamp_config_response is providing a string for its state, while the device class is 'timestamp', this is not valid and will be unsupported from Home Assistant 2022.2. Please report it to the custom component author.

```

lbbrhzn, 2021-12-06:
Thanks for reporting!

I guess we need to convert the string to a proper timestamp, should not be difficult to do.

lbbrhzn, 2021-12-27:
solved!

closed 2021-12-27 by lbbrhzn