lbbrhzn/ocpp #221: EVBox ELVI: Setting Maximum_Current causes Rejected / Rejected while setting MeterValuesSampledData to ...

dreimer1986, 2021-12-06:
## Version of the custom_component

0.3.21

## Configuration

```yaml
            {
                "entry_id": "fe8d66744ea7798573a91ffdfb389031",
                "version": 1,
                "domain": "ocpp",
                "title": "central",
                "data": {
                    "host": "0.0.0.0",
                    "port": 9000,
                    "csid": "central",
                    "cpid": "evbox",
                    "meter_interval": 60,
                    "monitored_variables": ""
                },
                "options": {},
                "pref_disable_new_entities": false,
                "pref_disable_polling": false,
                "source": "user",
                "unique_id": null,
                "disabled_by": null
            },
```

## Describe the bug

Beforehand I have to say that I LOVE the work you all realized here. After the long time I lurk around here it became rock solid and shows all information I can get from the limited Wallbox I have. Thanks very much!

But sooner or later you will start to hate my Wallbox, that for sure. ^^ I kept the following a while by now and out of pure luck realized that it could be a problem I have with my Wallbox and getting information into my Home Assistant by the OCPP addin.

Some history beforehand. I wondered all the time why my 11 kWh Wallbox only charges with about 4 kWh. Now I found out why. I can set a minimum charge current (default 6 A -> ~4 kWh) and a maximum charge current (default 16 A -> ~11 kWh) in the App of the Wallbox. When I connect the box to the Everon service EVBOX provides it seems to happlily charge with full speed. When it's connected to the OCPP addin it seems to stay on 6 A aka the lowest setting. I can force the box higher by increasing the minimum setting in App, but that was not needed on Everon. So I thought, the addin should support setting the current as the Wallbox should have that feature. I tried the Maximum_Current bar selector while charging and moved it to 16 A aka a legit value. "Rejected" Then for fun I tried 10 A (at that time there still was a 6 A Minimum set up in App.) "Rejected". Setting the Maximum_Current to ANY value causes a "Rejected" answer.  See log Part 1. Maybe something to do with the earlier configuration problem "Rejected while setting MeterValuesSampledData to" And yes, after the ...to there comes nothing in log. See Log Part 2.

## Debug log

```text
2021-12-06 20:07:15 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x80\x19\x99X', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:15 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2021-12-06 20:07:15 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x80\x19\x99X', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:15 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 80199958
2021-12-06 20:07:21 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<248 bytes>)
2021-12-06 20:07:21 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"29411be9-df13-451d-b782-469b63477702","MeterValues",{"connectorId":1,"transactionId":1638815636,"meterValue":[{"timestamp":"2021-12-06T19:07:21Z","sampledValue":[{"value":"0","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:21 INFO (MainThread) [ocpp] EVB-P20482857: receive message [2,"29411be9-df13-451d-b782-469b63477702","MeterValues",{"connectorId":1,"transactionId":1638815636,"meterValue":[{"timestamp":"2021-12-06T19:07:21Z","sampledValue":[{"value":"0","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]
2021-12-06 20:07:21 INFO (MainThread) [ocpp] EVB-P20482857: send [3,"29411be9-df13-451d-b782-469b63477702",{}]
2021-12-06 20:07:21 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"29411be9-df13-451d-b782-469b63477702",{}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:26 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<296 bytes>)
2021-12-06 20:07:26 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"29411bee-df13-451d-b782-469b7c1aed1b","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Charging,NoError,No error.,1,Blue,1,{20,8005,26},{0,0},{C,11902,5666,11896},0,0,250,2021-12-06T19:07:21Z,1638815636,3,{0,0,0,0,0,0,0,0,0},190,2,0,112,0,250,160,0"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:26 INFO (MainThread) [ocpp] EVB-P20482857: receive message [2,"29411bee-df13-451d-b782-469b7c1aed1b","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Charging,NoError,No error.,1,Blue,1,{20,8005,26},{0,0},{C,11902,5666,11896},0,0,250,2021-12-06T19:07:21Z,1638815636,3,{0,0,0,0,0,0,0,0,0},190,2,0,112,0,250,160,0"}]
2021-12-06 20:07:26 DEBUG (MainThread) [custom_components.ocpp] Datatransfer received from EVB-P20482857: {'message_id': 'evbStatusNotification', 'data': '1,Charging,NoError,No error.,1,Blue,1,{20,8005,26},{0,0},{C,11902,5666,11896},0,0,250,2021-12-06T19:07:21Z,1638815636,3,{0,0,0,0,0,0,0,0,0},190,2,0,112,0,250,160,0'}
2021-12-06 20:07:26 INFO (MainThread) [ocpp] EVB-P20482857: send [3,"29411bee-df13-451d-b782-469b7c1aed1b",{"status":"Accepted"}]
2021-12-06 20:07:26 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"29411bee-df13-451d-b782-469b7c1aed1b",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:35 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xf6\xe6\xe8\x9e', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:35 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2021-12-06 20:07:35 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xf6\xe6\xe8\x9e', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:35 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: f6e6e89e
2021-12-06 20:07:55 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x1b\x00\xa9-', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:55 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2021-12-06 20:07:55 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x1b\x00\xa9-', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:07:55 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 1b00a92d
2021-12-06 20:08:15 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'Z\xaa\xd9\x0b', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:08:15 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2021-12-06 20:08:15 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'Z\xaa\xd9\x0b', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:08:15 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 5aaad90b
2021-12-06 20:08:16 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"8627fab5-0072-4da8-ba35-58b54434c84d","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2021-12-06 20:08:16 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"8627fab5-0072-4da8-ba35-58b54434c84d","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 20:08:16 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<163 bytes>)
[... 100 lines cut ...]
2021-12-06 19:51:57 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"b30bad4e-41df-48dc-87f6-cae96b93cd12",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"}]}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 19:51:57 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"b30bad4e-41df-48dc-87f6-cae96b93cd12",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"}]}]
[... cut ...]
2021-12-06 19:51:58 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"58cf0de2-fd9e-4a27-a790-c013003600c9",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-06 19:51:58 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"58cf0de2-fd9e-4a27-a790-c013003600c9",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2021-12-06 19:51:58 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
```

drc38, 2021-12-06:
Does the charger provide a more detailed debug as to why the `SetChargingProfile` message is being rejected? You could try reducing the `stackLevel` in case it needs to be <6.

dreimer1986, 2021-12-07:
Maybe if I find out how to let it send more information over the service ocpp.get_diagnistics. Otherwise it does not give much information I fear...

Where is the StackLevel defined?

drc38, 2021-12-07:
In `api.py` you could try changing to `om.stack_level.value: stack_level -1`

drc38, 2021-12-07:
Or can you sniff message from App to charger to compare?

dreimer1986, 2021-12-07:
Not really. The App communicates via Bluetooth. It's more a way to limit the borders of whatever way of load management is being used can tinker in. The app normally is for electricians setting up the box offline and online. (Still the only way to set a custom OCPP.) In the long horror report (#2) where we slowly but steadily got the box to behave I put a few network sniffs when the box communicates with Everon. Are these useful maybe? https://github.com/lbbrhzn/ocpp/issues/2#issuecomment-863425690

Regarding the -1: I need to charge for that to work at all I guess? If not then we are still on Rejected, if yes, will try later at home after work.

lbbrhzn, 2021-12-07:
On IOS you can sniff bluetooth. See https://mezdanak.de/2019/07/12/ios-bluetooth-packet-logging/. 
Might be something similar for Android as well.

dreimer1986, 2021-12-07:
Indeed there is... at least a HCI Log is possible. Gonna try later too. No clue if that helps at all as this is no OCPP connection, but we will see.

dreimer1986, 2021-12-07:
[btsnoop_hci.zip](https://github.com/lbbrhzn/ocpp/files/7670774/btsnoop_hci.zip)

BT Log. The device you are looking for is: u-blox_27:d8:f4

Regarding the charging active and om.stack_level.value: stack_level -1:

```
2021-12-07 19:37:23 INFO (MainThread) [custom_components.ocpp] {'host': '0.0.0.0', 'port': 9000, 'csid': 'central', 'cpid': 'evbox', 'meter_interval': 60, 'monitored_variables': ''}
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.protocol] server - state = CONNECTING
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.protocol] server - event = connection_made(<_SelectorSocketTransport fd=11 read=idle write=<idle, bufsize=0>>)
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<230 bytes>)
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.server] server < GET http://192.168.181.42:9000/EVB-P20482857 HTTP/1.1
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.server] server < Headers([('Host', '192.168.181.42:9000'), ('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Key', 'KUJmbd8TRh23gkabrOfwBA=='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Sec-WebSocket-Version', '13')])
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.server] server > HTTP/1.1 101 Switching Protocols
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.server] server > Headers([('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Accept', 'CM1bZBuog0B2jut32b2W5dqjTr8='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Date', 'Tue, 07 Dec 2021 18:37:33 GMT'), ('Server', 'Python/3.9 websockets/9.1')])
2021-12-07 19:37:33 DEBUG (MainThread) [websockets.protocol] server - state = OPEN
2021-12-07 19:37:33 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2021-12-07 19:37:33 INFO (MainThread) [custom_components.ocpp] Charger websocket path=http://192.168.181.42:9000/EVB-P20482857
2021-12-07 19:37:33 INFO (MainThread) [custom_components.ocpp] Charger EVB-P20482857 connected to 0.0.0.0:9000.
2021-12-07 19:37:35 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<222 bytes>)
2021-12-07 19:37:35 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"9d27ec91-7a00-4147-85cf-2f1c5a1af120",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"9d27ec91-7a00-4147-85cf-2f1c5a1af120",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]
2021-12-07 19:37:37 INFO (MainThread) [custom_components.ocpp] Supported feature profiles: Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"1a65390c-d0aa-49de-a0af-3774a93bd0ca","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"1a65390c-d0aa-49de-a0af-3774a93bd0ca","TriggerMessage",{"requestedMessage":"StatusNotification"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<70 bytes>)
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"1a65390c-d0aa-49de-a0af-3774a93bd0ca",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"1a65390c-d0aa-49de-a0af-3774a93bd0ca",{"status":"Accepted"}]
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"2392986c-ee4e-466e-a4e9-98726e638873","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"2392986c-ee4e-466e-a4e9-98726e638873","ChangeAvailability",{"connectorId":0,"type":"Operative"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<237 bytes>)
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"2392986c-ee4e-466e-a4e9-98726e638873",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"29426671-df13-461d-b782-469bf966371f","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError","timestamp":"2021-12-07T18:37:37Z"}]', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"2392986c-ee4e-466e-a4e9-98726e638873",{"status":"Accepted"}]
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: receive message [2,"29426671-df13-461d-b782-469bf966371f","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError","timestamp":"2021-12-07T18:37:37Z"}]
2021-12-07 19:37:37 INFO (MainThread) [ocpp] EVB-P20482857: send [3,"29426671-df13-461d-b782-469bf966371f",{}]
[... cut ...]
2021-12-07 19:39:13 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'6\xc1\xed\xa8', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:39:13 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2021-12-07 19:39:13 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'6\xc1\xed\xa8', rsv1=False, rsv2=False, rsv3=False)
2021-12-07 19:39:13 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 36c1eda8
```

drc38, 2021-12-07:
I'm afraid without some detail from the charger on why the message is being rejected you're limited to trial and error changes to:
`"SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":5,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":15.0}]}}}]`
An incorrect (too high) `stackLevel` initially stopped my charger from accepting the charging profile, hence the suggestion. 

If diagnostics or the manufacturer won't help some other things you could try:
- set a power limit rather than current limit
- clear the charging profile before changing limit (HA service can be used) 

Good luck getting to the bottom of it.

drc38, 2021-12-08:
The readonly just means it is not a configuration key you can change. So to get diagnostics you'll have to use an ftp server rather than http.

drc38, 2021-12-08:
It would be worth trying the get diagnostics service to see if it provides a more detailed error log

dreimer1986, 2021-12-08:
So, I added all parameters I think are interesting. I guess I just need a anonymous FTP running for that?

drc38, 2021-12-08:
I guess so, mine only uses http so can't help much I'm afraid.

dreimer1986, 2021-12-08:
Even easier: I used the official FTP addin from Home Assistant and enabled uploading etcetc for one user and then used this on the diagnostic data service: ftp://hassio:hassio@192.168.181.42:21/config/ (last / is important!)

I can verify that it's a rolling log file keeping one hour of data in csv format. As the last try is from yesterday I will do the real thing later at home, but for the ones curious, here a log without the events, but with all the settings:

[EVB-P20482857_211208_121633.csv](https://github.com/lbbrhzn/ocpp/files/7675951/EVB-P20482857_211208_121633.csv)

Real one follows later. :D

Vanhouttethomas, 2021-12-12:
Hi
The documentation states the EVBox Elvi is not supported (yet).
How were you able to connect it?
I own an Elvi and was wondering if I could provide some logs so that the developers could investigate further how to support this charger.

dreimer1986, 2021-12-13:
It's sort of supported by now. If you wanna connect it, open the EVBox Connect App and in the Advanced settings set a custom server instead of the Everon Default setting. ws://IP.OF.HA:9000/ is the way to go. Dont forget the / at the end.

Vanhouttethomas, 2021-12-13:
Hi @dreimer1986 
Thanks for the followup of my question.

Could you point me in the right direction where exactly I should define my the server URL of my HA?

These are the steps I take:

- [1] Open the EVBOX connect app (latest app version on latest stable iOS)
- [2] Open ‘Installer Mode’ (access via security code)
- [3] ??

I seem to be able to find the APN url, although that seems to be for the mobile data connection and not what I need.

thanks for your help!

Screenshot of the options in the Installer Mode:
[image]

dreimer1986, 2021-12-13:
https://user-images.githubusercontent.com/29841368/130663735-3808bcd7-8a78-47b9-a52a-1ebb07bc9912.png

This is what I have there. You are missing the 3rd setting somehow?

Vanhouttethomas, 2021-12-13:
Hi
Well that seemed to have done the trick. 
The entities are now populated with the relevant data.

A second confirmation the Elvi is indeed supported ;-)

thanks a lot!

dreimer1986, 2021-12-13:
What happens to you if you try to change the Max Current?

Vanhouttethomas, 2021-12-15:
Hi

At home I only charge at 3,5kWh (1 phase).
the maximum current is set to 32A (I did not change this yet). When I set it to 16A, nothing happens. When checking the ‘history’, indeed I see the value stays the same at 32A.

Is there something else you want me to test?

[image]

dreimer1986, 2021-12-16:
Could you check the log entries for a Rejected coming from the addin?

dreimer1986, 2021-12-29:
Hmm, no reply, but likely the same problem anyways.

Btw, with the new notifications I get these two now:
[image]
The MeterValuesSampledData to be set to.... ?Nothing? being rejected is not new, but what data is it talking about? The logs (All debugging of OCPP and WebSocket active) says nothing in that matter on first sight.

drc38, 2021-12-31:
The MeterValuesSampledData should contain the list of measurands the charger supports. They are selected when setting up the ocpp integration. If you look at your log for MeterValues you will see what your charger sends by default.

dreimer1986, 2021-12-31:
Hmm, right you are. I added a few more checkboxes in Plugin config and now it is filled with a small list of measurands. Still rejected though ^^ 

```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:728
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 30. Dezember 2021, 04:06:03 (2 occurrences)
Last logged: 12:25:28

Rejected while setting MeterValuesSampledData to
Rejected while setting MeterValuesSampledData to Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature,Current.Export
```

drc38, 2021-12-31:
If you look at your HA log for the MeterValues message you will see what your charger sends by default (ie supports) or refer to charger manual, then only select those when setting up Ocpp.

dreimer1986, 2022-01-01:
Yes, I see the "Energy.Active.Import.Register". Still it dislikes and rejects both what was mentioned in the topic. Btw, no reply from the manufacturer yet

drc38, 2022-01-01:
> Yes, I see the "Energy.Active.Import.Register". Still it dislikes and rejects both what was mentioned in the topic. Btw, no reply from the manufacturer yet

It maybe that it only supports Energy.Active.Import.Register it definitely will not support SoC (as it is for DC chargers) or export functions unless it supports V2G.

dreimer1986, 2022-01-02:
Yes, but even when I only select this one in addin configuration, it answers with rejected and same for setting the Max Current

drc38, 2022-01-02:
While you wait for response you could use the get configuration service to check that  MeterValuesSampledData is not read only.

dreimer1986, 2022-01-02:
```
2022-01-02 16:29:42 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Running websocket_api script
2022-01-02 16:29:42 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Executing step call service
2022-01-02 16:29:42 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"679ea70b-ad59-4036-b834-96c42ae39588","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-01-02 16:29:42 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"679ea70b-ad59-4036-b834-96c42ae39588","GetConfiguration",{"key":["MeterValuesSampledData"]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-02 16:29:42 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<163 bytes>)
2022-01-02 16:29:42 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"679ea70b-ad59-4036-b834-96c42ae39588",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"}]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-02 16:29:42 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"679ea70b-ad59-4036-b834-96c42ae39588",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register"}]}]
2022-01-02 16:29:42 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for MeterValuesSampledData: Energy.Active.Import.Register
2022-01-02 16:29:56 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x9f\x9a\x8c\xc6', rsv1=False, rsv2=False, rsv3=False)
2022-01-02 16:29:56 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-02 16:29:56 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x9f\x9a\x8c\xc6', rsv1=False, rsv2=False, rsv3=False)
2022-01-02 16:29:56 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 9f9a8cc6
```

dreimer1986, 2022-01-17:
No news from the manufacturer, I think I will write another mail that weekend.
But new behavior with v0.4.6. Now the max current control is unavailable. I can't test while charging right now though (charged the car to 100% yesterday). This is in idle mode.

The "Rejected while setting MeterValuesSampledData to" error is gone though.

[image]

```
2022-01-17 15:16:18 INFO (MainThread) [custom_components.ocpp] {'host': '0.0.0.0', 'port': 9000, 'csid': 'central', 'cpid': 'evbox', 'meter_interval': 60, 'monitored_variables': 'Energy.Active.Import.Register'}
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.protocol] server - state = CONNECTING
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.protocol] server - event = connection_made(<_SelectorSocketTransport fd=11 read=idle write=<idle, bufsize=0>>)
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<230 bytes>)
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.server] server < GET http://192.168.181.42:9000/EVB-P20482857 HTTP/1.1
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.server] server < Headers([('Host', '192.168.181.42:9000'), ('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Key', 'KXg2ud8TRh23gkabdEq+Xg=='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Sec-WebSocket-Version', '13')])
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.server] server > HTTP/1.1 101 Switching Protocols
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.server] server > Headers([('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Accept', 'Xvzg4mpuxtwLfZyZ5Rw/t+4e3AY='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Date', 'Mon, 17 Jan 2022 14:16:25 GMT'), ('Server', 'Python/3.9 websockets/9.1')])
2022-01-17 15:16:25 DEBUG (MainThread) [websockets.protocol] server - state = OPEN
2022-01-17 15:16:25 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-01-17 15:16:25 INFO (MainThread) [custom_components.ocpp] Charger websocket path=http://192.168.181.42:9000/EVB-P20482857
2022-01-17 15:16:25 INFO (MainThread) [custom_components.ocpp] Charger EVB-P20482857 connected to 0.0.0.0:9000.
2022-01-17 15:16:27 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"b74685a1-a64b-4648-88e4-05ad1aa9a0a7","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-01-17 15:16:27 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"b74685a1-a64b-4648-88e4-05ad1aa9a0a7","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<222 bytes>)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"b74685a1-a64b-4648-88e4-05ad1aa9a0a7",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"b74685a1-a64b-4648-88e4-05ad1aa9a0a7",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]
2022-01-17 15:16:28 INFO (MainThread) [custom_components.ocpp] Supported feature profiles: Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"710f81c2-6c6a-4645-84f1-e3a5bac0976e","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"710f81c2-6c6a-4645-84f1-e3a5bac0976e","TriggerMessage",{"requestedMessage":"StatusNotification"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<70 bytes>)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"710f81c2-6c6a-4645-84f1-e3a5bac0976e",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"710f81c2-6c6a-4645-84f1-e3a5bac0976e",{"status":"Accepted"}]
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"1716ef9b-0fc0-49d1-8088-5e6b28405b41","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"1716ef9b-0fc0-49d1-8088-5e6b28405b41","ChangeAvailability",{"connectorId":0,"type":"Operative"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<70 bytes>)
2022-01-17 15:16:28 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"1716ef9b-0fc0-49d1-8088-5e6b28405b41",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"1716ef9b-0fc0-49d1-8088-5e6b28405b41",{"status":"Accepted"}]
2022-01-17 15:16:28 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"aefd3d7a-5e14-4def-915e-e2f811249923","GetConfiguration",{"key":["HeartbeatInterval"]}]
[... cut ...]
2022-01-17 15:17:45 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xcc\xf9\xb7\xa7', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:17:46 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 15:17:46 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xcc\xf9\xb7\xa7', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 15:17:46 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: ccf9b7a7
```

drc38, 2022-01-17:
What version of HA are you running? What dors your .Features metric report?

dreimer1986, 2022-01-17:
[home-assistant-pre-Reboot.log](https://github.com/lbbrhzn/ocpp/files/7884304/home-assistant-pre-Reboot.log)
[home-assistant-post-Reboot.log](https://github.com/lbbrhzn/ocpp/files/7884303/home-assistant-post-Reboot.log)

First these two as lil additions. This time charging (Started at around 89%) and then accidentally updating HAOS in between. Before and after there was no control button and entity unavailable.

Now your answer:

[image]

All in one picture to at least semi proof that I dont wanna ruin your monday or something. In text form for easy copy/paste: Profiles.AUTH|REM|RES|SMART|FW|CORE

drc38, 2022-01-17:
Only thing I can think of is HA has created a new number entity. Check the devices tab or developer tools to see if there are two. One should be available, the other can be deleted.

dreimer1986, 2022-01-17:
Sadly not... Thats from the full entity list sorted by integration.

[image]

[image]

drc38, 2022-01-17:
In your screenshot it is showing as available?

dreimer1986, 2022-01-17:
Not really, the red exclamation means it's not available:

https://www.youtube.com/watch?v=VLcq6veGGjU

GrantKeymer, 2022-01-17:
Good morning all, responding to your tag @drc38 

For me, this particular entity has always shown 'Unavailable', therefore the associated slide control cannot be used.  Which is a real pity because it means I have to adjust the charging current either from My Wallbox portal or the app.

Keep in mind that I've only recently got this integration working.  My first successful OCPP connection was with v 0.4.3 but so many things were broken with that release, it wasn't much use.  However, I do recall the Maximum_Current entity was unavailable then, and it has remained so with subsequent releases.

drc38, 2022-01-17:
Hard to know what the issue is with your HA environments given the issue does not occur for me,  the restore entity was added a few versions ago and may be worth looking at on the HA forums.

drc38, 2022-01-17:
> Not really, the red exclamation means it's not available:
> 
> https://www.youtube.com/watch?v=VLcq6veGGjU
> 
> Sorry for crushing every last straw here... In this case I just wanted to inform you. It did not work before and now at least does not throw any errors and shows that this is the case. If the behavior is not logical, in this way it's more to the reality than before. That for sure. But if any of the above helps to improve the situation, I would be quite happy about that. Regardting happy.... I am buying you a coffee for the messy wallbox I always go on your nerves with. ^^ EDIT: OH wait... that coffee did not find the way to you I guess? Or is the account shared with lbbrhzn?

It would have gone to @lbbrhzn as I haven't set up github sponsoring. He'll have to enjoy it instead! 😁

GrantKeymer, 2022-01-17:
> Hard to know what the issue is with your HA environments given the issue does not occur for me, the restore entity was added a few versions ago and may be worth looking at on the HA forums.

Am I right in thinking you're using a different charger @drc38 ?

It seems that @dreimer1986 is using an Elvi and I'm using a Pulsar Plus.

Therefore, could be a charger bug with both of our chargers.  My next move is to study the OCPP 1.6 Spec in more detail and understand the appropriate send/receive transactions related to this setting.

lbbrhzn, 2022-01-17:
I've been running the main branch with my Walbox Pulsar, and the slider works fine if I start from the devcontainer. 
However, it is not enabled when use it with my normal home assistant configuration! 
It might have something to do with the RestoreEntity class, that was added to the Max Charging Current entity in order to restore the value at reboot.

I recommend to remove the integration and re-add it, givoing the charger a different name. That should make sure the entity is not restored from an old version in the database. Hope it helps!

lbbrhzn, 2022-01-17:
@drc38, I read about some peculiarities of the RestoreEntity class here. https://aarongodfrey.dev/programming/restoring-an-entity-in-home-assistant/
Apparently it restores the state value, but not the attributes. Might this be the source of our troubles?

dreimer1986, 2022-01-17:
It is. I renamed it to elvi after a reinstall... Back it is. But of course it hates me as before ^^

```
2022-01-17 22:51:09 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'>\xae\x0f<', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:09 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:51:09 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'>\xae\x0f<', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:09 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 3eae0f3c
2022-01-17 22:51:29 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xedw\x15\xe7', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:29 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:51:29 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xedw\x15\xe7', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:29 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: ed7715e7
2022-01-17 22:51:49 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b']r\x9aq', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:49 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:51:49 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b']r\x9aq', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:51:49 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 5d729a71
2022-01-17 22:52:09 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xbf\xbem\x8d', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:09 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:52:09 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xbf\xbem\x8d', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:09 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: bfbe6d8d
2022-01-17 22:52:29 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x88\xe0c\xcb', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:29 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:52:29 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x88\xe0c\xcb', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:29 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 88e063cb
2022-01-17 22:52:49 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x93\x8bv\xf4', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:49 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:52:49 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x93\x8bv\xf4', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:52:49 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 938b76f4
2022-01-17 22:53:09 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x16\x96\xb8Q', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:09 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:53:09 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x16\x96\xb8Q', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:09 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 1696b851
2022-01-17 22:53:29 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x91\xb1pl', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:29 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:53:29 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x91\xb1pl', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:29 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 91b1706c
2022-01-17 22:53:49 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\x0c\x06,V', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:49 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:53:49 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\x0c\x06,V', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:53:49 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: 0c062c56
2022-01-17 22:54:09 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xb6W\xcb\x86', rsv1=False, rsv2=False, rsv3=False)
2022-01-17 22:54:09 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-17 22:54:09 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xb6W\xcb\x86', rsv1=False, rsv2=False, rsv3=False)
[... cut ...]
2022-01-17 23:02:46 DEBUG (MainThread) [websockets.protocol] server - aborted pending ping: 43c9def7
2022-01-17 23:02:46 DEBUG (MainThread) [websockets.protocol] server ! failing CLOSED WebSocket connection with code 1006
2022-01-17 23:02:46 DEBUG (MainThread) [websockets.protocol] server x half-closing TCP connection
2022-01-17 23:02:46 DEBUG (MainThread) [custom_components.ocpp] Websockets exception: code = 1006 (connection closed abnormally [internal]), no reason
2022-01-17 23:02:46 INFO (MainThread) [custom_components.ocpp] Charger EVB-P20482857 disconnected from 0.0.0.0:9000.
```

GrantKeymer, 2022-01-17:
> I've been running the main branch with my Walbox Pulsar, and the slider works fine if I start from the devcontainer. However, it is not enabled when use it with my normal home assistant [...]

I did this just now, also removing and reinstalling the OCPP repo, just to be sure.  Before adding cards to Lovelace, I changed the name from 'charger' to Wallbox Pulsar Plus and added all the cards again.  Unfortunately, nothing has changed.  The Maximum_Current control is still unavailable.

Should I be using the 'Main' version of OCPP instead of v0.4.6?

lbbrhzn, 2022-01-17:
Ive release 0.4.7 as a pre-release (you need to enable beta versions in HACS to see it). The changes should not impact the behavior of the max current control entity though, they only impact development and testing.
@GrantKeymer , I trust you did wait until your charger successfully reconnected? The entity will definitely remain unavailable until it has.

lbbrhzn, 2022-01-17:
Removing the integration, and re-adding after an upgrading to v0.4.7 did result in a working slider on my main system.
[image]

GrantKeymer, 2022-01-17:
> Removing the integration, and re-adding after an upgrading to v0.4.7 did result in a working slider on my main system.

Fantastic, thank you!  I'll try it again.

Yes, I did wait for the charger to reconnect, and I also restarted HA after removing and re-adding the integration.  But hopefully v0.4.7 will do the trick.

Sweet as Mate :D

lbbrhzn, 2022-01-17:
And when I change the slider it correctly updates  the ChargingProfile:
```text
2022-01-17 23:36:55 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2022-01-17 23:36:55 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current
2022-01-17 23:36:55 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2022-01-17 23:36:55 INFO (MainThread) [ocpp] charger: send [2,"8a04aec1-b7d7-48f8-a1eb-7080958e9a87","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2022-01-17 23:36:55 INFO (MainThread) [ocpp] charger: receive message [3,"8a04aec1-b7d7-48f8-a1eb-7080958e9a87",
{
"unknownKey": [],
"configurationKey": [
{
"readonly": true,
"key": "ChargeProfileMaxStackLevel",
"value": "0"
} 
]
}]
2022-01-17 23:36:55 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 0
2022-01-17 23:36:55 INFO (MainThread) [ocpp] charger: send [2,"4f2047e2-a653-4109-a3eb-d76746de6085","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":0,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":19.0}]}}}]
2022-01-17 23:36:55 INFO (MainThread) [ocpp] charger: receive message [3,"4f2047e2-a653-4109-a3eb-d76746de6085",
{
"status": "Accepted"
}]
```

lbbrhzn, 2022-01-17:
awesome!

lbbrhzn, 2022-01-17:
@dreimer1986 , looks a bit sus. Maybe your charger is paruku?
Anyway, my battery is chocka block, good as gold. I'm off to bed!

drc38, 2022-01-17:
@lbbrhzn pakaru = broken 😉

dreimer1986, 2022-01-18:
Likely it is not. I have not much ways to test the OCPP variant to reduce the max current, as the offical included site does not really allow that just for fun. But the difference I already witnessed is that using the official one causes the Wallbox to charge with max current by default. In the addin case I have to persuade it with the config tool. This is what I told here: https://github.com/lbbrhzn/ocpp/issues/221#issuecomment-988765666

GrantKeymer, 2022-01-18:
> Removing the integration, and re-adding after an upgrading to v0.4.7 did result in a working slider on my main system.

Unfortunately my slider is still Pakaru :(

I am definitely running v0.4.7 of the repo.  There must be something else different in my config vs. yours.

lbbrhzn, 2022-01-18:
@dreimer1986 , did you try clearing the charging profiles as suggested by @drc38?

lbbrhzn, 2022-01-18:
Another thing to try is to change the chargingProfilePurpose. According to section 5.16.3 of the OCPP spec, the chargingProfilePurpose of the ChargingProfile MUST be set to TxProfile when setting a charging profile during a transaction.

dreimer1986, 2022-01-18:
> @dreimer1986 , did you try clearing the charging profiles as suggested by @drc38?

Yes I did. More out of mistake when I verified that I have set a lower than 16A evb_MinimumChargeCurrent value and accidentally just reused the already selected ocpp.clear_profile service. But I did not tinker with chargingProfilePurpose as this is a niew one for me, too. Only thing we tried already was om.stack_level.value: stack_level -1.

drc38, 2022-01-18:
> Another thing to try is to change the chargingProfilePurpose. We're using a chargingProfilePurpose=ChargePointMaxProfile, which is meant for load balancing scenarios, but maybe your charger does not support that, or does not support setting or updating that during a transaction.
> 
> The OCPP spec on smart Charging is quite elaborate, and sometimes ambiguous:
> 
> Section 5.16.3 states: "In order to ensure that the updated charging profile applies only to the current transaction, the chargingProfilePurpose of the ChargingProfile MUST be set to TxProfile when setting a charging profile during a transaction."
> 
> Does this mean using ChargeProfile=ChargePointMaxProfile is not allowed during a transaction? That would defeat the purpose of load balancing.

My guess is because you can stack different profiles on top of each other, and for a charging session you do not know how long it will take, you should use TxProfile if the limit only applies for the duration of the session. Afterwards it will revert to the time of day limit profiles. 

We simply use ChargePointMaxProfile because it trumps all other purposes and should limit overall current to the setting until we change the setting via the slider, automation etc

dreimer1986, 2022-01-18:
> Another thing to try is to change the chargingProfilePurpose. We're using a chargingProfilePurpose=ChargePointMaxProfile, which is meant for load balancing scenarios, but maybe your charger does not [...]

OK, this sounds like a step in the right direction! In the VERY CORRECT DIRECTION!

I changed line 519 to:
```
                    om.charging_profile_purpose.value: ChargingProfilePurposeType.tx_default_profile.value,
```
as this sounded like a solution according to the specs as I understood it and now:

[image]

```
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.protocol] server - state = CONNECTING
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.protocol] server - event = connection_made(<_SelectorSocketTransport fd=35 read=idle write=<idle, bufsize=0>>)
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<230 bytes>)
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.server] server < GET http://192.168.181.42:9000/EVB-P20482857 HTTP/1.1
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.server] server < Headers([('Host', '192.168.181.42:9000'), ('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Key', 'KXk7pd8TRh23gkabjVCmGQ=='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Sec-WebSocket-Version', '13')])
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.server] server > HTTP/1.1 101 Switching Protocols
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.server] server > Headers([('Upgrade', 'websocket'), ('Connection', 'Upgrade'), ('Sec-WebSocket-Accept', 'DaI8rNIlkrZggY5VGPTmZm16nvw='), ('Sec-WebSocket-Protocol', 'ocpp1.6'), ('Date', 'Tue, 18 Jan 2022 08:49:41 GMT'), ('Server', 'Python/3.9 websockets/9.1')])
2022-01-18 09:49:41 DEBUG (MainThread) [websockets.protocol] server - state = OPEN
2022-01-18 09:49:41 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-01-18 09:49:41 INFO (MainThread) [custom_components.ocpp] Charger websocket path=http://192.168.181.42:9000/EVB-P20482857
2022-01-18 09:49:41 INFO (MainThread) [custom_components.ocpp] Charger EVB-P20482857 connected to 0.0.0.0:9000.
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"5b17d590-c100-4acc-a388-43147bb56c52","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"5b17d590-c100-4acc-a388-43147bb56c52","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<222 bytes>)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"5b17d590-c100-4acc-a388-43147bb56c52",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"5b17d590-c100-4acc-a388-43147bb56c52",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation"}]}]
2022-01-18 09:49:43 INFO (MainThread) [custom_components.ocpp] Supported feature profiles: Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger,Reservation
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"295e6f69-ce2d-47b8-a968-ae5b0ba9d2d9","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"295e6f69-ce2d-47b8-a968-ae5b0ba9d2d9","TriggerMessage",{"requestedMessage":"StatusNotification"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<70 bytes>)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"295e6f69-ce2d-47b8-a968-ae5b0ba9d2d9",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"295e6f69-ce2d-47b8-a968-ae5b0ba9d2d9",{"status":"Accepted"}]
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"0d9a3e44-5284-47cb-91c7-95a2fd85594e","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[2,"0d9a3e44-5284-47cb-91c7-95a2fd85594e","ChangeAvailability",{"connectorId":0,"type":"Operative"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<70 bytes>)
2022-01-18 09:49:43 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.TEXT: 1>, data=b'[3,"0d9a3e44-5284-47cb-91c7-95a2fd85594e",{"status":"Accepted"}]', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: receive message [3,"0d9a3e44-5284-47cb-91c7-95a2fd85594e",{"status":"Accepted"}]
2022-01-18 09:49:43 INFO (MainThread) [ocpp] EVB-P20482857: send [2,"ea9b8084-05ce-41e1-9905-7965dddaa091","GetConfiguration",{"key":["HeartbeatInterval"]}]
[... cut ...]
2022-01-18 09:50:01 DEBUG (MainThread) [websockets.protocol] server > Frame(fin=True, opcode=<Opcode.PING: 9>, data=b'\xafhD\xef', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:50:01 DEBUG (MainThread) [websockets.protocol] server - event = data_received(<10 bytes>)
2022-01-18 09:50:01 DEBUG (MainThread) [websockets.protocol] server < Frame(fin=True, opcode=<Opcode.PONG: 10>, data=b'\xafhD\xef', rsv1=False, rsv2=False, rsv3=False)
2022-01-18 09:50:01 DEBUG (MainThread) [websockets.protocol] server - received solicited pong: af6844ef
```
Once more this is without any chrging session being active

[47 more replies]

closed 2022-02-04 by lbbrhzn