lbbrhzn/ocpp #224: Compatible with Grizzl-E Smart EVSE

malaiwah, 2021-12-10:
I did test my Grizzl-E Smart charger against my Home Assistant installation.
https://grizzl-e.com/grizzl-e-smart/

The EVSE Firmware is currently 5.43.

Once the OCPP integration was installed via HACS, I did a wifi-reset on the EVSE, got into the UC_something wifi SSID and opened up a browser to 192.168.4.1 (default static IP of the EVSE) and I entered in `<local_ip>:9000` for the portal configuration URL (instead of their acharger.ca)

All seems to be well with it, it could be added to the list of compatible/tested-with devices.

[image]
[image]
[image]
[image]
[image]

dreimer1986, 2021-12-10:
Hmm, I wonder what are the differences between yours and this EVSE: https://github.com/lbbrhzn/ocpp/issues/176

tube0013, 2021-12-10:
I just tried mine again, I thought it had not been connected for a while to any server, it's now showing firmware 05.51 if you 05.43 - which I think mine was at originally. any chance you can unplug the charger and hook into the esp32 and read the FW off it with esptool?

lbbrhzn, 2022-02-04:
no action needed. closing.

closed 2022-02-04 by lbbrhzn

wipeout666, 2022-03-09:
@tube0013 I can do that for you, I still have FW 05.43. I have esptool ready, just have no idea what is the flash size. If you can find out, I'll do a flash_read for you. And the baudrate, if that matters (I've never done such a flash dump for a ESP32)

wipeout666, 2022-03-16:
United Charge just told me that v5.6 is adding back the ability to provide your own OCPP URL.

lbbrhzn, 2022-03-16:
See #442 for results with a recent firmware version.