lbbrhzn/ocpp #236: v0.3.23 Problem with current

lbbrhzn, 2021-12-13:
### Discussed in https://github.com/lbbrhzn/ocpp/discussions/233

<div type='discussions-op-text'>

<sup>Originally posted by **McFranky** December 13, 2021</sup>
Since v0.3.23 the current_import sensor is 0A on my ABB Terra AC Wallbox. Before I got the actual current...</div>

McFranky, 2021-12-14:
In v0.3.24 the problem with no current always exists.
[image]

And now, I get no voltage! New Bug
[image]

McFranky, 2021-12-15:
Test Version 0.3.25 on ABB Terra AC
Now the current is to small!
[image]
And same by the voltage...
[image]

McFranky, 2021-12-17:
Test v0.3.26 on ABB Terra AC
[image]
OK
[image]
OK
[image]
OK

Super!! Great Job!

closed 2021-12-17 by lbbrhzn