lbbrhzn/ocpp #263: No way to change max current on Schneider Smart WallBox

ClemKC, 2021-12-25:
Hi,
I get an issue on HomeAssistant version 2021.12.5 with OCPP version 0.4.1 and SmartWall box Schneider version 3400, I can't change the max current sent to the EV Car to avoid a shutdown of the whole house.
I also have another issue, when I restart HomeAssistant, sometimes the charger appears like 'Unvailable' while it is, everything appears like "Unvailable". 

I don't know if it helps for the load balancing but the charger send this error 'Sent SetChargingProfileConf: Rejected (Load balancer is disabled)".

To be followed
Thanks 
Clément (French)

sotoniak, 2021-12-25:
Hi, i have same problem with latest version OCPP on Wallbox PulsarPlus firmware 5.5.10 in Home Assistant. Version v0.3.26 is OK.

ClemKC, 2021-12-26:
Hi, Actually it didn't work on oldest version neither ;)

lbbrhzn, 2021-12-26:
@drc38, the number entity is unavailable, even though the status is 'available'. It appears to be related to the introduction of RestoreEntity. If I remove that the entity becomes available. Strange.

lbbrhzn, 2021-12-26:
Please try v0.4.2-beta

ClemKC, 2021-12-30:
Hello, same issue with v0.4.3-beta. Entity number.3n1906212005002_maximum_current is now unvailable either in charging mode than in available. I had the max current in version v0.3.26 but not on the last betat 0.4.3. I would like to export logs to help you but don't know how to do

drc38, 2021-12-30:
Check the Features metric, if SMART is not listed your charger does not support the Ocpp smart charging feature set which includes changing the current.

ClemKC, 2021-12-30:
Yes the Smart WallBox sold by Schneider Electric is compatible with changing the current, it's compatible with everyhing (the first builder manufacturer since 2010 ;) ). I confirmed this with LMS system from Schneider which allow to change current rate charging. So is there something to add in the configuration files to work?

drc38, 2021-12-30:
What does your Features metric return?

ClemKC, 2021-12-30:
Active power imported, exported. Same for reactive power. Soc level of battery, Max power during session, Session duration and more than I forgot (HA is off at this time)

ClemKC, 2022-01-04:
Hello, any update or news from the bug?

julien-lebot, 2022-01-04:
I have the same Wallbox and I would be interested in getting this to work too.
I'm running ocpp v0.4.3-beta and here's the output of my `features` metrics: `Profiles.RES|SMART|FW|CORE`
Let me know if I can help.

drc38, 2022-01-04:
See #221 for how to get the diagnostics log from the charger in case it has further information on why it is not allowing setting of the charging profile.  

Otherwise you'll need to go back to Schneider with the ocpp message and response for support.

ClemKC, 2022-01-07:
Julien Lebot could you please help us with your wallbox ? I'm going to send the diag to Schneider for support

ClemKC, 2022-01-07:
After discussion with Schneider, OCPP server needs to set a profile charging to dynamic or something like this. What is not done at this time by the OCCP server so this is the reason why the charge point says "balancer needs to be enabled". Could you have a look drc38?

drc38, 2022-01-07:
> After discussion with Schneider, OCPP server needs to set a profile charging to dynamic or something like this. What is not done at this time by the OCCP server so this is the reason why the charge point says "balancer needs to be enabled". Could you have a look drc38?

The Ocpp v1.6 standard has no reference to a "dynamic" profile, can you ask them for an example of a valid ocpp request?

drc38, 2022-01-07:
Take a look at line 506 of of the api.py as that defines the ocpp message sent to set the maximum charge rate.
```
req = call.SetChargingProfilePayload(
                connector_id=0,
                cs_charging_profiles={
                    om.charging_profile_id.value: 8,
                    om.stack_level.value: stack_level,
                    om.charging_profile_kind.value: ChargingProfileKindType.relative.value,
                    om.charging_profile_purpose.value: ChargingProfilePurposeType.charge_point_max_profile.value,
                    om.charging_schedule.value: {
                        om.charging_rate_unit.value: units,
                        om.charging_schedule_period.value: [
                            {om.start_period.value: 0, om.limit.value: lim}
                        ],
                    },
                },
            )
```

ClemKC, 2022-01-18:
Hi! I got news, how to upload files to the chat? # #

drc38, 2022-01-18:
> Hi! I got news, how to upload files to the chat? # #

[image]

lbbrhzn, 2022-02-02:
We now support two different kind of charging profiles to set the maximum current. It is confirmed to work for Wallbox Plus.
Closing this issue.

closed 2022-02-02 by lbbrhzn

ClemKC, 2022-02-03:
I've tried tonight on Schneider SmartWallBox but wallbox returns a message saying rejected

ClemKC, 2022-02-03:
Not tried with a car yet

drc38, 2022-02-03:
Looks like the TxDefaultProfile is being accepted, so when you are able to charge check that it does limit the current to your new setting.  It should work now.

ClemKC, 2022-02-04:
Exactly, you can close the ticket it works well. I've noticed an issue but I will write it in another post