lbbrhzn/ocpp #271: feature: always allow charging

mogorman, 2021-12-29:
**Is your feature request related to a problem? Please describe.**
I am using ocpp feature just to have local monitoring of my charger. I don't need the feature that disables the charger.  I'd like it to always be hot

**Describe the solution you'd like**
A config option to just always have charge_control enabled
**Describe alternatives you've considered**
I currently have a workaround that seems to work, automation that just enables when it goes unavailable.

```yaml
automation charger:
  action:
  - device_id: 58c923e15372dfada5e4622bb53747ee
    domain: switch
    entity_id: switch.wallbox_charge_control
    type: turn_on
  alias: Charge Car
  description: Flip switch to allow ocpp to charge car
  id: '1627073230476'
  mode: single
  trigger:
  - device_id: 58c923e15372dfada5e4622bb53747ee
    domain: switch
    entity_id: switch.wallbox_availability
    platform: device
    type: turned_off

```

**Additional context**
It's possible I am missing something but thought id ask here even if its just a guide for others.

drc38, 2021-12-29:
Refer to discussion #231, the feature is set by the charger manufacturer.

lbbrhzn, 2022-01-11:
won't fix. The behavior is charger dependent. For chargers that need it, you can use the suggested automation.

closed 2022-02-02 by lbbrhzn