lbbrhzn/ocpp #272: Make Home Assistant notifications configurable by log level

N3rdix, 2021-12-29:
Some of the HA notifications and warnings are not important and useless. 
**Example**:
```
Charger Garage booted
```

[image]

**Describe the solution you'd like**
Introduce some kind of log level within the integration to filter the notifications, e.g. debug, info, warning, error, fatal

drc38, 2022-01-03:
By way of background to the notifications, there are 3 types:
 - Warnings thrown by the charger in response to a request from the integration (usually due to a feature not supported by the manufacturer)
 - Booting of charger (to confirm the charger has booted when pressing reset, or if there is some other reason that is useful to know about eg power failure/resore)
 - Data transfer from the charger which is unsolicited and will vary by charger as to when and why a transfer is sent

drc38, 2022-01-03:
See #221 for how to setup the integration so it doesn't throw a MeterValuesSampledData warning.

closed 2022-04-13 by github-actions[bot]