lbbrhzn/ocpp #337: Getting a zillion persistent notifications

confucius78, 2022-01-31:
After rebooting following an update to 0.12, every few seconds a persistent notification is created to report:

'Warning: charger reported Rejected while setting MeterValuesSampledData=Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,SoC'

Or:

'Charger CKcharger booted'

I have an Alfen charger, which has been working fine until V0.10. The only issue I have seen so far is a single persistent notification:
'Warning: charger reported Rejected while setting MeterValuesSampledData=Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,SoC'

Running Home Assistant 2022.2b4

I will try and upload debug log later. For now I have reverted to V0.10

lbbrhzn, 2022-01-31:
ok, thanks for reporting. 
This may be caused by triggering a boot notification upon connect. If a charger not only re-sends the boot notification after the trigger, but also reboots/reconnects, it will enter an infinite loop. I'll remove the trigger from reconnect.

confucius78, 2022-01-31:
I'll do some more digging in the meantime. It seems like this is related to me running the HA beta as well, as V0.10 shows the same behavior

lbbrhzn, 2022-01-31:
hmm, in that case there may be something else causing the reconnect loop.
I'll try 2022.2b4 as well.

warksit, 2022-02-01:
Also have this issue with an Alfen Eve Mini. (Only installed yesterday so no previous data)

lbbrhzn/ocpp 0.12
ha 2021.12.10

N3rdix, 2022-02-01:
Same issue for me with a Vestel EVC04, getting those 2 errors in the log:

```
Logger: homeassistant
Source: custom_components/ocpp/api.py:740
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 17:08:53 (8 occurrences)
Last logged: 17:15:15

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 270, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 774, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 303, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 379, in post_connect
    await self.configure(
  File "/config/custom_components/ocpp/api.py", line 740, in configure
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 273, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"a368283c-7c0b-41c0-80ee-8ddb4849b4ff","GetConfiguration",{"key":["MeterValueSampleInterval"]}].
```

```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 270, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 774, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 303, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 390, in post_connect
    resp = await self.get_configuration(ckey.number_of_connectors.value)
  File "/config/custom_components/ocpp/api.py", line 714, in get_configuration
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 273, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"80ee2d5e-fdad-40c7-bf77-38afbb56ec55","GetConfiguration",{"key":["NumberOfConnectors"]}].
```

confucius78, 2022-02-01:
Ok, so I have found out a bit more.

1. Actually the 2 messages I am getting are 2 distinct issues: 
- one is about failing to set the values the charger should report back (it seems to report this setting is r/w, but it does not accept a change). Again: separate issue and looking back I seem to have had this issue longer
- the second is the apparent 'bootloop', although the charger actually is not rebooting. It looks as if the integration gets stuck in a loop when the charger reports its status as 'booted', although I don't see it sending that in its actual logs.

I'll try to get those logs posted tomorrow, both for hass as well as the charger

lbbrhzn, 2022-02-02:
please check v0.4.13

confucius78, 2022-02-02:
Don't see a HACS update yet, but I will as soon as I do

fwestenberg, 2022-02-02:
Same error here, also with the Alfen Pro. I manually updated the library to v0.4.13 (HACS didn't show the update yet), now the persistent notifications are gone. Now I got this error in the log:

```
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:186
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 09:06:13 (4205 occurrences)
Last logged: 09:11:43

Exception occurred: 'WebSocketServerProtocol' object has no attribute 'id'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 186, in on_connect
    f"Charger {cp_id} connected to {self.host}:{self.port} websocket={websocket.id}."
AttributeError: 'WebSocketServerProtocol' object has no attribute 'id'
```

and 

```
Logger: websockets.server
Source: custom_components/ocpp/api.py:203
Integration: Open Charge Point Protocol (OCPP) (documentation, issues)
First occurred: 09:06:13 (4205 occurrences)
Last logged: 09:11:43

Error in connection handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/server.py", line 293, in handler
    await self.ws_handler(self, path)
  File "/config/custom_components/ocpp/api.py", line 203, in on_connect
    f"Charger {cp_id} disconnected from {self.host}:{self.port} websocket={websocket.id}."
AttributeError: 'WebSocketServerProtocol' object has no attribute 'id'
```

drc38, 2022-02-02:
I had the same error, it looks like HA can get confused with the websockets version installed (as other integrations only work with 9.1). You need to go to bash and run `pip websockets --upgrade`

lbbrhzn, 2022-02-02:
i've removed the references to websocket.id, because it seems we can't rely on HACS installing the required version.
see v0.4.14

fwestenberg, 2022-02-02:
v0.4.14 works again! Only this message still exists as notification (only 1 notification exists):

` ` `
Ocpp integration
Warning: charger reported Rejected while setting MeterValuesSampledData=Energy.Active.Import.Register,Energy.Reactive.Import.Register,Energy.Active.Import.Interval,Energy.Reactive.Import.Interval,Power.Active.Import,Power.Reactive.Import,Power.Offered,Power.Factor,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature,Current.Export,Energy.Active.Export.Register,Energy.Reactive.Export.Register,Energy.Active.Export.Interval,Energy.Reactive.Export.Interval,Power.Active.Export,Power.Reactive.Export
```

Any idea? Values like actual current / voltage / frequency are not available. I really like to see this, the actual meter value is working however.

fwestenberg, 2022-02-02:
Correction: after setting some metered values in the Alfen MyEve app, I see more data in HA as well!

In the app, go to Advanced ->Connectivity->Meter value->Meter values sampled data and select the fields you'd like to see in HA.

confucius78, 2022-02-02:
Ok cool no more notifications, however it seems there is still some kind of loop going on.

The charger.Availability-entity is switching on/off every second, similar frequency to the earlier notifications

lbbrhzn, 2022-02-02:
@confucius78 can you upload a log fragment?

confucius78, 2022-02-02:
Yeah, working on it :-)

I see this in the log of the charger:
[image]

confucius78, 2022-02-02:
Home assistant log:
2022-02-02 13:04:49 INFO (MainThread) [ocpp] CKcharger: send [3,"295",{}]
2022-02-02 13:04:49 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:50 INFO (MainThread) [ocpp] CKcharger: receive message [2,"296","StatusNotification",{"connectorId":0,"errorCode":"OtherError","info":"Modbus TCP/IP lost, in safe mode max 14.0A","status":"Faulted","timestamp":"2022-01-31T12:36:27Z","vendorErrorCode":"210"}]
2022-02-02 13:04:50 INFO (MainThread) [ocpp] CKcharger: send [3,"296",{}]
2022-02-02 13:04:50 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:52 INFO (MainThread) [ocpp] CKcharger: receive message [2,"297","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T12:36:27Z"}]
2022-02-02 13:04:52 INFO (MainThread) [ocpp] CKcharger: send [3,"297",{}]
2022-02-02 13:04:52 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:53 INFO (MainThread) [ocpp] CKcharger: receive message [2,"298","StatusNotification",{"connectorId":0,"errorCode":"OtherError","info":"Modbus TCP/IP lost, in safe mode max 14.0A","status":"Faulted","timestamp":"2022-01-31T12:36:29Z","vendorErrorCode":"210"}]
2022-02-02 13:04:53 INFO (MainThread) [ocpp] CKcharger: send [3,"298",{}]
2022-02-02 13:04:53 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:55 INFO (MainThread) [ocpp] CKcharger: receive message [2,"299","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T12:36:29Z"}]
2022-02-02 13:04:55 INFO (MainThread) [ocpp] CKcharger: send [3,"299",{}]
2022-02-02 13:04:55 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:56 INFO (MainThread) [ocpp] CKcharger: receive message [2,"300","StatusNotification",{"connectorId":0,"errorCode":"OtherError","info":"Modbus TCP/IP lost, in safe mode max 14.0A","status":"Faulted","timestamp":"2022-01-31T12:36:31Z","vendorErrorCode":"210"}]
2022-02-02 13:04:56 INFO (MainThread) [ocpp] CKcharger: send [3,"300",{}]
2022-02-02 13:04:56 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:58 INFO (MainThread) [ocpp] CKcharger: receive message [2,"301","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T12:36:31Z"}]
2022-02-02 13:04:58 INFO (MainThread) [ocpp] CKcharger: send [3,"301",{}]
2022-02-02 13:04:58 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:04:59 INFO (MainThread) [ocpp] CKcharger: receive message [2,"302","StatusNotification",{"connectorId":0,"errorCode":"OtherError","info":"Modbus TCP/IP lost, in safe mode max 14.0A","status":"Faulted","timestamp":"2022-01-31T12:36:33Z","vendorErrorCode":"210"}]
2022-02-02 13:04:59 INFO (MainThread) [ocpp] CKcharger: send [3,"302",{}]
2022-02-02 13:04:59 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:05:00 WARNING (MainThread) [homeassistant.helpers.script] Error in 'choose[1]' evaluation: In 'state':
  In 'state' condition: unknown entity switch.garage_vent_highon
2022-02-02 13:05:01 INFO (MainThread) [ocpp] CKcharger: receive message [2,"303","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T12:36:33Z"}]
2022-02-02 13:05:01 INFO (MainThread) [ocpp] CKcharger: send [3,"303",{}]
2022-02-02 13:05:01 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:05:02 INFO (MainThread) [ocpp] CKcharger: receive message [2,"304","StatusNotification",{"connectorId":0,"errorCode":"OtherError","info":"Modbus TCP/IP lost, in safe mode max 14.0A","status":"Faulted","timestamp":"2022-01-31T12:36:35Z","vendorErrorCode":"210"}]
2022-02-02 13:05:02 INFO (MainThread) [ocpp] CKcharger: send [3,"304",{}]
2022-02-02 13:05:02 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.
2022-02-02 13:05:03 INFO (MainThread) [ocpp] CKcharger: receive message [2,"305","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T12:36:35Z"}]
2022-02-02 13:05:03 INFO (MainThread) [ocpp] CKcharger: send [3,"305",{}]
2022-02-02 13:05:03 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_connection_latency not found.

lbbrhzn, 2022-02-02:
Can you remove and re-add the ocpp integration? that should get rid of the Forced update failed warnings.
Other than that, I just see the charger continuously sending status notifications, there are no messages going from the central  to the charger.  Apparently it is in safe mode, due to TCP/IP lost. But these messages are sent anyway?!

lbbrhzn, 2022-02-02:
If you add debug to configuration.yaml, you will also see ping/pong latencies.
```
logger:
  default: info
  logs:
    custom_components.ocpp: debug
    websocket: info
```

confucius78, 2022-02-02:
Ok done... I have setup a separate box with HASS to test, loaded with latest ocpp pulled from github. Logging settings per your recommendation.

Note the charger seems to report status of 2 connectors (id 0 and 1), only one of which is in this error state.
It seems the availability switch in the HA frontend seems to be flipping with the then last reported connector status

It does that despite reporting back it has only one connector, which is true :-)

Logs below:

2022-02-02 13:33:22 INFO (MainThread) [websockets.server] connection open
2022-02-02 13:33:22 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2022-02-02 13:33:22 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/CKcharger
2022-02-02 13:33:22 INFO (MainThread) [custom_components.ocpp] Charger CKcharger connected to 0.0.0.0:9000.
2022-02-02 13:33:23 INFO (MainThread) [ocpp] CKcharger: receive message [2,"209","MeterValues",{"connectorId":1,"transactionId":1643539948,"meterValue":[{"timestamp":"2022-01-31T14:44:39Z","sampledValue":[{"value":"2311522.000","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"0.000","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"21.125","context":"Sample.Periodic","measurand":"Temperature","location":"Body","unit":"Celsius"}]}]}]
2022-02-02 13:33:23 INFO (MainThread) [ocpp] CKcharger: send [3,"209",{}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: send [2,"905e6660-3042-4855-86aa-3e1b0251312b","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: receive message [3,"905e6660-3042-4855-86aa-3e1b0251312b",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,Reservation,LocalAuthListManagement,RemoteTrigger,SmartCharging"}]}]
2022-02-02 13:33:24 INFO (MainThread) [custom_components.ocpp] Supported feature profiles: Core,FirmwareManagement,Reservation,LocalAuthListManagement,RemoteTrigger,SmartCharging
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: send [2,"a05a7b51-2720-4ca7-80af-f1dd07d20f15","TriggerMessage",{"requestedMessage":"BootNotification"}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: receive message [3,"a05a7b51-2720-4ca7-80af-f1dd07d20f15",{"status":"Accepted"}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: send [2,"9106aee1-4a72-48ed-8a9f-80ce920e911c","TriggerMessage",{"requestedMessage":"StatusNotification"}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: receive message [3,"9106aee1-4a72-48ed-8a9f-80ce920e911c",{"status":"Accepted"}]
2022-02-02 13:33:24 INFO (MainThread) [ocpp] CKcharger: send [2,"d41a8811-e3d6-462b-b6a4-a2f45b7f0fad","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"d41a8811-e3d6-462b-b6a4-a2f45b7f0fad",{"status":"Accepted"}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: send [2,"77d220bc-8e25-43f0-9a19-8b766f9ea15f","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"77d220bc-8e25-43f0-9a19-8b766f9ea15f",{"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2022-02-02 13:33:25 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: send [2,"aac15c37-e29e-40db-9445-e8fe9df9a9bc","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"aac15c37-e29e-40db-9445-e8fe9df9a9bc",{"configurationKey":[{"key":"WebSocketPingInterval","readonly":false,"value":"60"}]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: send [2,"bc59a99c-e692-4851-80be-8f3c021b3256","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"bc59a99c-e692-4851-80be-8f3c021b3256",{"configurationKey":[{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import,Power.Offered,SoC,Temperature"}]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: send [2,"75c01ff9-6fee-43e4-92d3-31c91cd7e0e2","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature"}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"75c01ff9-6fee-43e4-92d3-31c91cd7e0e2",{"status":"Rejected"}]
2022-02-02 13:33:25 WARNING (MainThread) [custom_components.ocpp] Rejected while setting MeterValuesSampledData to Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Power.Offered,Current.Import,Current.Offered,Voltage,Frequency,RPM,SoC,Temperature
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: send [2,"67e62d32-3082-48fe-abdb-5e8fd830f91b","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2022-02-02 13:33:25 INFO (MainThread) [ocpp] CKcharger: receive message [3,"67e62d32-3082-48fe-abdb-5e8fd830f91b",{"configurationKey":[{"key":"MeterValueSampleInterval","readonly":false,"value":"60"}]}]
[... cut ...] rCode":"210"}]
2022-02-02 13:33:40 INFO (MainThread) [ocpp] CKcharger: send [3,"213",{}]
2022-02-02 13:33:42 INFO (MainThread) [ocpp] CKcharger: receive message [2,"214","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2022-01-31T14:44:30Z"}]
2022-02-02 13:33:42 INFO (MainThread) [ocpp] CKcharger: send [3,"214",{}]
2022-02-02 13:33:42 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'CKcharger': ping=1.0 ms, pong=9.0 ms

lbbrhzn, 2022-02-02:
connectorId=0 is for the charger itself, connectorid=1 is for the first connector. Currently the integration discards the connectorId completely, that will result in the toggling behavior.

@drc38, should we only display status for connectorId=0? Or create separate sensors?

confucius78, 2022-02-02:
FYI: I think the connection error it is showing is due to it not having a backoffice installed. Charger was bought and installed by me, no backoffice and no sim enabled. It is hardwired ethernet however

confucius78, 2022-02-02:
Sorry for posting so often ;-) I just noticed something that may be important: the charger is currently reporting in 'Generic state' that modbus is 'NOT IN USE (COMMUNICATION_IDLE)'

I seem to recall it was showing a different state there before after having connected to OCPP in home assistant

lbbrhzn, 2022-02-02:
Please try v0.4.15. It has separate sensors for main status and connector.

lbbrhzn, 2022-02-02:
The original problem has been solved, so I am closing this issue.

closed 2022-02-02 by lbbrhzn

confucius78, 2022-02-03:
Thanks for the help! I didn't think we could get it done so quickly! GJ