lbbrhzn/ocpp #361: No data from Alfen EVE Pro chargepoint

confucius78, 2022-02-03:
Hi,

to make sure there is no confusion: this is not concerning the same chargepoint as discussed in issue #337 

I am trying to connect an Alfen EVE Pro charger to the integration, but despite it having the identical firmware and settings (as far as I can tell) as my own it is throwing an error after making the connection.

log:
> 2022-02-03 10:51:55 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'NL-GFX-A2246': ping=1.0 ms, pong=8.0 ms
> 2022-02-03 10:51:58 INFO (MainThread) [ocpp] NL-GFX-A2246: receive message [2,"4","MeterValues",{"connectorId":1,"transactionId":9223372036854775807,"meterValue":[{"timestamp":"2020-09-19T15:14:23Z","sampledValue":[{"value":"15.630","context":"Sample.Periodic","unit":"A"},{"value":"16.000","context":"CurrentMax","unit":"A"}]}]}]
> 2022-02-03 10:51:58 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to 'NL-GFX-A2246': 'Payload '{'connectorId': 1, 'transactionId': 9223372036854775807, 'meterValue': [{'timestamp': '2020-09-19T15:14:23Z', 'sampledValue': [{'value': '15.630', 'context': 'Sample.Periodic', 'unit': 'A'}, {'value': '16.000', 'context': 'CurrentMax', 'unit': 'A'}]}]} for action 'MeterValues' is not valid: 'CurrentMax' is not one of ['Interruption.Begin', 'Interruption.End', 'Sample.Clock', 'Sample.Periodic', 'Transaction.Begin', 'Transaction.End', 'Trigger', 'Other']
> 
> Failed validating 'enum' in schema['properties']['meterValue']['items']['properties']['sampledValue']['items']['properties']['context']:
>     {'additionalProperties': False,
>      'enum': ['Interruption.Begin',
>               'Interruption.End',
>               'Sample.Clock',
>               'Sample.Periodic',
>               'Transaction.Begin',
>               'Transaction.End',
>               'Trigger',
>               'Other'],
>      'type': 'string'}
> 
> On instance['meterValue'][0]['sampledValue'][1]['context']:
>     'CurrentMax''
> Traceback (most recent call last):
>   File "/srv/homeassistant/lib/python3.9/site-packages/ocpp/messages.py", line 192, in validate_payload
>     validator.validate(message.payload)
>   File "/srv/homeassistant/lib/python3.9/site-packages/jsonschema/validators.py", line 353, in validate
>     raise error
> jsonschema.exceptions.ValidationError: 'CurrentMax' is not one of ['Interruption.Begin', 'Interruption.End', 'Sample.Clock', 'Sample.Periodic', 'Transaction.Begin', 'Transaction.End', 'Trigger', 'Other']
> 
> Failed validating 'enum' in schema['properties']['meterValue']['items']['properties']['sampledValue']['items']['properties']['context']:
>     {'additionalProperties': False,
>      'enum': ['Interruption.Begin',
>               'Interruption.End',
>               'Sample.Clock',
>               'Sample.Periodic',
>               'Transaction.Begin',
>               'Transaction.End',
>               'Trigger',
>               'Other'],
>      'type': 'string'}
> 
> On instance['meterValue'][0]['sampledValue'][1]['context']:
>     'CurrentMax'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 846, in run
>     await asyncio.gather(*self.tasks)
>   File "/srv/homeassistant/lib/python3.9/site-packages/ocpp/charge_point.py", line 131, in start
>     await self.route_message(message)
>   File "/srv/homeassistant/lib/python3.9/site-packages/ocpp/charge_point.py", line 149, in route_message
>     await self._handle_call(msg)
>   File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 831, in _handle_call
>     await super()._handle_call(msg)
>   File "/srv/homeassistant/lib/python3.9/site-packages/ocpp/charge_point.py", line 173, in _handle_call
>     validate_payload(msg, self._ocpp_version)
>   File "/srv/homeassistant/lib/python3.9/site-packages/ocpp/messages.py", line 194, in validate_payload
>     raise ValidationError(f"Payload '{message.payload} for action "
> ocpp.exceptions.ValidationError: Payload '{'connectorId': 1, 'transactionId': 9223372036854775807, 'meterValue': [{'timestamp': '2020-09-19T15:14:23Z', 'sampledValue': [{'value': '15.630', 'context': 'Sample.Periodic', 'unit': 'A'}, {'value': '16.000', 'context': 'CurrentMax', 'unit': 'A'}]}]} for action 'MeterValues' is not valid: 'CurrentMax' is not one of ['Interruption.Begin', 'Interruption.End', 'Sample.Clock', 'Sample.Periodic', 'Transaction.Begin', 'Transaction.End', 'Trigger', 'Other']
> 
> Failed validating 'enum' in schema['properties']['meterValue']['items']['properties']['sampledValue']['items']['properties']['context']:
>     {'additionalProperties': False,
>      'enum': ['Interruption.Begin',
>               'Interruption.End',
>               'Sample.Clock',
>               'Sample.Periodic',
>               'Transaction.Begin',
>               'Transaction.End',
>               'Trigger',
>               'Other'],
>      'type': 'string'}
> 
> On instance['meterValue'][0]['sampledValue'][1]['context']:
>     'CurrentMax'
> 2022-02-03 10:51:58 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'NL-GFX-A2246'

Charger log (only a small section of continuously repeating the below):
> 2022-02-03T10:56:22.750Z:INFO:websockets.c:440:Connected to Socket 172.16.220.111:9000
> 2022-02-03T10:56:22.761Z:INFO:websockets.c:2172:WS timeout: 30000 ms
> 2022-02-03T10:56:22.785Z:INFO:websockets.c:806:WebSocket upgrade successful
> 2022-02-03T10:56:22.804Z:USER:taskOCPP.c:462:Back office connection available
> 2022-02-03T10:56:58.027Z:INFO:taskMaster.c:4946:Socket #1: next sample interval metervalue in 60 seconds
> 2022-02-03T10:56:58.375Z:INFO:active_lb.c:292:SMt:(-1.0 -1.0 -1.0)A Free:(26.0 26.0 26.0)A ALB:(26.0 26.0 26.0)A
> 2022-02-03T10:57:22.738Z:INFO:taskMaster.c:3685:MeterValues delayed by 3 sec
> 2022-02-03T10:57:28.214Z:INFO:taskMaster.c:3393:TX item @0x00025002 timestamp corrected with 330 seconds
> 2022-02-03T10:57:28.226Z:COM:ocpp_rpc.c:663:-> [2,"8","MeterValues",{"connectorId":1,"transactionId":9223372036854775807,"me
> 2022-02-03T10:57:28.226Z:COM:ocpp_rpc.c:663:terValue":[{"timestamp":"2020-09-19T15:14:23Z","sampledValue":[{"value":"15.630"
> 2022-02-03T10:57:28.226Z:COM:ocpp_rpc.c:663:,"context":"Sample.Periodic","unit":"A"},{"value":"16.000","context":"CurrentMax
> 2022-02-03T10:57:28.226Z:COM:ocpp_rpc.c:663:","unit":"A"}]}]}]
> 2022-02-03T10:57:28.292Z:ERROR:ocpp_rpc.c:425:reply error DISCONNECTED
> 2022-02-03T10:57:28.300Z:USER:taskOCPP.c:479:Back office connection lost

I have previously connected my own charger to this exact same HSS and OCPP instance, and it does work for me.

lbbrhzn, 2022-02-03:
The MeterValues message you get from this particular charger does not comply to OCPP standards: The context field is incorrect. 
Which version of the ocpp integration are your using? With version v0.4.16 we've upgraded the upstream ocpp package version to 13.1, this has some improvements in handling validation errors, see https://github.com/mobilityhouse/ocpp/commit/4ee2cae7b4cd4d218804533871054145b93909ee

confucius78, 2022-02-03:
I'll make sure the instance is updated. The challenge is this charger is that it is sending all of its history of not-yet-sent meter messages when first connecting to the ocpp instance.

No way to tell what fw version or settings it was at the time these messages were stored.

Thanks, I'll make sure the instance is fully updated

lbbrhzn, 2022-02-03:
In release v0.4.17 I added the config option to disable validation.

confucius78, 2022-02-03:
I saw, man that was fast! Many thanks, I'll test over the weekend and keep you posted!

confucius78, 2022-02-05:
So I did a quick check, but even with the option switched off I see disconnects of the charger by the ocpp host, could it be the 'don't check' is not fully implemented?

Sorry still new to python and github so maybe a redundant question..

lbbrhzn, 2022-02-08:
@confucius78, you need to enable the option to skip validation. By default it is off.
If you still see disconnects with the skip validation enabled, please upload your log file.

confucius78, 2022-02-10:
Unfortunately the chargepoint is no longer sending out erroneous messages, so I cannot test any further.

I uninstalled ocpp and re-installed, enabling the flag for error checking bypass. After that, the disconnects were still occurring.

As I cannot test however, propose to close the issue until someone encounters this again.

lbbrhzn, 2022-02-10:
the original issue is solved. closing

closed 2022-02-10 by lbbrhzn