lbbrhzn/ocpp #394: Similar issue to #374 - active power import missing

smbunn, 2022-02-14:
I have been using active.power.import in my Home Assistant for 5 months now but noticed recently it was giving me big problems, mainly because most of the time it reads zero
[image]

I use this in my calculations and this is screwing up my solar yield calculations as the house takes 5kW but with this value at zero is says the car is taking nothing when I know it is taking 3.2 kW

drc38, 2022-02-14:
Does the sensor work if you revert to a previous version of the ocpp integration? I don't recall having any issues with that sensor on mine. Will take a look tonight at the history.

drc38, 2022-02-14:
Mine is fine on v0.4.20
[image]

smbunn, 2022-02-16:
I have been logging for a day.  Where is the log stored so I can send it to you?

GrantKeymer, 2022-02-17:
> I have been logging for a day. Where is the log stored so I can send it to you?

http://<your IP address>:8123/config/logs

Then click the `Load Full Home Assistant Log` button.

lbbrhzn, 2022-03-25:
any update on this?