lbbrhzn/ocpp #400: Wallbox (Pulsar, Commander) says OCPP is limited, what's the exact status?

Hywan, 2022-02-16:
Hello,

I'm not opening an issue but rather a question. I came to your repo and read the code. Thanks for making it open source.
I've contacted Wallbox because I want to buy a Pulsar Plus or Commander 2 charger. But the Wallbox' support is saying **it's not possible to start or to stop a charge with OCPP**. I was extremely surprised to read that.

My needs are super basic: I need to start and stop a charge, and to adjust the current, based on my solar PV production. Can you confirm that it works with this project? If yes, it means that Wallbox' support is not super aware of their own product features.

Thanks!

PS: I'm asking because Wallbox Pulsar Plus is listed in [the Supported devices Section](https://github.com/lbbrhzn/ocpp#supported-devices).

lbbrhzn, 2022-02-16:
Hi @Hywan, I have a Wallbox Pulsar Plus, and I can confirm it is indeed supported.

In OCPP mode, the charging will not start automatically. It will report the connector status 'Preparing', and then you need to use the Charging switch to start it (this will send a RemoteStartTransaction request). You can use an automation to do it for you if you like.

The max charging current can be adjusted dynamically, but the charger does not support a current lower than 6A.
Note that Wallbox has some support for smart charging built in, see https://wallbox.com/en_nl/energy-management/eco-smart-solar-ev-charging. However, this does require an additional external power meter.

lbbrhzn, 2022-02-16:
On the UI.

lbbrhzn, 2022-02-16:
> Yeah, but I want to build my own one :-).

Good, that is what this integration is for. You may want to talk to  @GrantKeymer, he want to do exactly the same.

lbbrhzn, 2022-02-16:
'Locked' is not an OCPP status. I guess you refer to the status indicated in the Wallbox App.
What is the status indicated by the OCPP integration?

The OCPP statuses are these:
[image]

GrantKeymer, 2022-02-16:
Welcome @Hywan.  As another owner of Wallbox Pulsar Plus who wants to automate charging from excess solar production, we have 2 things in common.  I started out with an EM112 Energy Meter (supplied by Wallbox) trying to get my Pulsar Plus to respond according to the load connected to the terminals of the meter.  I had made a test rig so I could simulate import from the grid or export to the grid with varying rates of kW.  But I could not get it to work successfully, although the owner of this repo (lbbrhzn) has since managed to.

When I found this integration for HA, I realised how much better I could make the charging from solar power, than the two very simple options offered by Wallbox with their Eco-Smart upgrade.  This OCPP integration can do everything the Wallbox app does, and much more.  However, it has one limitation...

**I haven't managed to find a way to Pause and Resume a charging session.**

Instead you have to stop one session and start a new one.  Does it matter?  Probably not, but if you are charging your EV throughout a day and need to start and stop many times as household loads change, or clouds come over, it means that your EV will record a lot of extra L1/L2 charges, instead of just one.  I guess it's just something that annoys me as being untidy, less than optimal, which shouldn't need to be that way.  But it's the only limitation of OCPP that I'm aware of, and who knows, it could be improved in OCPP V2.0, or a future upgrade.

So, where to from here?  I've made a simple dashboard in Node-RED as you can find from the `Discussions` tab.

Now I'm waiting for a response from the provider of my Solar Power system.  They have great web portal where all required data is available e.g. Solar Production kW, Battery Charge/Discharge kW, Grid Import/Export kW and Household Load kW.  I managed to scrape it from the web page with a script, but the session token expires after a while and there doesn't seem to be an easy way around it.  So I asked them for API access and they eventually provided it, but again, there are limitations.  The worst one is again, the expiry of session tokens.  Just one simple function doesn't expire but all others expire after one hour.  Then you need to re-authorise the function and it creates a new token, so all is good for another hour.  Very frustrating and I don't expect they will improve the situation as it seems they don't like customers requesting regular updates from their server with a script.  There are rate limits and so on, although the worst limitation is the expiry of session tokens.

Assuming I cannot get around the expiry of session tokens, I plan to install a separate energy meter in my switchboard, alongside the existing one.  I like the look of the IAmmeter series because they have WiFi as well as RS485 comms and an integration is provided for HA.  Also, the price is reasonable...

https://www.iammeter.com/store

If you have another suggestion for an Energy Meter supported by HA, please share the details as I haven't placed the order yet.

Well, that's my journey into the land of HA, Solar Power and EV Charging.  Interested to hear more about your experience.

drc38, 2022-02-16:
> I haven't managed to find a way to Pause and Resume a charging session.

Did you see the post from @dreimer1986 where specifying a current limit above what the charger supports the Elvi goes into the `SuspendedEVSE` state? Worth trying on your Wallbox.

GrantKeymer, 2022-02-16:
> When my Pulsar Plus is locked and I send it a charger.Unlock request from Home Assistant, I get a "NotSupported" error back. The charger LED stays yellow, which I understand is the "locked" state. Do I need to connect a car for Unlock to be a valid operation?

Welcome @dansarginson.  I also have a Wallbox Pulsar Plus and this condition with the yellow LEDs confused me too.  When OCPP connects successfully, the LEDs go yellow, so I thought it meant the charger was locked because that's what happens with the Wallbox app.  However, it doesn't mean that at all.  In fact, if the LEDs are green, it means the charger is `Unavailable` when connected by OCPP.  Confusing?  Definitely!

But once you get your head around the different meanings of the LED colours when connected via OCPP, it's OK.

As others here have explained, the Lock/Unlock function of OCPP is intended to release the charging connector lock on versions of the Wallbox Pulsar or Commander which have a type 2 socket.  If you have a tethered cable like I do, the Lock/Unlock function is of no use at all.

I hope that explains what is happening and you can make this great OCPP integration work successfully with your Pulsar Plus.

GrantKeymer, 2022-02-16:
> > I haven't managed to find a way to Pause and Resume a charging session.
> 
> Did you see the post from @dreimer1986 where specifying a current limit above what the charger supports the Elvi goes into the `SuspendedEVSE` state? Worth trying on your Wallbox.

Thanks for the suggestion.  Yes, I've tried that.  But if I set the charging rate 17 to 32 Amps, it makes no difference, the rate stays at 16 Amps.  Now you have a charging rate of 40 Amps supported, I should try that and see what happens.

drc38, 2022-02-16:
Yes it would need to be above the charger maximum ie 32A (as opposed to the vehicle maximum). Also take a look at https://github.com/TimSoethout/goodwe-sems-home-assistant as an example for how to deal with expiring api tokens.

dreimer1986, 2022-02-17:
> > I haven't managed to find a way to Pause and Resume a charging session.
> 
> Did you see the post from @dreimer1986 where specifying a current limit above what the charger supports the Elvi goes into the `SuspendedEVSE` state? Worth trying on your Wallbox.

Above and below. Maybe a setting of 5A will do the magic. Here it does. Minimum is 6A and can not set below that. Only higher is possible.

GrantKeymer, 2022-02-17:
> Above and below. Maybe a setting of 5A will do the magic. Here it does. Minimum is 6A and can not set below that. Only higher is possible.

Thanks for the suggestion.  I've already tried settings of 5 Amps and below but they all get rejected with charge current remaining at 6 Amps.

dreimer1986, 2022-02-17:
Interesting... I wonder if there is at least ONE Charger out there that really follows a standard. Even my certified one does everything, but not follow the standard they got certified for. Does below mean you tried 0 A, too?

GrantKeymer, 2022-02-17:
> Interesting... I wonder if there is at least ONE Charger out there that really follows a standard. Even my certified one does everything, but not follow the standard they got certified for. Does below mean you tried 0 A, too?

Yes I did.  Have a look at this discussion: https://github.com/lbbrhzn/ocpp/discussions/333