lbbrhzn/ocpp #418: Transaction History

warksit, 2022-03-01:
**Describe the solution you'd like**
Total kwh used by tag_id for the last week/month/year

This may already be possible but I'm not sure how.

drc38, 2022-03-01:
@lbbrhzn @grantkeymer suggest the tag_id gets added to the extra attributes of the session energy metric. Then it could be extracted in a value template or similar to produce custom sensors/aggregators by users. Thoughts?

GrantKeymer, 2022-03-01:
I saw this request and my immediate thought was that this kind of function is really up to users to implement for themselves.  The question then becomes where to store the tag_id so it can be associated with the energy supplied to a user in possession of that tag?

Your idea of storing the tag_id as an extra_attr of the session energy metric therefore makes a lot of sense.  However, users would need to ensure they retrieved the tag_id at the close of a charging session, otherwise it would be overwritten when a new session started.

drc38, 2022-03-01:
With a combination of a sensor template https://www.home-assistant.io/integrations/template/ and meter https://www.home-assistant.io/integrations/utility_meter/ users should be able to create a "tag_id" specific sensor and aggregate on whatever frequency they choose.

drc38, 2022-03-01:
@warksit see #419, once you've got it setup please document the approach for the wiki.

lbbrhzn, 2022-03-25:
@warsit, did you get it working?

N3rdix, 2022-04-20:
I use the built-in energy dashboard of HA to track long-term ststistics and it works quite well.
Habe built a template sensor on top of energy_active_import_register to avoid smaller or zero values (as I had issues with the values when a new charging session started).

you could also enhance this pattern and create separate sensors per tag_id I guess…

closed 2022-06-27 by github-actions[bot]