lbbrhzn/ocpp #432: lost connection after second

mistrals, 2022-03-08:
## Version of the custom_component

4.22

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

lost connection after 1 second

## Debug log

```text

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 880, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 398, in post_connect
    await self.configure(ckey.web_socket_ping_interval.value, "60")
  File "/config/custom_components/ocpp/api.py", line 780, in configure
    for key_value in resp.configuration_key:
TypeError: 'NoneType' object is not iterable

```

mistrals, 2022-03-08:
and after

`Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 393, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 467, in get_supported_features
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 271, in call
    await self._send(call.to_json())
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 325, in _send
    await self._connection.send(message)
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 619, in send
    await self.ensure_open()
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 920, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: sent 1000 (OK); then received 1005 (no status code [internal])`

dreimer1986, 2022-03-10:
How about... information? Manufacturer, charger name, charger firmware? All the stuff we might need to help?

mistrals, 2022-03-11:
charger is Orbis Viaris Uni firmware 4.0.21. Manufacturer report compatibility with OCPP 1.6J
--

mistrals, 2022-03-12:
`2022-03-12 10:10:04 INFO (MainThread) [ocpp] : receive message [2,"1","BootNotification",{"chargePointModel":"VIARIS UNI","chargePointSerialNumber":"EVVC3EB8B3FB4","chargePointVendor":"Orbis","firmwareVersion":"4.0.21"}]
2022-03-12 10:10:04 INFO (MainThread) [ocpp] : send [3,"1",{"currentTime":"2022-03-12T09:10:04.307415+00:00","interval":3600,"status":"Accepted"}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : send [2,"c3f42f25-1937-4bcf-95e9-5600821929aa","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : receive message [3,"c3f42f25-1937-4bcf-95e9-5600821929aa",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":""}],"unknownKey":[]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : send [2,"c6e8cd44-3689-4e70-a8b2-599469747181","GetConfiguration",{"key":["NumberOfConnectors"]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : receive message [3,"c6e8cd44-3689-4e70-a8b2-599469747181",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"2"}],"unknownKey":[]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : send [2,"47e4f75e-6d0a-4863-a504-e8f0a80ee2e3","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : receive message [3,"47e4f75e-6d0a-4863-a504-e8f0a80ee2e3",{"status":"Accepted"}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : send [2,"244f1fa3-db32-435e-bf89-a51affb38759","GetConfiguration",{"key":["HeartbeatInterval"]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : receive message [3,"244f1fa3-db32-435e-bf89-a51affb38759",{"unknownKey":["HeartbeatInterval"]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : send [2,"2f78a96b-17c2-4395-8da9-b436f9ffc24f","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2022-03-12 10:10:05 INFO (MainThread) [ocpp] : receive message [3,"2f78a96b-17c2-4395-8da9-b436f9ffc24f",{"unknownKey":["WebSocketPingInterval"]}]
2022-03-12 10:10:05 INFO (MainThread) [websockets.server] connection closed
2022-03-12 10:10:06 INFO (MainThread) [ocpp] : send [2,"cab42fd7-d183-40cf-bff3-4db920796ec7","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2022-03-12 10:10:06 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 393, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 467, in get_supported_features
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 271, in call
    await self._send(call.to_json())
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 325, in _send
    await self._connection.send(message)
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 619, in send
    await self.ensure_open()
  File "/usr/local/lib/python3.9/site-packages/websockets/legacy/protocol.py", line 920, in ensure_open
    raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: sent 1000 (OK); then received 1005 (no status code [internal])`

lbbrhzn, 2022-03-12:
I wonder if the connection is dropped because SupportedFeatureProfiles is empty. According to the OCPP standard it should never be empty...

closed 2022-03-15 by lbbrhzn

mistrals, 2022-03-15:
`Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 274, in call
    await self._get_specific_response(call.unique_id,
  File "/config/custom_components/ocpp/api.py", line 817, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 307, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(),
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 394, in post_connect
    resp = await self.get_configuration(ckey.number_of_connectors.value)
  File "/config/custom_components/ocpp/api.py", line 752, in get_configuration
    resp = await self.call(req)
  File "/usr/local/lib/python3.9/site-packages/ocpp/charge_point.py", line 277, in call
    raise asyncio.TimeoutError(
asyncio.exceptions.TimeoutError: Waited 30s for response on [2,"96de07e1-1719-4341-ae9f-dfef0a0df634","GetConfiguration",{"key":["NumberOfConnectors"]}].
`

drc38, 2022-03-15:
Looks like its response to SupportedFeatureProfiles is still an issue. You'll need to raise the issue with the manufacturer.

PR #438 (2022-03-15): Handle unknown key in configure
May fix #432
commit 126ff263 (2022-03-15): Handle unknown key in configure (#438)

* Handle unknown key in configure

May fix #432

* Check for None entry

* Fix linting

* Add tests

* Fix test linting