lbbrhzn/ocpp #454: Missing metrics on Wallbox Pulsar Plus?

Strixx76, 2022-03-24:
## Version of the custom_component
0.4.26

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

In the documentation `Voltage` and `Frequency` is listed as "_Useful Entities for Wallbox Pulsar Plus_". I have a Pulsar Plus with current firmware version 5.5.10 but I do not get any values for those metrics.
Something wrong with my charger, firmware or is the documentation not correct?

## Debug log

```text

Add your logs here.

```

GrantKeymer, 2022-03-24:
I wrote the User Guide with the list of `Useful Entities`.  I'm still using v0.4.21 of this integration, but I don't know whether that makes any difference compared to the version you're using.  Please note that Voltage and Frequency only update during a charging session, not when the charger is idle.

Strixx76, 2022-03-24:
While reading through some posts in the Discussions I have noticed that we have different models even though they are all named Pulsar Plus.
My model number is: PLP1-M-2-4

And I can not get any readings when charging.

GrantKeymer, 2022-03-24:
Good point!  I have PLP1-0-2-2.  It will be interesting to see which version @lbbrhzn is using.

GrantKeymer, 2022-03-24:
[image]

It seems you have a Type 2 version with 7m cable and 3-phase 22kW capability.

GrantKeymer, 2022-03-24:
Here's a further thought, given that you have a 3-phase version.  Did you try looking in the Extra Attributes of the Voltage Sensor?

[image]

Strixx76, 2022-03-24:
I do have the 3-phase, 22kW, type 2 box with 7m cable.
The sensor value is `unknown`, and has no attributes.

lbbrhzn, 2022-03-24:
I'm running FW version 5.5.10 on my PLP1-0-2-4

lbbrhzn, 2022-03-24:
@GrantKeymer, Is it possible you have the additional readings because of the external power meter you installed?

Strixx76, 2022-03-24:
I also have the external power meter (Power Boost, Carlo Gavazzi EM340) using Modbus to communicating with the Pulsar Plus

GrantKeymer, 2022-03-24:
@lbbrhzn - No, I removed the external energy meter and sent it back many weeks ago.

lbbrhzn, 2022-03-24:
@Strixx76, I also have that. But I thought @GrantKeymer hade one configured differently, for Eco-Smart charging.

lbbrhzn, 2022-03-25:
I don't get Frequency or Voltage.

Strixx76, 2022-03-25:
Could it be that it is the EV that is reporting it to the charger?

lbbrhzn, 2022-03-25:
Seems unlikely.  I think we can close this issue, it is not related to the integration.

Strixx76, 2022-03-25:
Sure. Should the documentation be updated?

GrantKeymer, 2022-03-25:
> Sure. Should the documentation be updated?

Yes, I'll update the User Guide with a note that the Voltage and Frequency metrics don't work on 3-phase versions of the Pulsar Plus.

There are a few improvements I want to make, so this will be one of them.  Thanks guys for checking up on this issue.

closed 2022-03-27 by Strixx76